oe-alliance/oe-alliance-core #538: astra-sm 0.2

atvcaptain, 2019-10-02:
| gcc  -Wall -Wextra -I../lua -Os -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/oe1/atv64arm/build-enviroment/builds/openatv/release/osmio4kplus/tmp/work/aarch64-oe-linux/astra-sm/0.2-r0=/usr/src/debug/astra-sm/0.2-r0                      -fdebug-prefix-map=/home/oe1/atv64arm/build-enviroment/builds/openatv/release/osmio4kplus/tmp/work/aarch64-oe-linux/astra-sm/0.2-r0=/usr/src/debug/astra-sm/0.2-r0                      -fdebug-prefix-map=/home/oe1/atv64arm/build-enviroment/builds/openatv/release/osmio4kplus/tmp/work/aarch64-oe-linux/astra-sm/0.2-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/oe1/atv64arm/build-enviroment/builds/openatv/release/osmio4kplus/tmp/work/aarch64-oe-linux/astra-sm/0.2-r0/recipe-sysroot-native=  -std=c99 -pthread -fomit-frame-pointer -funsafe-loop-optimizations -ffast-math -fwrapv -fstrict-aliasing \
|     -o mkscript ./mkscript.c
| gcc: error: unrecognized command line option ‘-fmacro-prefix-map=/home/oe1/atv64arm/build-enviroment/builds/openatv/release/osmio4kplus/tmp/work/aarch64-oe-linux/astra-sm/0.2-r0=/usr/src/debug/astra-sm/0.2-r0’
| Makefile:488: recipe for target 'mkscript' failed
| make[1]: *** [mkscript] Error 1

nickersk, 2019-10-02:
same thing i think :-) have you tried to upgrade host gcc to version 8+?

atvcaptain, 2019-10-02:
i have now install gcc 8,x and check again

atvcaptain, 2019-10-02:
ok with gcc-8 works

closed 2019-10-02 by atvcaptain

Huevos, 2020-01-03:
What version of Ubuntu are you running. Is this not available in 16.04

Huevos, 2020-01-03:
Ok, I figured it out now.