oe-alliance/oe-alliance-core #539: gcc 9.20 break any

atvcaptain, 2019-10-02:
looks kernel build is different no drivers can be loaded

root@osmio4kplus:~# insmod /lib/modules/5.3.0/extra/ci.ko
insmod: ERROR: could not insert module /lib/modules/5.3.0/extra/ci.ko: Unknown symbol in module
root@osmio4kplus:~# dmesg
random: crng init done
ci: Unknown symbol __stack_chk_guard (err -2)
root@osmio4kplus:~#

nickersk, 2019-10-03:
imho gcc -fstack-protector causes this ...

atvcaptain, 2019-10-03:
i have send info to os mio driver team lets wait

atvcaptain, 2019-10-03:
feedback :  use gcc 8.3.0 for a while. Gcc 9.2 enables stack protector by default. It causes such unknown symbol error.

nickersk, 2019-10-03:
it can be turned off by default i think, var is in security config ... see oe-core/meta/conf/distro/include/security_flags.inc SECURITY_STACK_PROTECTOR ?= "-fstack-protector-strong"

atvcaptain, 2019-10-03:
ah good yes we must change it good you have found the config option

atvcaptain, 2019-10-03:
4.3 was identical the SECURITY_STACK_PROTECTOR

nickersk, 2019-10-03:
this means that drivers have been recompiled with different flags or stack protector isnt set now ?

atvcaptain, 2019-10-03:
i read disable speed up boot so its allways better for slow boxes to disable it
https://patchwork.ozlabs.org/patch/310365/

nickersk, 2019-10-03:
pls try osmio4kplus build with SECURITY_STACK_PROTECTOR = "" in layer.conf i'll look at qt5 issue now

atvcaptain, 2019-10-03:
ok

atvcaptain, 2019-10-03:
first test:
EBUG: Executing shell function do_compile_kernelmodules
NOTE: make HOSTCC=gcc  -isystem/home/oe1/atv64arm/build-enviroment/builds/openatv/release/osmio4kplus/tmp/work/osmio4kplus-oe-linux/linux-edision/5.3-.2/recipe-sysroot-native/usr/include **-fno-stack-protector** -O2 -pipe -L/home/oe1/atv64arm/build-enviroment/builds/openatv/release/osmio4kplus/tmp/work/osmio4kplus-oe-linux/linux-edision/5.3-.2/recipe-sysroot-native/usr/lib  

but not success

Oct  3 17:33:37 osmio4kplus user.warn kernel: ci: loading out-of-tree module taints kernel.
Oct  3 17:33:37 osmio4kplus user.warn kernel: ci: Unknown symbol __stack_chk_guard (err -2)
Oct  3 17:33:37 osmio4kplus user.warn kernel: ci: Unknown symbol __stack_chk_guard (err -2)
Oct  3 17:33:37 osmio4kplus user.warn kernel: si2183: module license 'Proprietary' taints kernel.
Oct  3 17:33:37 osmio4kplus user.warn kernel: Disabling lock debugging due to kernel taint
Oct  3 17:33:37 osmio4kplus user.warn kernel: si2183: Unknown symbol __stack_chk_guard (err -2)
Oct  3 17:33:37 osmio4kplus user.warn kernel: avl6862: Unknown symbol __stack_chk_guard (err -2)
Oct  3 17:33:37 osmio4kplus user.warn kernel: avl6261: Unknown symbol __stack_chk_guard (err -2)

nickersk, 2019-10-03:
so lets try with oe core default ...

atvcaptain, 2019-10-03:
cinfig say CONFIG_HAVE_STACKPROTECTOR=y

https://cateee.net/lkddb/web-lkddb/HAVE_STACKPROTECTOR.html

normal this generate the symbol

nickersk, 2019-10-03:
with fno stackprotector too ?

atvcaptain, 2019-10-03:
i have build kernel and kernel modules with -fno-stack-protector
no changes
i build now with gcc 8.x image new

nickersk, 2019-10-03:
ok

atvcaptain, 2019-10-03:
oh new issue with 8.x

Initialising tasks...ERROR: Multiple versions of lib32-glibc are due to be built (virtual:multilib:lib32:/home/oe1/atv64arm/build-enviroment/openembedded-core/meta/recipes-core/glibc/glibc_2.30.bb virtual:multilib:lib32:/home/oe1/atv64arm/build-enviroment/meta-oe-alliance/meta-oe/recipes-core/glibc/glibc_2.23.bb). Only one version of a given PN should be built in any given build. You likely need to set PREFERRED_VERSION_lib32-glibc to select the correct version or don't depend on multiple versions.

nickersk, 2019-10-03:
this is strange 2.23 is too old to buildsystem think it can be compiled against 2.23 :-) but ok if he needs it for mc lets try to set it.

atvcaptain, 2019-10-03:
i set 2.30 and other issue

Parsing of 4125 .bb files complete (0 cached, 4125 parsed). 9466 targets, 1474 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
NOTE: preferred version 8.% of lib32-libgcc not available (for item lib32-libgcc)
NOTE: versions of lib32-libgcc available: 4.9.4 9.2.0
NOTE: preferred version 8.% of lib32-gcc-runtime not available (for item lib32-libstdc++)
NOTE: versions of lib32-gcc-runtime available: 4.9.4 9.2.0
NOTE: preferred version 8.% of gcc-cross-aarch64 not available (for item virtual/aarch64-oe-linux-gcc)
NOTE: versions of gcc-cross-aarch64 available: 4.9.4 9.2.0
NOTE: preferred version 8.% of gcc-runtime not available (for item virtual/aarch64-oe-linux-compilerlibs)
NOTE: versions of gcc-runtime available: 4.9.4 9.2.0
NOTE: preferred version 8.% of lib32-libgcc-initial not available (for item lib32-libgcc-initial)
NOTE: versions of lib32-libgcc-initial available: 4.9.4 9.2.0
NOTE: preferred version 8.% of lib32-gcc-cross-arm not available (for item virtual/lib32-arm-oemllib32-linux-gnueabi-gcc)
NOTE: versions of lib32-gcc-cross-arm available: 4.9.4 9.2.0
NOTE: preferred version 8.% of lib32-gcc-cross-arm not available (for item virtual/lib32-arm-oemllib32-linux-gnueabi-g++)
NOTE: versions of lib32-gcc-cross-arm available: 4.9.4 9.2.0
NOTE: preferred version 8.% of lib32-libgcc not available (for item lib32-libgcc)
NOTE: versions of lib32-libgcc available: 4.9.4 9.2.0
NOTE: preferred version 8.% of lib32-gcc-runtime not available (for item virtual/lib32-arm-oemllib32-linux-gnueabi-compilerlibs)
NOTE: versions of lib32-gcc-runtime available: 4.9.4 9.2.0
NOTE: preferred version 8.% of gcc-crosssdk-x86_64-oesdk-linux not available (for item virtual/x86_64-oesdk-linux-gcc-crosssdk)
NOTE: versions of gcc-crosssdk-x86_64-oesdk-linux available: 4.9.4 9.2.0
NOTE: preferred version 8.% of nativesdk-libgcc-initial not available (for item nativesdk-libgcc-initial)
NOTE: versions of nativesdk-libgcc-initial available: 4.9.4 9.2.0
NOTE: preferred version 8.% of nativesdk-gcc-runtime not available (for item virtual/nativesdk-x86_64-oesdk-linux-compilerlibs)
NOTE: versions of nativesdk-gcc-runtime available: 4.9.4 9.2.0
NOTE: preferred version 8.% of gcc-runtime not available (for item libatomic)
NOTE: versions of gcc-runtime available: 4.9.4 9.2.0
NOTE: preferred version 8.% of libgcc not available (for item libgcc-dev)
NOTE: versions of libgcc available: 4.9.4 9.2.0
NOTE: preferred version 8.% of gcc-runtime not available (for item libstdc++)
NOTE: versions of gcc-runtime available: 4.9.4 9.2.0
NOTE: preferred version 8.% of gcc-runtime not available (for item libstdc++-dev)
NOTE: versions of gcc-runtime available: 4.9.4 9.2.0
NOTE: preferred version 8.% of gcc-runtime not available (for item libatomic-dev)
NOTE: versions of gcc-runtime available: 4.9.4 9.2.0
NOTE: preferred version 8.% of libgcc not available (for item libgcc)
NOTE: versions of libgcc available: 4.9.4 9.2.0
NOTE: preferred version 8.% of lib32-gcc-runtime not available (for item lib32-libatomic-dev)
NOTE: versions of lib32-gcc-runtime available: 4.9.4 9.2.0
NOTE: preferred version 8.% of lib32-libgcc not available (for item lib32-libgcc-dev)
NOTE: versions of lib32-libgcc available: 4.9.4 9.2.0
NOTE: preferred version 8.% of lib32-gcc-runtime not available (for item lib32-libatomic)
NOTE: versions of lib32-gcc-runtime available: 4.9.4 9.2.0
NOTE: preferred version 8.% of lib32-gcc-runtime not available (for item lib32-libstdc++-dev)
NOTE: versions of lib32-gcc-runtime available: 4.9.4 9.2.0
NOTE: preferred version 8.% of gcc-cross-canadian-aarch64 not available (for item gcc-cross-canadian-aarch64)
NOTE: versions of gcc-cross-canadian-aarch64 available: 4.9.4 9.2.0
NOTE: preferred version 8.% of gcc-cross-canadian-arm not available (for item gcc-cross-canadian-arm)
NOTE: versions of gcc-cross-canadian-arm available: 4.9.4 9.2.0
NOTE: preferred version 2.30 of glibc not available (for item virtual/aarch64-oe-linux-libc-for-gcc)
NOTE: versions of glibc available: 2.23
NOTE: preferred version 8.% of libgcc-initial not available (for item libgcc-initial)
NOTE: versions of libgcc-initial available: 4.9.4 9.2.0
NOTE: preferred version 8.% of gcc-cross-aarch64 not available (for item virtual/aarch64-oe-linux-g++)
NOTE: versions of gcc-cross-aarch64 available: 4.9.4 9.2.0
NOTE: preferred version 8.% of libgcc not available (for item libgcc)
NOTE: versions of libgcc available: 4.9.4 9.2.0
NOTE: preferred version 8.% of lib32-gcc-cross-initial-arm not available (for item virtual/lib32-arm-oemllib32-linux-gnueabi-gcc-initial)
NOTE: versions of lib32-gcc-cross-initial-arm available: 4.9.4
NOTE: preferred version 2.30 of lib32-glibc not available (for item virtual/lib32-arm-oemllib32-linux-gnueabi-libc-for-gcc)
NOTE: versions of lib32-glibc available: 2.23
NOTE: preferred version 2.30% of nativesdk-glibc not available (for item virtual/nativesdk-x86_64-oesdk-linux-libc-for-gcc)
NOTE: versions of nativesdk-glibc available: 2.23
NOTE: preferred version 8.% of gcc-crosssdk-initial-x86_64-oesdk-linux not available (for item virtual/x86_64-oesdk-linux-gcc-initial-crosssdk)
NOTE: versions of gcc-crosssdk-initial-x86_64-oesdk-linux available: 4.9.4
NOTE: preferred version 8.% of gcc-cross-initial-aarch64 not available (for item virtual/nativesdk-x86_64-oesdk-linux-gcc-initial)
NOTE: versions of gcc-cross-initial-aarch64 available: 4.9.4
[... cut ...]
NOTE: versions of lib32-gcc-runtime available: 4.9.4 9.2.0
NOTE: preferred version 8.% of nativesdk-gcc-runtime not available (for item nativesdk-gcc-runtime)
NOTE: versions of nativesdk-gcc-runtime available: 4.9.4 9.2.0
NOTE: preferred version 8.% of nativesdk-libgcc not available (for item nativesdk-libgcc)
NOTE: versions of nativesdk-libgcc available: 4.9.4 9.2.0
NOTE: preferred version 8.% of gcc-cross-initial-aarch64 not available (for item virtual/aarch64-oe-linux-gcc-initial)
NOTE: versions of gcc-cross-initial-aarch64 available: 4.9.4

atvcaptain, 2019-10-03:
ah shit i have only read 8.3  but oe-core have remove 
but forget delete all files :)

atvcaptain, 2019-10-05:
now fixed https://github.com/oe-alliance/oe-alliance-core/commit/a5325a5b7407a652891ec307385edd1d8b38cd51

closed 2019-10-05 by atvcaptain