oe-alliance/oe-alliance-core #540: qtbase 5.8

atvcaptain, 2019-10-03:
NOTE: recipe qtbase-5.8.0+gitAUTOINC+49dc9aa409-r0: task do_compile: Started
ERROR: qtbase-5.8.0+gitAUTOINC+49dc9aa409-r0 do_compile: oe_runmake failed
ERROR: qtbase-5.8.0+gitAUTOINC+49dc9aa409-r0 do_compile: Execution of '/home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/temp/run.do_compile.5193' failed with exit code 1:
cd src/ && ( test -e Makefile || /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot-native/usr/bin/qt5/qmake -o Makefile /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/git/src/src.pro -qtconf /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/build/bin/qt.conf -- -v -opensource -confirm-license -sysroot /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot -prefix /usr -bindir /usr/bin/qt5 -libdir /usr/lib -datadir /usr/share/qt5 -sysconfdir /etc/qt5 -docdir /usr/share/doc/qt5 -headerdir /usr/include/qt5 -archdatadir /usr/lib/qt5 -libexecdir /usr/lib/qt5/libexec -plugindir /usr/lib/qt5/plugins -importdir /usr/lib/qt5/imports -qmldir /usr/lib/qt5/qml -translationdir /usr/share/qt5/translations -testsdir /usr/share/qt5/tests -examplesdir /usr/share/qt5/examples -hostprefix /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot-native -hostlibdir /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot/usr/lib -hostbindir /usr/bin/qt5 -external-hostbindir /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot-native/usr/bin/qt5 -hostdatadir /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot/usr/lib/qt5 -platform linux-oe-g++ -xplatform linux-oe-g++ -shared -silent -no-pch -no-rpath -pkg-config -no-accessibility -no-cups -dbus -directfb -no-eglfs -evdev -nomake examples -no-fontconfig -system-freetype -no-gbm -no-gif -opengl es2 -no-glib -no-gtk -no-harfbuzz -no-ico -no-iconv -no-icu -no-journald -system-libjpeg -no-kms -no-libinput -system-libpng -no-libproxy -make libs -linuxfb -no-mtdev -openssl -qt-pcre -no-qml-debug -release -no-sm -no-sql-db2 -no-sql-ibase -no-sql-mysql -no-sql-oci -no-sql-odbc -no-sql-psql -no-sql-sqlite -no-sql-sqlite2 -no-sql-tds -nomake tests -make tools -no-tslib -libudev -widgets -no-xcb -no-xinput2 -no-xkb -no-xkbcommon -no-xkbcommon-evdev -no-xrender -system-zlib ) && make -f Makefile 
cd doc/ && ( test -e Makefile || /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot-native/usr/bin/qt5/qmake -o Makefile /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/git/doc/doc.pro -qtconf /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/build/bin/qt.conf -- -v -opensource -confirm-license -sysroot /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot -prefix /usr -bindir /usr/bin/qt5 -libdir /usr/lib -datadir /usr/share/qt5 -sysconfdir /etc/qt5 -docdir /usr/share/doc/qt5 -headerdir /usr/include/qt5 -archdatadir /usr/lib/qt5 -libexecdir /usr/lib/qt5/libexec -plugindir /usr/lib/qt5/plugins -importdir /usr/lib/qt5/imports -qmldir /usr/lib/qt5/qml -translationdir /usr/share/qt5/translations -testsdir /usr/share/qt5/tests -examplesdir /usr/share/qt5/examples -hostprefix /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot-native -hostlibdir /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot/usr/lib -hostbindir /usr/bin/qt5 -external-hostbindir /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot-native/usr/bin/qt5 -hostdatadir /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot/usr/lib/qt5 -platform linux-oe-g++ -xplatform linux-oe-g++ -shared -silent -no-pch -no-rpath -pkg-config -no-accessibility -no-cups -dbus -directfb -no-eglfs -evdev -nomake examples -no-fontconfig -system-freetype -no-gbm -no-gif -opengl es2 -no-glib -no-gtk -no-harfbuzz -no-ico -no-iconv -no-icu -no-journald -system-libjpeg -no-kms -no-libinput -system-libpng -no-libproxy -make libs -linuxfb -no-mtdev -openssl -qt-pcre -no-qml-debug -release -no-sm -no-sql-db2 -no-sql-ibase -no-sql-mysql -no-sql-oci -no-sql-odbc -no-sql-psql -no-sql-sqlite -no-sql-sqlite2 -no-sql-tds -nomake tests -make tools -no-tslib -libudev -widgets -no-xcb -no-xinput2 -no-xkb -no-xkbcommon -no-xkbcommon-evdev -no-xrender -system-zlib ) && make -f Makefile 
cd qmake/ && ( test -e Makefile.qmake-aux || /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot-native/usr/bin/qt5/qmake -o Makefile.qmake-aux /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/git/qmake/qmake-aux.pro -qtconf /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/build/bin/qt.conf -- -v -opensource -confirm-license -sysroot /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot -prefix /usr -bindir /usr/bin/qt5 -libdir /usr/lib -datadir /usr/share/qt5 -sysconfdir /etc/qt5 -docdir /usr/share/doc/qt5 -headerdir /usr/include/qt5 -archdatadir /usr/lib/qt5 -libexecdir /usr/lib/qt5/libexec -plugindir /usr/lib/qt5/plugins -importdir /usr/lib/qt5/imports -qmldir /usr/lib/qt5/qml -translationdir /usr/share/qt5/translations -testsdir /usr/share/qt5/tests -examplesdir /usr/share/qt5/examples -hostprefix /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot-native -hostlibdir /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot/usr/lib -hostbindir /usr/bin/qt5 -external-hostbindir /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot-native/usr/bin/qt5 -hostdatadir /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot/usr/lib/qt5 -platform linux-oe-g++ -xplatform linux-oe-g++ -shared -silent -no-pch -no-rpath -pkg-config -no-accessibility -no-cups -dbus -directfb -no-eglfs -evdev -nomake examples -no-fontconfig -system-freetype -no-gbm -no-gif -opengl es2 -no-glib -no-gtk -no-harfbuzz -no-ico -no-iconv -no-icu -no-journald -system-libjpeg -no-kms -no-libinput -system-libpng -no-libproxy -make libs -linuxfb -no-mtdev -openssl -qt-pcre -no-qml-debug -release -no-sm -no-sql-db2 -no-sql-ibase -no-sql-mysql -no-sql-oci -no-sql-odbc -no-sql-psql -no-sql-sqlite -no-sql-sqlite2 -no-sql-tds -nomake tests -make tools -no-tslib -libudev -widgets -no-xcb -no-xinput2 -no-xkb -no-xkbcommon -no-xkbcommon-evdev -no-xrender -system-zlib ) && make -f Makefile.qmake-aux 
make[1]: Entering directory '/home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/build/doc'
make[1]: Nothing to be done for 'first'.
[... cut ...]
| make: *** [sub-src-make_first] Error 2
| WARNING: /home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/temp/run.do_compile.5193:1 exit 1 from 'exit 1'
| 
NOTE: recipe qtbase-5.8.0+gitAUTOINC+49dc9aa409-r0: task do_compile: Failed
ERROR: Task (/home/oe1/atv64arm/build-enviroment/meta-qt5/recipes-qt/qt5/qtbase_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3181 tasks of which 3180 didn't need to be rerun and 1 failed.

nickersk, 2019-10-03:
rebuild didnt help ?

nickersk, 2019-10-04:
next todo: fix or remove zerotier for arm ...

nickersk, 2019-10-04:
nn there is a line saying -march=armv5 unknown target

atvcaptain, 2019-10-04:
strange this info
| /bin/sh: dpkg: command not found

nickersk, 2019-10-04:
this bb needs to be completely reworked, building zerotier from sources will not be that easy

atvcaptain, 2019-10-04:
we can first remove it, we have many other todo´s

atvcaptain, 2019-10-04:
Makefile  no idea we need the  -march option, when we remove it its solve some build issue

	@echo 'BUILD_OPTIMIZATION = "-march=$(OPT_ARCH) -O2 -pipe"' >> $@

nickersk, 2019-10-04:
-march=native is useless imho unless you build for intel arch ... :-D

atvcaptain, 2019-10-04:
ok i dont know why the option exist in pe-a , i revert my last revert :)

nickersk, 2019-10-04:
then i can revert nfs-utils fix too :-D

atvcaptain, 2019-10-04:
yes and i think zerotier works too

nickersk, 2019-10-04:
will try later as all of my servers + laptop are busy :-)

atvcaptain, 2019-10-04:
qt not 100% fixed
Makefile:44: recipe for target 'sub-src-make_first' failed
| make: *** [sub-src-make_first] Error 2
| ERROR: oe_runmake failed

ERROR: Task (/home/oe1/atv64arm/build-enviroment/meta-qt5/recipes-qt/qt5/qtlocation_git.bb:do_compile) failed with exit code '1'

nickersk, 2019-10-04:
yes, its unclear what is hapening ...segmentation fault means nothing ...

atvcaptain, 2019-10-04:
ah yes i see too

Makefile:866: recipe for target 'sensorproxy_interface.h' failed
make[4]: *** [sensorproxy_interface.h] Segmentation fault
make[4]: Leaving directory '/home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtsensors/5.8.0+gitAUTOINC+1220a31e6c-r0/build/src/plugins/sensors/iio-sensor-proxy'
Makefile:94: recipe for target 'sub-iio-sensor-proxy-make_first' failed
make[3]: *** [sub-iio-sensor-proxy-make_first] Error 2

nickersk, 2019-10-04:
i will try to bake them alone i got feeling something overheats hw resources

atvcaptain, 2019-10-04:
mutant say qt 5.10 is working too
i start compile

NOTE: recipe qtbase-native-5.10.1+gitAUTOINC+6c6ace9d23-r0:

atvcaptain, 2019-10-04:
first test i remove all patches and build without

atvcaptain, 2019-10-04:
build error too switch to 5.12

atvcaptain, 2019-10-04:
collect2: error: ld returned 1 exit status
Makefile:74: recipe for target 'extended' failed
make[4]: *** [extended] Error 1
make[4]: Leaving directory '/home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd51/tmp/work/hd51-oe-linux-gnueabi/qtdeclarative/5.12.3+gitAUTOINC+95c938cd67-r0/build/examples/qml/referenceexamples/extended'
Makefile:181: recipe for target 'sub-extended-make_first' failed
make[3]: *** [sub-extended-make_first] Error 2
make[3]: *** Waiting for unfinished jobs....

| Makefile:126: recipe for target 'sub-referenceexamples-make_first' failed
| make[2]: *** [sub-referenceexamples-make_first] Error 2
| make[2]: *** Waiting for unfinished jobs....

atvcaptain, 2019-10-04:
warrior build fail too :(

atvcaptain, 2019-10-04:
now check last 5.13

nickersk, 2019-10-04:
i will try to lower bbtasks number BB_NUMBER_THREADS = "14"
PARALLEL_MAKE = "-j 12"

betacentauri, 2019-10-04:
Don’t think that old napi plugin for hbbtv will work in 5.10 or higher.

nickersk, 2019-10-04:
plugin will not work,because netscape api was removed starting 5.9 or 5.10

atvcaptain, 2019-10-04:
hehe

last same issue

| Makefile:100: recipe for target 'sub-examples-make_first' failed
| make: *** [sub-examples-make_first] Error 2

nickersk, 2019-10-04:
https://github.com/oe-alliance/oe-alliance-core/commit/37908c719a231a89ec6734e3e113fe1de5051b75

atvcaptain, 2019-10-04:
good work

hd51 first box start 6.4 and hbbtv works good

closed 2019-10-04 by atvcaptain

zgemma-star, 2019-11-15:
> 
> 
> plugin will not work,because netscape api was removed starting 5.9 or 5.10

5.11 and 5.13 works good with netscape api

nickersk, 2019-11-15:
good to know, in fact it is hard to test qt for me as i have vusolose v2 only ... :-)