oe-alliance/oe-alliance-core #542: Samba build 4.4

betacentauri, 2019-10-05:
With latest change build fails.
OE don't expects to remove cluster support (ctdb) as install_append needs the files:
https://git.openembedded.org/meta-openembedded/tree/meta-networking/recipes-connectivity/samba/samba_4.10.8.bb#n179

Special install_append in bbappend is needed.

nickersk, 2019-10-05:
i'll take a look on this.

betacentauri, 2019-10-05:
My proposal would be to re enable cluster support.
In OE-A 4.3 was the problem that cluster support delivered several test files. It seems new samba receipt in 4.4 don't do it.

betacentauri, 2019-10-05:
But only cluster support and let the rest as it is.

nickersk, 2019-10-05:
reenabling cluster support and removing profiling only from configure causes this install error: /image/usr/sbin/samba_upgradedns: No such file or directory

betacentauri, 2019-10-05:
Strange

atvcaptain, 2019-10-05:
i see no python-samba in the 4.4 image !

nickersk, 2019-10-05:
w8 a moment pls, fix is on the way :-D

betacentauri, 2019-10-05:
So we can close?

atvcaptain, 2019-10-05:
i check now build too

nickersk, 2019-10-05:
@betacentauri 4.3 is ok? i cant check it right now ...

atvcaptain, 2019-10-05:
4.3 is ok

atvcaptain, 2019-10-05:
4.4 is fixed  image 3 mb smaller

closed 2019-10-05 by atvcaptain