oe-alliance/oe-alliance-core #543: osmio 4k kodi not start branch 4.4

atvcaptain, 2019-10-05:
kodi not start log show this

19:20:53.647 T:547600736144  NOTICE: special://profile/ is mapped to: special://masterprofile/
19:20:53.647 T:547600736144  NOTICE: -----------------------------------------------------------------------
19:20:53.647 T:547600736144  NOTICE: Starting Kodi (17.6 Git:2019-10-05-7e52c-stb). Platform: Linux ARM 64-bit
19:20:53.647 T:547600736144  NOTICE: Using Release Kodi x64 build
19:20:53.647 T:547600736144  NOTICE: Kodi compiled Oct  5 2019 by GCC 9.2.0 for Linux ARM 64-bit version 5.2.0 (328192)
19:20:53.661 T:547600736144  NOTICE: Running on Unknown Linux Distribution 0.0, kernel: Linux ARM 64-bit version 5.3.0
19:20:53.661 T:547600736144  NOTICE: FFmpeg version/source: ffmpeg-3.1-kodi
19:20:53.661 T:547600736144  NOTICE: Host CPU: ARMv8 Processor rev 0 (v8l), 2 cores available
19:20:53.661 T:547600736144  NOTICE: ARM Features: Neon disabled
19:20:53.661 T:547600736144  NOTICE: special://xbmc/ is mapped to: /usr/share/kodi
19:20:53.661 T:547600736144  NOTICE: special://xbmcbin/ is mapped to: /usr/lib64/kodi
19:20:53.661 T:547600736144  NOTICE: special://xbmcbinaddons/ is mapped to: /usr/lib64/kodi/addons
19:20:53.661 T:547600736144  NOTICE: special://masterprofile/ is mapped to: /media/hdd/.kodi/userdata
19:20:53.661 T:547600736144  NOTICE: special://envhome/ is mapped to: /media/hdd
19:20:53.661 T:547600736144  NOTICE: special://home/ is mapped to: /media/hdd/.kodi
19:20:53.661 T:547600736144  NOTICE: special://temp/ is mapped to: /tmp/kodi
19:20:53.661 T:547600736144  NOTICE: special://logpath/ is mapped to: /tmp/kodi
19:20:53.661 T:547600736144  NOTICE: The executable running is: /usr/lib64/kodi/kodi.bin
19:20:53.661 T:547600736144  NOTICE: Local hostname: osmio4kplus
19:20:53.661 T:547600736144  NOTICE: Log File is located: /tmp/kodi//kodi.log
19:20:53.661 T:547600736144  NOTICE: -----------------------------------------------------------------------
19:20:53.662 T:547600736144  NOTICE: load settings...
19:20:53.663 T:547600736144   ERROR: CSettings: error loading settings definition from special://xbmc/system/settings/settings.xml, Line 0
                                            Failed to open file
19:20:53.663 T:547600736144   FATAL: Unable to load settings definitions

all files exist with branch 4.3 all working no idea way

nickersk, 2019-10-05:
permissions ?

atvcaptain, 2019-10-05:
oh yes funny i have test other 4.4 box and np

nickersk, 2019-10-05:
weird :-)

atvcaptain, 2019-10-05:
strange kodi create new files and cant read any more next issue

ERROR: CSettings: unable to load settings from special://masterprofile/guisettings.xml, creating new default settings

nickersk, 2019-10-05:
/media/hdd is flash or usb device?

atvcaptain, 2019-10-05:
usb hdd root have 777 rights but kodi create new files and rights lost

atvcaptain, 2019-10-05:
but strange test today sf8008 and kodi have works

nickersk, 2019-10-05:
something must modify them but how to find what...

atvcaptain, 2019-10-05:
6.3 create kodi files with 750
and 6.4 with 010

atvcaptain, 2019-10-05:
kodi is starting when i use a 6.3 userhome 
but log show this

20:39:35.828 T:547981541264   ERROR: GetConverter: iconv_open() for "UTF-8" -> "UTF-32LE" failed, errno = 22 (Invalid argument)
20:39:35.828 T:547981541264   ERROR: Previous line repeats 11 times.

and some gui png not show correct

nickersk, 2019-10-05:
there is gconv in rrecommends but maybe its job for iconv

vuthanhtungLLC, 2019-11-01:
Yes

68foxboris, 2019-11-03:
And kodi not start on my gi et 11000 4k.

21:36:14.334 T:3019590416  NOTICE: Starting Kodi (17.6 Git:2019-10-29-7e52c-stb). Platform: Linux ARM 32-bit
21:36:14.334 T:3019590416  NOTICE: Using Release Kodi x32 build
21:36:14.334 T:3019590416  NOTICE: Kodi compiled Oct 29 2019 by GCC 8.2.0 for Linux ARM 32-bit version 4.18.0 (266752)
21:36:14.334 T:3019590416  NOTICE: Running on openpli develop, kernel: Linux ARM 32-bit version 4.1.37
21:36:14.334 T:3019590416  NOTICE: FFmpeg version/source: ffmpeg-3.1-kodi
21:36:14.334 T:3019590416  NOTICE: Host CPU: ARMv7 Processor rev 3 (v7l), 2 cores available
21:36:14.334 T:3019590416  NOTICE: ARM Features: Neon enabled
21:36:14.690 T:3019590416  NOTICE: Loaded playercorefactory configuration
21:36:14.690 T:3019590416  NOTICE: Loading player core factory settings from special://masterprofile/playercorefactory.xml.
21:36:14.690 T:3019590416  NOTICE: special://masterprofile/playercorefactory.xml does not exist. Skipping.
19:36:14.700 T:3019590416  NOTICE: Running database version Addons27
19:36:14.737 T:3019590416  NOTICE: ADDONS: Using repository repository.xbmc.org
19:36:14.740 T:2988438448   ERROR: SetPriority: error Permission denied
19:36:14.966 T:3019590416  NOTICE: InitWindowSystem: Using EGL Implementation: NexTV
and
2019-10-29 21:36:14+0200 [-] [Console] command: ps | grep kodi.bin | grep -v grep
2019-10-29 21:36:14+0200 [-] [Console] finished: ps | grep kodi.bin | grep -v grep
2019-10-29 21:36:14+0200 [-] [KodiLauncher] FBLock
2019-10-29 21:36:14+0200 [-] [KodiLauncher] startup: kodi is not running, starting...
2019-10-29 21:36:14+0200 [-] [Console] command: unset PYTHONPATH;kodi;kodiext -T
2019-10-29 21:36:14+0200 [-] 
2019-10-29 21:36:14+0200 [-] [Console] finished: unset PYTHONPATH;kodi;kodiext -T
2019-10-29 21:36:14+0200 [-] [KodiLauncher] kodi stopped: retval = 0
2019-10-29 21:36:15+0200 [-] [KodiLauncher] FBUnlock

atvcaptain, 2019-11-03:
have nothing todo with 4.4 build see build with 4.3 ->   GCC 8.2.0

68foxboris, 2019-11-03:
Sorry:
07:40:04.003 T:3020056192  NOTICE: special://profile/ is mapped to: special://masterprofile/
07:40:04.003 T:3020056192  NOTICE: -----------------------------------------------------------------------
07:40:04.003 T:3020056192  NOTICE: Starting Kodi (17.6 Git:2019-11-02-7e52c-stb). Platform: Linux ARM 32-bit
07:40:04.003 T:3020056192  NOTICE: Using Release Kodi x32 build
07:40:04.003 T:3020056192  NOTICE: Kodi compiled Nov  2 2019 by GCC 9.2.0 for Linux ARM 32-bit version 5.2.0 (328192)
07:40:04.003 T:3020056192  NOTICE: Running on openpli develop, kernel: Linux ARM 32-bit version 4.1.37
07:40:04.029 T:3020056192  NOTICE: FFmpeg version/source: ffmpeg-3.1-kodi
07:40:04.030 T:3020056192  NOTICE: Host CPU: ARMv7 Processor rev 3 (v7l), 2 cores available
07:40:04.030 T:3020056192  NOTICE: ARM Features: Neon enabled
07:40:04.030 T:3020056192  NOTICE: special://xbmc/ is mapped to: /usr/share/kodi
07:40:04.030 T:3020056192  NOTICE: special://xbmcbin/ is mapped to: /usr/lib/kodi
07:40:04.030 T:3020056192  NOTICE: special://xbmcbinaddons/ is mapped to: /usr/lib/kodi/addons
07:40:04.030 T:3020056192  NOTICE: special://masterprofile/ is mapped to: /media/hdd/.kodi/userdata
07:40:04.030 T:3020056192  NOTICE: special://envhome/ is mapped to: /media/hdd
07:40:04.030 T:3020056192  NOTICE: special://home/ is mapped to: /media/hdd/.kodi
07:40:04.030 T:3020056192  NOTICE: special://temp/ is mapped to: /tmp/kodi
07:40:04.030 T:3020056192  NOTICE: special://logpath/ is mapped to: /tmp/kodi
07:40:04.030 T:3020056192  NOTICE: The executable running is: /usr/lib/kodi/kodi.bin
07:40:04.030 T:3020056192  NOTICE: Local hostname: et1x000
07:40:04.030 T:3020056192  NOTICE: Log File is located: /tmp/kodi//kodi.log
07:40:04.051 T:3020056192  NOTICE: -----------------------------------------------------------------------
07:40:04.176 T:3020056192  NOTICE: load settings...
07:40:04.226 T:3020056192 WARNING: CSettingInt: unknown options filler "audiocdactions" of "audiocds.autoaction"
07:40:04.670 T:3020056192  NOTICE: Found 1 Lists of Devices
07:40:04.670 T:3020056192  NOTICE: Enumerated ALSA devices:
07:40:04.670 T:3020056192  NOTICE:     Device 1
07:40:04.670 T:3020056192  NOTICE:         m_deviceName      : default
07:40:04.670 T:3020056192  NOTICE:         m_displayName     : Default (BRCM PCM BRCM PCM)
07:40:04.670 T:3020056192  NOTICE:         m_displayNameExtra:
07:40:04.670 T:3020056192  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
07:40:04.670 T:3020056192  NOTICE:         m_channels        : FL,FR,BL,BR,FC,LFE,SL,SR,UNKNOWN1,UNKNOWN2,UNKNOWN3,UNKNOWN4,UNKNOWN5,UNKNOWN6,UNKNOWN7,UNKNOWN8
07:40:04.670 T:3020056192  NOTICE:         m_sampleRates     : 8000,11025,16000,22050,32000,44100,48000,64000,88200,96000,176400,192000
07:40:04.670 T:3020056192  NOTICE:         m_dataFormats     : AE_FMT_S32NE,AE_FMT_S16NE,AE_FMT_S16LE
07:40:04.670 T:3020056192  NOTICE:         m_streamTypes     : No passthrough capabilities
07:40:04.713 T:3020056192  NOTICE: No settings file to load (special://xbmc/system/advancedsettings.xml)
07:40:04.713 T:3020056192  NOTICE: No settings file to load (special://masterprofile/advancedsettings.xml)
07:40:04.713 T:3020056192  NOTICE: Default Video Player: VideoPlayer
07:40:04.713 T:3020056192  NOTICE: Default Audio Player: paplayer
07:40:04.713 T:3020056192  NOTICE: Disabled debug logging due to GUI setting. Level 0.
07:40:04.713 T:3020056192  NOTICE: Log level changed to "LOG_LEVEL_NORMAL"
07:40:04.713 T:3020056192  NOTICE: Loading player core factory settings from special://xbmc/system/playercorefactory.xml.
07:40:04.726 T:3020056192 WARNING: CPlayerCoreFactory::GetPlayer(ExtEplayer3): no such player: ExtEplayer3
07:40:04.726 T:3020056192 WARNING: CPlayerCoreFactory::GetPlayer(GstPlayer): no such player: GstPlayer
07:40:04.726 T:3020056192 WARNING: CPlayerCoreFactory::GetPlayer(E2Player): no such player: E2Player
07:40:04.726 T:3020056192  NOTICE: Loaded playercorefactory configuration
07:40:04.726 T:3020056192  NOTICE: Loading player core factory settings from special://masterprofile/playercorefactory.xml.
07:40:04.726 T:3020056192  NOTICE: special://masterprofile/playercorefactory.xml does not exist. Skipping.
05:40:04.835 T:3020056192  NOTICE: Running database version Addons27
05:40:05.437 T:3020056192  NOTICE: ADDONS: Using repository repository.xbmc.org
05:40:05.441 T:2989487008   ERROR: SetPriority: error Permission denied
05:40:05.778 T:3020056192  NOTICE: InitWindowSystem: Using EGL Implementation: NexTV

atvcaptain, 2019-11-03:
have nothing todo with oe-a .> Running on openpli develop, kernel:

atvcaptain, 2019-11-04:
i am rollback opkg to old version no change

issue was wrong file and folder permissions

ipk show all correct but after install permissions corrupt

root@osmio4kplus:~# ls -l /usr/lib64/kodi/
**------x---**    1 root     root          2481 Oct 21 15:08 AddOptions.cmake
------x---    1 root     root         12822 Oct 21 15:08 AddonHelpers.cmake
------x---    1 root     root          5730 Oct 21 15:08 ArchSetup.cmake
------x---    1 root     root          1924 Oct 21 15:08 CMakeHelpers.cmake
------x---    1 root     root          2321 Oct 21 15:08 CheckCommits.cmake
------x---    1 root     root          2760 Oct 21 15:08 CheckTargetPlatform.cmake
------x---    1 root     root           902 Oct 21 15:08 GenerateVersionedFiles.cmake
------x---    1 root     root          1965 Oct 21 15:08 GeneratorSetup.cmake
------x---    1 root     root          9679 Oct 21 15:08 HandleDepends.cmake
------x---    1 root     root          1072 Oct 21 15:08 KodiConfig.cmake
------x---    1 root     root         27746 Oct 21 15:08 Macros.cmake
------x---    1 root     root          3743 Oct 21 15:08 PrepareEnv.cmake
------x---    1 root     root          3701 Oct 21 15:08 ProjectMacros.cmake
------x---    1 root     root           758 Oct 21 15:08 Uninstall.cmake
drwxr-xr-x   12 root     root          1024 Oct 21 15:08 addons
------x---    1 root     root      27055560 Oct 21 15:08 kodi.bin
drwxr-xr-x    2 root     root          1024 Oct 21 15:08 system

nickersk, 2019-11-04:
hmm, cannot this be issue with busybox nice command not being suid?

atvcaptain, 2019-11-04:
with oe- 4.3 all ok

root@osmio4kplus:~# ls -l /usr/bin/ko*
-rwxr-x--T    1 root     root          5909 Oct  5 15:05 /usr/bin/kodi
-rwxr-x--T    1 root     root          1614 Oct  5 15:05 /usr/bin/kodi-standalone
-rwxr-x--T    1 root     root         14552 Oct  5 15:05 /usr/bin/kodiext

root@osmio4kplus:~# ls -l /usr/lib64/kodi/
-rwxr-x--T    1 root     root          2481 Oct  5 15:05 AddOptions.cmake
-rwxr-x--T    1 root     root         12822 Oct  5 15:05 AddonHelpers.cmake
-rwxr-x--T    1 root     root          5730 Oct  5 15:05 ArchSetup.cmake
-rwxr-x--T    1 root     root          1924 Oct  5 15:05 CMakeHelpers.cmake
-rwxr-x--T    1 root     root          2321 Oct  5 15:05 CheckCommits.cmake
-rwxr-x--T    1 root     root          2760 Oct  5 15:05 CheckTargetPlatform.cmake
-rwxr-x--T    1 root     root           902 Oct  5 15:05 GenerateVersionedFiles.cmake
-rwxr-x--T    1 root     root          1965 Oct  5 15:05 GeneratorSetup.cmake
-rwxr-x--T    1 root     root          9679 Oct  5 15:05 HandleDepends.cmake
-rwxr-x--T    1 root     root          1072 Oct  5 15:05 KodiConfig.cmake
-rwxr-x--T    1 root     root         27746 Oct  5 15:05 Macros.cmake
-rwxr-x--T    1 root     root          3743 Oct  5 15:05 PrepareEnv.cmake
-rwxr-x--T    1 root     root          3701 Oct  5 15:05 ProjectMacros.cmake
-rwxr-x--T    1 root     root           758 Oct  5 15:05 Uninstall.cmake
drwxr-xr-x   12 root     root          4096 Oct  5 15:05 addons
-rwxr-x--T    1 root     root      26994152 Oct  5 15:05 kodi.bin
drwxr-xr-x    2 root     root          4096 Oct  5 15:05 system

nickersk, 2019-11-04:
they all have sticky bit set?

atvcaptain, 2019-11-04:
ipk looks ok for me

next issue was when i fix the permission

kodi start and the profile was create and the permission was not ok too

12:15:08.316 T:547748835216   ERROR: CSettings: unable to load settings from special://masterprofile/guisettings.xml, creating new default settings

the file have only x bit no rw kodi cant start, i fix this and kodi works

nickersk, 2019-11-04:
oh, maybe create postinst_ontarget with setting up correct permissions ?

atvcaptain, 2019-11-04:
no i use the internel emmc flash box box 1,3gb free, 

very strange, and why permission are different after remove and install again

other box like hd51 all ok

nickersk, 2019-11-04:
driver issue ?

atvcaptain, 2019-11-04:
or opkg have issue when to many ipks installed to same time

nickersk, 2019-11-04:
what about trying to change permissions in do_install_append ? like chmod 0750 ${D}/${libdir}/ ...
this will prevent wrong modes for ipk creation...

atvcaptain, 2019-11-05:
The file permission problem goes away if you exclude libopen.so from the preload list in enigma2.sh.

#check for sundtek tuner helper lib
if [ -e /opt/lib/libmediaclient.so ]; then
	LIBS="/opt/lib/libmediaclient.so /usr/lib64/libopen.so.0.0.0"
else
	LIBS="/usr/lib64/libopen.so.0.0.0"
fi

the problem is not limited to kodi but appears in every file installed by the e2 plugin browser. I think this is related to the glibc version and libopen.so. We must check to remove it if possible. It doesn’t seem necessary now.

possible other crash like uianimation we can fix it too

nickersk, 2019-11-05:
--- a/tools/libopen.c
+++ b/tools/libopen.c
@@ -17,7 +17,7 @@ int open64(const char *pathname, int flags, ...)
        {
                void *handle;
                char *error;
-               handle = dlopen(LIBC_SO, RTLD_LAZY);
+               handle = dlopen("libc.so.6", RTLD_LAZY);
                if (!handle)
                {
                        fputs(dlerror(), stderr);
@@ -55,7 +55,7 @@ int open(const char *pathname, int flags, ...)
        {
                void *handle;
                char *error;
-               handle = dlopen(LIBC_SO, RTLD_LAZY);
+               handle = dlopen("libc.so.6", RTLD_LAZY);
                if (!handle)
                {
                        fputs(dlerror(), stderr);
@@ -93,7 +93,7 @@ FILE *fopen64(const char *pathname, const char *mode)
        {
                void *handle;
                char *error;
-               handle = dlopen(LIBC_SO, RTLD_LAZY);
+               handle = dlopen("libc.so.6", RTLD_LAZY);
                if (!handle)
                {
                        fputs(dlerror(), stderr);
@@ -132,7 +132,7 @@ FILE *fopen(const char *pathname, const char *mode)
        {
                void *handle;
                char *error;
-               handle = dlopen(LIBC_SO, RTLD_LAZY);
+               handle = dlopen("libc.so.6", RTLD_LAZY);
                if (!handle)
                {
                        fputs(dlerror(), stderr);
@@ -171,7 +171,7 @@ int socket(int domain, int type, int protocol)
        {
                void *handle;
                char *error;
-               handle = dlopen(LIBC_SO, RTLD_LAZY);
+               handle = dlopen("libc.so.6", RTLD_LAZY);
                if (!handle)
                {
                        fputs(dlerror(), stderr);
@@ -209,7 +209,7 @@ int accept(int sockfd, struct sockaddr *addr, socklen_t *addrlen)
        {
                void *handle;
                char *error;
-               handle = dlopen(LIBC_SO, RTLD_LAZY);
+               handle = dlopen("libc.so.6", RTLD_LAZY);
                if (!handle)
                {
                        fputs(dlerror(), stderr);
@@ -246,7 +246,7 @@ int socketpair(int d, int type, int protocol, int sv[2])
        {
                void *handle;
                char *error;
-               handle = dlopen(LIBC_SO, RTLD_LAZY);
+               handle = dlopen("libc.so.6", RTLD_LAZY);
                if (!handle)
                {
                        fputs(dlerror(), stderr);
@@ -289,7 +289,7 @@ int pipe(int modus[2])
        {
                void *handle;
                char *error;
-               handle = dlopen(LIBC_SO, RTLD_LAZY);
+               handle = dlopen("libc.so.6", RTLD_LAZY);
                if (!handle)
                {
                        fputs(dlerror(), stderr);

nickersk, 2019-11-05:
"                void *handle;
                char *error;
-               handle = dlopen(LIBC_SO, RTLD_LAZY);
+               handle = dlopen("libc.so.6", RTLD_LAZY);
                if (!handle)
                {
                        fputs(dlerror(), stderr);
@@ -246,7 +246,7 @@ int socketpair(int d, int type, int protocol, int sv[2])
        {
                void *handle;
                char *error;
-               handle = dlopen(LIBC_SO, RTLD_LAZY);
+               handle = dlopen("libc.so.6", RTLD_LAZY);
                if (!handle)
                {
                        fputs(dlerror(), stderr);
@@ -289,7 +289,7 @@ int pipe(int modus[2])
        {
                void *handle;
                char *error;
-               handle = dlopen(LIBC_SO, RTLD_LAZY);
+               handle = dlopen("libc.so.6", RTLD_LAZY);
                if (!handle)
                {
                        fputs(dlerror(), stderr);
"

nickersk, 2019-11-05:
LIBC_SO = /lib/libc.so.6 ? if yes, then:
https://github.com/nickersk/enigma2/commit/cedb22fdd2915e3aa43c25b8d45fe500a921e179

68foxboris, 2019-11-18:
kodi it works now

68foxboris, 2019-11-28:
A very strange error kodi worked yesterday.But now it does not start-terminal
root@et1x000:~# kodi
### unable to connect to ipc server
*** Unable to join nxclient_ipc_thread_regular
/usr/bin/kodi: line 195:  1715 Segmentation fault      "$LIBDIR/${bin_name}/${bin_name}.bin" $SAVED_ARGS

nickersk, 2019-11-28:
out of system resources,maybe?

68foxboris, 2019-11-28:
Clean installation only kodi.I uninstalled and installed on a new one that does not work.

68foxboris, 2019-11-28:
And this should not be for kodi
https://gitlab.freedesktop.org/gstreamer/meson-ports/gl-headers

nickersk, 2019-11-28:
newflash older image and try again?

nickersk, 2019-11-28:
when it worked and now d

68foxboris, 2019-11-28:
Nothing has changed.

nickersk, 2019-11-28:
does kodi start from e2 plugin screen?

68foxboris, 2019-11-29:
I think I found why the kodi does not work, but I still need to check.It seems to be related to the file defconfig.
05:50:27.798 T:3022065680  NOTICE: Starting Kodi (17.6 Git:2019-11-28-7e52c-stb). Platform: Linux ARM 32-bit
05:50:27.798 T:3022065680  NOTICE: Using Release Kodi x32 build
05:50:27.798 T:3022065680  NOTICE: Kodi compiled Nov 28 2019 by GCC 9.2.0 for Linux ARM 32-bit version 5.2.0 (328192)
05:50:27.801 T:3022065680  NOTICE: FFmpeg version/source: ffmpeg-3.1-kodi
05:50:27.801 T:3022065680  NOTICE: Host CPU: ARMv7 Processor rev 3 (v7l), 2 cores available
05:50:27.801 T:3022065680  NOTICE: ARM Features: Neon enabled
03:50:28.512 T:2991584144   ERROR: SetPriority: error Permission denied
03:50:28.761 T:3022065680  NOTICE: InitWindowSystem: Using EGL Implementation: NexTV
03:50:28.799 T:3022065680  NOTICE: Found resolution 1280 x 720 for display 0 with 1280 x 720 @ 50.000000 Hz
03:50:28.799 T:3022065680  NOTICE: Previous line repeats 2 times.
03:50:28.799 T:3022065680  NOTICE: Found resolution 1280 x 720 for display 0 with 1920 x 1080 @ 50.000000 Hz
03:50:28.799 T:3022065680  NOTICE: Found (1280x720@50.000000) at 16, setting to RES_DESKTOP at 16
03:50:28.799 T:3022065680  NOTICE: Checking resolution 16
03:50:28.805 T:3022065680  NOTICE: GL_VENDOR = Broadcom
03:50:28.805 T:3022065680  NOTICE: GL_RENDERER = VideoCore V HW (V3D-530)
03:50:28.805 T:3022065680  NOTICE: GL_VERSION = OpenGL ES 3.1
03:50:28.805 T:3022065680  NOTICE: GL_SHADING_LANGUAGE_VERSION = OpenGL ES GLSL ES 3.10
03:50:28.805 T:3022065680  NOTICE: GL_EXTENSIONS = GL_BRCM_mirror_once_border GL_BRCM_polygon_mode GL_BRCM_texture_1D GL_BRCM_texture_norm16 GL_EXT_color_buffer_float GL_EXT_debug_marker GL_EXT_discard_framebuffer GL_EXT_draw_elements_base_vertex GL_EXT_multi_draw_indirect GL_EXT_multisampled_render_to_texture GL_EXT_primitive_bounding_box GL_EXT_protected_textures GL_EXT_robustness GL_EXT_shader_integer_mix GL_EXT_shader_texture_lod GL_EXT_texture_filter_anisotropic GL_EXT_texture_format_BGRA8888 GL_KHR_blend_equation_advanced GL_KHR_blend_equation_advanced_coherent GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_EGL_image GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_compressed_ETC1_RGB8_texture GL_OES_compressed_paletted_texture GL_OES_depth24 GL_OES_draw_elements_base_vertex GL_OES_mapbuffer GL_OES_packed_depth_stencil GL_OES_primitive_bounding_box GL_OES_rgb8_rgba8 GL_OES_shader_image_atomic GL_OES_standard_derivatives GL_OES_surfaceless_context GL_OES_texture_npot GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_OES_vertex_array_object GL_OES_vertex_half_float
03:50:29.341 T:3002069904  NOTICE: Running database version Addons27
03:50:29.344 T:3002069904  NOTICE: Running database version ViewModes6
03:50:29.347 T:3002069904  NOTICE: Running database version Textures13
03:50:29.382 T:3002069904  NOTICE: Running database version MyMusic60
03:50:29.442 T:3002069904  NOTICE: Running database version MyVideos107
03:50:29.445 T:3002069904  NOTICE: Running database version TV29
03:50:29.452 T:3002069904  NOTICE: Running database version Epg11
03:50:29.767 T:3022065680 WARNING: JSONRPC: Could not parse type "Setting.Details.SettingList"
03:50:30.037 T:3022065680  NOTICE: initialize done
03:50:30.037 T:3022065680  NOTICE: Running the application...
03:50:30.050 T:3022065680  NOTICE: starting zeroconf publishing
03:50:30.051 T:3022065680  NOTICE: starting upnp client
03:50:30.070 T:2588566416  NOTICE: ES: Starting UDP Event server on port 9777
03:50:30.071 T:2588566416  NOTICE: UDP: Listening on port 9777 (ipv6 : false)

atvcaptain, 2020-03-27:
fixed with last updates

closed 2020-03-27 by atvcaptain