oe-alliance/oe-alliance-core #544: android-tools_5.1.1.r37  (4.4)

atvcaptain, 2019-10-07:
/home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd60/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/android-tools/5.1.1.r37-r0/git/system/core/adb/adb_auth_host.c:86:5: warning: implicit declaration of function 'RSA_get0_key'; did you mean 'RSA_check_key'? [-Wimplicit-function-declaration]
   86 |     RSA_get0_key(rsa, &n, &e, NULL);
      |     ^~~~~~~~~~~~
      |     RSA_check_key
arm-oe-linux-gnueabi-gcc-ar rcs libcutils.a atomic.o hashmap.o native_handle.o config_utils.o cpu_info.o load_file.o str_parms.o fs.o multiuser.o socket_inaddr_any_server.o socket_local_client.o socket_local_server.o socket_loopback_client.o socket_loopback_server.o socket_network_client.o sockets.o ashmem-host.o dlmalloc_stubs.o
arm-oe-linux-gnueabi-gcc  -mfpu=neon-vfpv4 -mfloat-abi=hard -mcpu=cortex-a15 --sysroot=/home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd60/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/android-tools/5.1.1.r37-r0/recipe-sysroot -o adb -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed adb.o console.o transport.o transport_local.o transport_usb.o commandline.o adb_client.o adb_auth_host.o sockets.o services.o file_sync_client.o get_my_path_linux.o usb_linux.o usb_vendors.o fdevent.o libcutils.a -lpthread -lcrypto
/home/oe1/atv64arm/build-enviroment/builds/openatv/release/hd60/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/android-tools/5.1.1.r37-r0/recipe-sysroot-native/usr/bin/arm-oe-linux-gnueabi/../../libexec/arm-oe-linux-gnueabi/gcc/arm-oe-linux-gnueabi/9.2.0/ld: adb_auth_host.o: in function `RSA_to_RSAPublicKey':
adb_auth_host.c:(.text+0x1bc): undefined reference to `RSA_get0_key'
collect2: error: ld returned 1 exit status

atvcaptain, 2019-10-07:
i am back in 12 h now on work

atvcaptain, 2019-10-07:
i found this looks like a solution  we can use too:  https://github.com/cossacklabs/themis/pull/258/files#diff-b1a8d20c2dabfcf4753b5240d2d3fba0R261

nickersk, 2019-10-07:
yes .

closed 2019-10-07 by atvcaptain