oe-alliance/oe-alliance-core #545: all vu uianimation crashing (Build 4.4)

atvcaptain, 2019-10-13:
16:25:50.3096 [   ] bsod.cpp:402 print_backtrace Backtrace:
16:25:50.3098 [   ] bsod.cpp:410 print_backtrace enigma2(_Z17handleFatalSignaliP9siginfo_tPv) [0x91044]
16:25:50.3099 [   ] bsod.cpp:410 print_backtrace /lib/libc.so.6(__default_rt_sa_restorer) [0xB5DEC8F0]
16:25:50.3100 [   ] bsod.cpp:410 print_backtrace /lib/libc.so.6(gsignal) [0xB5DEBAB0]
16:25:50.3100 [   ] bsod.cpp:410 print_backtrace /lib/libc.so.6(abort) [0xB5DD8840]
16:25:50.3101 [   ] bsod.cpp:410 print_backtrace /usr/lib/libv3ddriver.so(_ZN9__gnu_cxx27__verbose_terminate_handlerEv) [0xB631CFDC]
16:25:50.3102 [   ] bsod.cpp:410 print_backtrace /usr/lib/libv3ddriver.so(_ZN9__gnu_cxx27__verbose_terminate_handlerEv) [0xB631CFDC]
16:25:50.3102 [   ] bsod.cpp:410 print_backtrace /usr/lib/libv3ddriver.so(_ZN9__gnu_cxx27__verbose_terminate_handlerEv) [0xB631CFDC]
16:25:50.3103 [   ] bsod.cpp:410 print_backtrace /usr/lib/libv3ddriver.so(_ZN9__gnu_cxx27__verbose_terminate_handlerEv) [0xB631CFDC]
16:25:50.3104 [   ] bsod.cpp:410 print_backtrace /usr/lib/libv3ddriver.so(_ZN9__gnu_cxx27__verbose_terminate_handlerEv) [0xB631CFDC]
16:25:50.3105 [   ] bsod.cpp:410 print_backtrace /usr/lib/libv3ddriver.so(_ZN9__gnu_cxx27__verbose_terminate_handlerEv) [0xB631CFDC]
16:25:50.3106 [   ] bsod.cpp:410 print_backtrace /usr/lib/libv3ddriver.so(_ZN9__gnu_cxx27__verbose_terminate_handlerEv) [0xB631CFDC]
16:25:50.3106 [   ] bsod.cpp:410 print_backtrace /usr/lib/libv3ddriver.so(_ZN9__gnu_cxx27__verbose_terminate_handlerEv) [0xB631CFDC]
16:25:50.3107 [   ] bsod.cpp:410 print_backtrace /usr/lib/libv3ddriver.so(_ZN9__gnu_cxx27__verbose_terminate_handlerEv) [0xB631CFDC]
16:25:50.3108 [   ] bsod.cpp:410 print_backtrace /usr/lib/libv3ddriver.so(_ZN9__gnu_cxx27__verbose_terminate_handlerEv) [0xB631CFDC]
16:25:50.3108 [   ] bsod.cpp:422 handleFatalSignal -------FATAL SIGNAL

atvcaptain, 2019-10-13:
root@vuduo4k:~# ldd -v /usr/lib/libv3ddriver.so
        linux-vdso.so.1 (0xbed32000)
        libpthread.so.0 => /lib/libpthread.so.0 (0xb6c66000)
        libdl.so.2 => /lib/libdl.so.2 (0xb6c53000)
        libm.so.6 => /lib/libm.so.6 (0xb6bee000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0xb6bc0000)
        libc.so.6 => /lib/libc.so.6 (0xb6a9e000)
        /lib/ld-linux-armhf.so.3 (0xb6f37000)

        Version information:
        /usr/lib/libv3ddriver.so:
                ld-linux-armhf.so.3 (GLIBC_2.4) => /lib/ld-linux-armhf.so.3
                libpthread.so.0 (GLIBC_2.4) => /lib/libpthread.so.0
                libm.so.6 (GLIBC_2.4) => /lib/libm.so.6
                libgcc_s.so.1 (GCC_3.3) => /lib/libgcc_s.so.1
                libgcc_s.so.1 (GCC_3.0) => /lib/libgcc_s.so.1
                libgcc_s.so.1 (GCC_3.5) => /lib/libgcc_s.so.1
                libc.so.6 (GLIBC_2.17) => /lib/libc.so.6
                libc.so.6 (GLIBC_2.4) => /lib/libc.so.6
        /lib/libpthread.so.0:
                ld-linux-armhf.so.3 (GLIBC_2.4) => /lib/ld-linux-armhf.so.3
                ld-linux-armhf.so.3 (GLIBC_PRIVATE) => /lib/ld-linux-armhf.so.3
                libc.so.6 (GLIBC_2.7) => /lib/libc.so.6
                libc.so.6 (GLIBC_PRIVATE) => /lib/libc.so.6
                libc.so.6 (GLIBC_2.4) => /lib/libc.so.6
        /lib/libdl.so.2:
                ld-linux-armhf.so.3 (GLIBC_PRIVATE) => /lib/ld-linux-armhf.so.3
                libc.so.6 (GLIBC_PRIVATE) => /lib/libc.so.6
                libc.so.6 (GLIBC_2.4) => /lib/libc.so.6
        /lib/libm.so.6:
                libc.so.6 (GLIBC_PRIVATE) => /lib/libc.so.6
                libc.so.6 (GLIBC_2.4) => /lib/libc.so.6
        /lib/libgcc_s.so.1:
                libc.so.6 (GLIBC_2.4) => /lib/libc.so.6
        /lib/libc.so.6:
                ld-linux-armhf.so.3 (GLIBC_2.4) => /lib/ld-linux-armhf.so.3
                ld-linux-armhf.so.3 (GLIBC_PRIVATE) => /lib/ld-linux-armhf.so.3

nickersk, 2019-10-13:
as we cant do nothing more about this (vu dont care) i think we could complete remove uianimation ( here too https://github.com/oe-alliance/oe-alliance-core/commit/89502824584545156ccde560eb6ce00ba30507e1 )

atvcaptain, 2019-10-13:
i must test first a new image, possible vu proxy driver use the lib when yes , vu dead or vu give us a update

atvcaptain, 2019-10-14:
no crash after rebuild the image

but tuner not working good and only black screen no live service works any more
looks my box lost protection

nickersk, 2019-10-14:
so we need new driver ...

atvcaptain, 2019-10-14:
box cant read partions with protection any more here

oot@vuduo4k:~# bp3flash.py
RPMB ioctl failed: Invalid argument
Traceback (most recent call last):
  File "/usr/bin/bp3flash.py", line 389, in <module>
    bp3Main(newBp3Bin)
  File "/usr/bin/bp3flash.py", line 306, in bp3Main
    rpmb_checkok = rpmb.checkMagicNumber()
  File "/usr/bin/bp3flash.py", line 193, in checkMagicNumber
    magic_number, = struct.unpack('i',data[0:4])
struct.error: unpack requires a string argument of length 4

atvcaptain, 2019-10-14:
hm firmware files ok in bp30 and bp31
Number  Start (sector)    End (sector)  Size       Code  Name
   1            1408            1535   64.0 KiB    8300  nvram
   2            1536            1663   64.0 KiB    8300  macadr
   3            1664            1791   64.0 KiB    8300  devtree
   4            1792            1919   64.0 KiB    8300  bp30
   5            1920            2047   64.0 KiB    8300  bp31
   6            2048           34815   16.0 MiB    8300  kernel
   7           34816           67583   16.0 MiB    8300  initrd
   8           67584           71679   2.0 MiB     8300  splash
   9           71680         7632895   3.6 GiB     8300  rootfs

root@vuduo4k:~# ls /tmp/bp30
bp3.bin     

root@vuduo4k:~# ls /tmp/bp31
bp3.bin     

but no video

nickersk, 2019-10-14:
so it cant read magic number ... or reads something unusual ..

atvcaptain, 2019-10-14:
i have send today a mail to vuplus main developer

betacentauri, 2019-10-14:
Cap did you check python script? Maybe some python module or normal binary is used  to read the data. And this module needs other parameters in new version or binary is missing.

atvcaptain, 2019-10-14:
problem is here

the /dev/mmcblk0rpmb  we get a
 read error: Input/output error

so no data can read and check failed

thawtes, 2020-01-18:
is there any improvement about this issue ?

nickersk, 2020-01-18:
afaik no new info.

thawtes, 2020-01-18:
> afaik no new info.

thank you

atvcaptain, 2021-12-24:
vu support close, we get no new updates, feature now removed

closed 2021-12-24 by atvcaptain