oe-alliance/oe-alliance-core #546: vmc Plugin crash with openatv 6.4

atvcaptain, 2019-10-13:
21:07:14.7184 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-13 21:07:14+0200 [-] [ActionMap] Keymap 'vmcscreenactions' -> Action = 'ok'.
21:07:14.7397 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-13 21:07:14+0200 [-] Traceback (most recent call last):
21:07:14.7403 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-13 21:07:14+0200 [-]   File "/usr/lib/enigma2/python/Components/ActionMap.py", line 57, in action
21:07:14.7409 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-13 21:07:14+0200 [-]   File "/usr/lib/enigma2/python/Plugins/Extensions/VMC/VMC_Screens_VMCConfig.py", line 1548, in keySelect
21:07:14.7427 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-13 21:07:14+0200 [-]     self['baselist'].getcurrentselection()[0][1]()
21:07:14.7433 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-13 21:07:14+0200 [-]   File "/usr/lib/enigma2/python/Plugins/Extensions/VMC/VMC_Screens_VMCConfig.py", line 275, in createPathManager
21:07:14.7440 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-13 21:07:14+0200 [-]     cur = VMC_DBFunctions.getPathes()
21:07:14.7444 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-13 21:07:14+0200 [-]   File "/usr/lib/enigma2/python/Plugins/Extensions/VMC/VMC_DBFunctions.py", line 880, in getPathes
21:07:14.7453 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-13 21:07:14+0200 [-]     pathes = json.loads(pathes)
21:07:14.7458 { E } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-13 21:07:14+0200 [-] ValueError: Expected object or value

nickersk, 2019-10-14:
looks like problem getting pathes from json ...

atvcaptain, 2020-01-04:
fixed with image 05.01.2020

closed 2020-01-04 by atvcaptain