oe-alliance/oe-alliance-core #547: linux-ceryon_4.0.1  compile issue (4.4)

atvcaptain, 2019-10-15:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "mipsel-oe-linux"
MACHINE              = "7220s"
DISTRO               = "openatv"
DISTRO_VERSION       = "6.4"

/home/oe1/atv64/build-enviroment/builds/openatv/release/7220s/tmp/work-shared/7220s/kernel-source/include/linux/log2.h:22:1: error: ignoring attribute 'noreturn' because it conflicts with attribute 'const' [-Werror=attributes]
   22 | int ____ilog2_NaN(void);
      | ^~~

  /home/oe1/atv64/build-enviroment/meta-oe-alliance/meta-brands/meta-ceryon/recipes-linux/linux-ceryon_4.0.1.bb:do_compile

nickersk, 2019-10-15:
which machine is that?

atvcaptain, 2019-10-15:
read post 1

MACHINE = "7220s"

nickersk, 2019-10-15:
https://github.com/oe-alliance/oe-alliance-core/commit/ebd3c3b42c73a44026f6c8f107d52eac9b407d11

closed 2019-10-15 by atvcaptain