oe-alliance/oe-alliance-core #548: dm 500hd and dm800hd image to big

atvcaptain, 2019-10-15:
Log data follows:
| DEBUG: Executing python function set_image_size
| DEBUG: 238076.800000 = 183136 * 1.300000
| DEBUG: 238076.800000 = max(238076.800000, 65536)[238076.800000] + 0
| DEBUG: 238077.000000 = int(238076.800000)
| DEBUG: 238077 = aligned(238077)
| DEBUG: returning 238077
| DEBUG: Python function set_image_size finished
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['virtual:native:/home/oe1/atv64/build-enviroment/openembedded-core/meta/recipes-core/update-rc.d/update-rc.d_0.8.bb:do_populate_sysroot', '/home/oe1/atv64/build-enviroment/openembedded-core/meta/recipes-core/glibc/ldconfig-native_2.12.1.bb:do_populate_sysroot', 'virtual:native:/home/oe1/atv64/build-enviroment/openembedded-core/meta/recipes-devtools/opkg-utils/opkg-utils_0.4.1.bb:do_populate_sysroot', 'virtual:native:/home/oe1/atv64/build-enviroment/openembedded-core/meta/recipes-devtools/makedevs/makedevs_1.0.1.bb:do_populate_sysroot', '/home/oe1/atv64/build-enviroment/openembedded-core/meta/recipes-devtools/qemu/qemuwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/home/oe1/atv64/build-enviroment/openembedded-core/meta/recipes-devtools/opkg/opkg_0.4.1.bb:do_populate_sysroot', 'virtual:native:/home/oe1/atv64/build-enviroment/openembedded-core/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', '/home/oe1/atv64/build-enviroment/meta-oe-alliance/meta-brands/meta-dream/recipes-bsp/dreambox-buildimage-native_1.5.bb:do_populate_sysroot', '/home/oe1/atv64/build-enviroment/openembedded-core/meta/recipes-core/glibc/cross-localedef-native_2.30.bb:do_populate_sysroot', 'virtual:native:/home/oe1/atv64/build-enviroment/openembedded-core/meta/recipes-devtools/mtd/mtd-utils_git.bb:do_populate_sysroot', '/home/oe1/atv64/build-enviroment/openembedded-core/meta/recipes-kernel/kmod/depmodwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/home/oe1/atv64/build-enviroment/openembedded-core/meta/recipes-extended/pigz/pigz_2.4.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: ['dreambox-buildimage-native', 'mtd-utils-native', 'acl-native']
| NOTE: Skipping as already exists in sysroot: ['update-rc.d-native', 'ldconfig-native', 'opkg-utils-native', 'makedevs-native', 'qemuwrapper-cross', 'opkg-native', 'pseudo-native', 'cross-localedef-native', 'depmodwrapper-cross', 'pigz-native', 'xz-native', 'kmod-native', 'qemu-native', 'shadow-native', 'quilt-native', 'gdk-pixbuf-native', 'glib-2.0-native', 'gtk+3-native', 'autoconf-native', 'libarchive-native', 'gnu-config-native', 'pkgconfig-native', 'libtool-native', 'automake-native', 'openssl10-native', 'debianutils-native', 'lzo-native', 'util-linux-native', 'zlib-native', 'e2fsprogs-native', 'gettext-minimal-native', 'gtk-doc-native', 'python3-native', 'ninja-native', 'libpng-native', 'shared-mime-info-native', 'gobject-introspection-native', 'libjpeg-turbo-native', 'meson-native', 'libffi-native', 'libpcre-native', 'gettext-native', 'libxi-native', 'libxcomposite-native', 'libxfixes-native', 'pango-native', 'libxrender-native', 'libx11-native', 'fontconfig-native', 'at-spi2-atk-native', 'libxext-native', 'libxcursor-native', 'libxrandr-native', 'libxdamage-native', 'atk-native', 'cairo-native', 'texinfo-dummy-native', 'm4-native', 'bzip2-native', 'upx-native', 'ncurses-native', 'libcap-ng-native', 'attr-native', 'gdbm-native', 'sqlite3-native', 'readline-native', 'libtirpc-native', 'libnsl2-native', 're2c-native', 'intltool-native', 'libxml2-native', 'flex-native', 'cmake-native', 'python3-setuptools-native', 'xorgproto-native', 'util-macros-native', 'freetype-native', 'fribidi-native', 'libxft-native', 'harfbuzz-native', 'xtrans-native', 'libxcb-native', 'gperf-native', 'expat-native', 'at-spi2-core-native', 'dbus-native', 'libsm-native', 'pixman-native', 'perl-native', 'ucl-native', 'libxml-parser-perl-native', 'curl-native', 'unzip-native', 'icu-native', 'libxau-native', 'libxdmcp-native', 'libpthread-stubs-native', 'xcb-proto-native', 'libxtst-native', 'libice-native', 'db-native']
| DEBUG: sed -e 's:^[^/]*/:/home/oe1/atv64/build-enviroment/builds/openatv/release/dm500hd/tmp/work/dm500hd-oe-linux/openatv-image/6.4-0/recipe-sysroot-native/:g' /home/oe1/atv64/build-enviroment/builds/openatv/release/dm500hd/tmp/sysroots-components/x86_64/mtd-utils-native/fixmepath | xargs sed -i -e 's:FIXMESTAGINGDIRTARGET:/home/oe1/atv64/build-enviroment/builds/openatv/release/dm500hd/tmp/work/dm500hd-oe-linux/openatv-image/6.4-0/recipe-sysroot:g; s:FIXMESTAGINGDIRHOST:/home/oe1/atv64/build-enviroment/builds/openatv/release/dm500hd/tmp/work/dm500hd-oe-linux/openatv-image/6.4-0/recipe-sysroot-native:g' -e 's:FIXME_PSEUDO_SYSROOT:/home/oe1/atv64/build-enviroment/builds/openatv/release/dm500hd/tmp/sysroots-components/x86_64/pseudo-native:g' -e 's:FIXME_HOSTTOOLS_DIR:/home/oe1/atv64/build-enviroment/builds/openatv/release/dm500hd/tmp/hosttools:g' -e 's:FIXME_PKGDATA_DIR:/home/oe1/atv64/build-enviroment/builds/openatv/release/dm500hd/tmp/pkgdata/dm500hd:g' -e 's:FIXME_PSEUDO_LOCALSTATEDIR:/home/oe1/atv64/build-enviroment/builds/openatv/release/dm500hd/tmp/work/dm500hd-oe-linux/openatv-image/6.4-0/pseudo/:g' -e 's:FIXME_LOGFIFO:/home/oe1/atv64/build-enviroment/builds/openatv/release/dm500hd/tmp/work/dm500hd-oe-linux/openatv-image/6.4-0/temp/fifo.18141:g'
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_image_jffs2nfi

after add samba 4.x

nickersk, 2019-10-16:
question is what to do next. i see 3 options:
1. readd s3
2. remove s4 from smallflash boxes
3. find something similar to samba, but minimalist

Schimmelreiter, 2019-10-16:
4. Drop those junk boxes completely.

kueken, 2019-12-13:
Ok, you don't want to build from the old Samba 3 anymore, but maybe building from 'Source3' would be an alternative for all boxes with small memory?
https://github.com/samba-team/samba/tree/v4-11-stable/source3
In principle, a slimmer Samba can still be created from it.
But then in the current environment.

atvcaptain, 2021-04-05:
support now stopped

closed 2021-04-05 by atvcaptain