oe-alliance/oe-alliance-core #549: linux-odin_3.18.24 build issue (4.4)

atvcaptain, 2019-10-15:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "mipsel-oe-linux"
MACHINE              = "odinm7"
DISTRO               = "openatv"
DISTRO_VERSION       = "6.4"

/home/oe1/atv64/build-enviroment/builds/openatv/release/odinm7/tmp/work-shared/odinm7/kernel-source/include/linux/log2.h:22:1: error: ignoring attribute 'noreturn' because it conflicts with attribute 'const' [-Werror=attributes]
   22 | int ____ilog2_NaN(void);
      | ^~~

atvcaptain, 2019-10-15:
https://github.com/scylladb/dpdk/issues/5

atvcaptain, 2019-10-15:
ah i see you have add some fixes but not for all system

the log2 allready fixed

nickersk, 2019-10-15:
i will fix them all but busy right now ... evening or tomorrow task ...

closed 2019-10-15 by atvcaptain