oe-alliance/oe-alliance-core #551: linux-vuplus_3.9.6 build issue (4.4)

atvcaptain, 2019-10-15:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "mipsel-oe-linux"
MACHINE              = "vuduo"
DISTRO               = "openatv"
DISTRO_VERSION       = "6.4"

/home/oe1/atv64/build-enviroment/builds/openatv/release/vuduo/tmp/work-shared/vuduo/kernel-source/include/linux/init.h:306:7: error: 'cleanup_module' specifies less restrictive attribute than its target 'wil6210_driver_exit': 'cold' [-Werror=missing-attributes]
  306 |  void cleanup_module(void) __attribute__((alias(#exitfn)));
      |       ^~~~~~~~~~~~~~

atvcaptain, 2019-10-15:
looks is the same issue with a fix

https://github.com/zfsonlinux/zfs/issues/8329

nickersk, 2019-10-15:
one issue i fix imediately i fall to other, this is no life :-)

atvcaptain, 2019-10-15:
i found more  :) , i check now all boxes in 5-7 days i have all build

nickersk, 2019-10-15:
https://github.com/oe-alliance/oe-alliance-core/commit/428d206e0ef1e70a02fcff78a286d767af15951c

closed 2019-10-15 by atvcaptain