oe-alliance/oe-alliance-core #554: python-beautifulsoup4 not work (OEA 4.4)

thawtes, 2019-10-18:
hi,
python-beautifulsoup4 not working properly on 4.4 branch
`/usr/lib/python2.7/site-packages/bs4/element.py:16: UserWarning: The soupsieve package is not installed. CSS selectors cannot be used.`

but,
good work on 4.3 branch

thx

thawtes, 2019-10-18:
i think it needs soupsieve package

soupsieve source
https://pypi.org/project/soupsieve/
https://github.com/facelessuser/soupsieve

thawtes, 2019-10-18:
yes,
does not work properly because there is missing **soupsieve** !

thawtes, 2019-10-29:
`/usr/lib/python2.7/site-packages/bs4/element.py:16: UserWarning: The soupsieve package is not installed. CSS selectors cannot be used.`

**python-soupsieve package installed, but the problem continues ?**

Is there something missing?

thawtes, 2019-10-29:
missing : **backports.functools_lru_cache-1.5** from **soupsieve**

nickersk, 2019-10-29:
this comes from soupsieve right ? not from bs4 ..

thawtes, 2019-10-30:
> 
> 
> shuld be fixed [51b86e7](https://github.com/oe-alliance/oe-alliance-core/commit/51b86e74a3bd641d900ad673c68ea5049b1da149) (hopefully)

:( not fixed !

python-backports-functools-lru-cache : **not installable !**  (because not in feed)
this package depend **python-setuptools-scm-native** : **not installable !**  (because not in feed)

nickersk, 2019-10-31:
depends are used only at build time .. can you try manual install from your local feed? if it will work we can add it public feed...thx

thawtes, 2019-10-31:
> 
> 
> depends are used only at build time .. can you try manual install from your local feed? if it will work we can add it to public feed...thx i tried it locally and it works. opkg install python-soupsieve: Installing python-backports-functools-lru-cache (1.5) on root. Installing python-soupsieve (1.9.4) on root. Configuring python-backports-functools-lru-cache.
> Configuring python-soupsieve. did you run opkg update? and maybe it just got not built in public feed yet...but i cant check that..reinstalling bs4 should do the job too...

unfortunately it didn't work on me. (opkg updated)
Still not on feed list
pls check latest build list
http://feeds2.mynonpublic.com/6.4/vuduo2/mips32el/Packages.gz
you will see that packages are not listed

nickersk, 2019-10-31:
@thawtes good remark man, i forgot to readd rdepends to soupsieve after revert...will fix soon. thx again

thawtes, 2019-10-31:
> 
> 
> @thawtes good remark man, i forgot to readd rdepends to soupsieve after revert...will fix soon. thx again. finally [7d249c6](https://github.com/oe-alliance/oe-alliance-core/commit/7d249c64d3551336883ad2af74879335b6f26d63) :-)

dear @nickersk , i thank you ...

thawtes, 2019-11-01:
issue fixed !
thanks

closed 2019-11-01 by thawtes