oe-alliance/oe-alliance-core #555: gst1.0 - meson

nickersk, 2019-10-19:
@atvcaptain https://github.com/oe-alliance/oe-alliance-core/commit/640bc390f6437618cce49688a6a8a0054477dff2 can you test it on 4.4 outside offficial images pls? i created it and tested for nextp3 branches (mipsel, arm, aarch64, hisi). if all ok can be cherrypicked to 4.4. thx

nickersk, 2019-10-19:
p3 meta-OE is broken currently (guys are playing with licenses) will wait for them to fix it.

atvcaptain, 2019-10-19:
yes i can test it  offline, need some time we test a new box at the moment dev server are busy :)

68foxboris, 2019-11-13:
Tell me please:
Log data follows:
| DEBUG: Executing shell function do_configure
| NOTE: Executing meson -Ddoc=disabled -Dexamples=disabled -Dqt5=disabled -Ddecklink=enabled -Ddvb=enabled -Dfbdev=enabled -Dnetsim=enabled -Dshm=enabled -Ddts=enabled -Dmpegdemux=enabled -Dandroidmedia=disabled -Dapplemedia=disabled -Dbs2b=disabled -Dchromaprint=disabled -Ddirectsound=disabled -Dfdkaac=disabled -Dgme=disabled -Dgsm=disabled -Diqa=disabled -Dladspa=disabled -Dlv2=disabled -Dmpeg2enc=disabled -Dmplex=disabled -Dmsdk=disabled -Dmusepack=disabled -Dnvcodec=disabled -Dofa=disabled -Dopenexr=disabled -Dopenni2=disabled -Dopensles=disabled -Dsoundtouch=disabled -Dspandsp=disabled -Dteletext=disabled -Dtinyalsa=disabled -Dwasapi=disabled -Dwebrtcdsp=disabled -Dwildmidi=disabled -Dwinks=disabled -Dwinscreencap=disabled -Dzbar=disabled -Dassrender=disabled -Dbluez=disabled -Dbz2=enabled -Dcurl=enabled -Dcurl-ssh2=disabled -Ddash=enabled -Ddc1394=disabled -Ddebug=false -Ddirectfb=disabled -Ddtls=enabled -Dfaac=enabled -Dfaad=enabled -Dflite=disabled -Dfluidsynth=disabled -Dgl=disabled -Dhls=enabled -Dhls-crypto=openssl -Dkms=disabled -Dlibde265=enabled -Dlibmms=enabled -Dmodplug=disabled -Dmsdk=disabled -Dneon=enabled -Dopenal=disabled -Dopencv=disabled -Dopenh264=disabled -Dopenjpeg=disabled -Dopenmpt=disabled -Dopus=enabled -Dorc=enabled --disable-qt -Dresindvd=disabled -Drsvg=enabled -Drtmp=enabled -Dsbc=enabled -Dsmoothstreaming=enabled -Dsndfile=enabled -Dsrtp=disabled -Dtinyalsa=disabled -Dttml=enabled -Duvch264=enabled -Dvoaacenc=disabled -Dvoamrwbenc=disabled -Dwayland=disabled -Dwebp=enabled -Dwebrtc=disabled -Dwebrtcdsp=disabled -Dx265=enabled...
| usage: meson [-h]
|              {setup,configure,dist,install,introspect,init,test,wrap,subprojects,help,rewrite}
|              ...
| meson: error: unrecognized arguments: --disable-qt
| ERROR: meson failed
| WARNING: /home/foxbob/hd51/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/gstreamer1.0-plugins-bad/1.17.0.1+gitAUTOINC+b55f380b20-r0/temp/run.do_configure.28635:1 exit 1 from 'exit 1'
| ERROR: Execution of '/home/foxbob/hd51/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/gstreamer1.0-plugins-bad/1.17.0.1+gitAUTOINC+b55f380b20-r0/temp/run.do_configure.28635' failed with exit code 1:
| usage: meson [-h]
|              {setup,configure,dist,install,introspect,init,test,wrap,subprojects,help,rewrite}
|              ...
| meson: error: unrecognized arguments: --disable-qt

nickersk, 2019-11-13:
update meta-qt5 and try again

68foxboris, 2019-11-24:
Good evening. How can I fix this error?
[
[image]
[image]
[image]
](url)

68foxboris, 2019-11-24:
Good evening. How can I fix this error?
[image]
[image]
[image]

nickersk, 2019-11-24:
its ok, ffmpeg 4.2.1 wants to install its own libav files, but they are provided by gstreamer1.0-libav already - and are diferrent version (4.1.4). only way to fix this would be to remove gst-libav and then install ffmpeg, or use 4.2.1 ffmpeg header in gst-libav...

68foxboris, 2019-11-24:
And if you do so during assembly,so that there is one version ffmpeg.Or so wrong?
DEPENDS = "gstreamer1.0 gstreamer1.0-plugins-base ffmpeg"

nickersk, 2019-11-24:
well gstreamer meson uses ffpmeg 4.1.4 . if u will try to depend on ffmpeg 4.2.1 it would use 4.2.1 which is not recomended. maybe i can try to to move gstlibav libs under gstreamer1.0 libdir (which as far i can remember was in previous non mesonbuild version so). thanks for reporting this, i will research it tomorrow ...

nickersk, 2019-11-25:
till now i found, that mesonbuild is causing libav libs to be installed by every build of ffmpeg meson port... so possibly the correct solution would be to use depend on external ffmpeg as i have not discovered how to tell meson to not install these libs.

68foxboris, 2019-11-27:
It turns out that you need to leave it as it is.

nickersk, 2019-11-27:
btw. using ffmpeg 4.2.1 image grows by ~5Mb

68foxboris, 2019-12-08:
WARNING: gstreamer1.0-libav-1.17.0.1+gitAUTOINC+77f63c2457-r0 do_package: libpostproc-4.2.1 was registered as shlib provider for libpostproc.so.55, changing it to gstreamer1.0-libav-1.17.0.1+git2254+77f63c2 because it was built later
WARNING: gstreamer1.0-libav-1.17.0.1+gitAUTOINC+77f63c2457-r0 do_package: libswresample-4.2.1 was registered as shlib provider for libswresample.so.3, changing it to gstreamer1.0-libav-1.17.0.1+git2254+77f63c2 because it was built later
WARNING: gstreamer1.0-libav-1.17.0.1+gitAUTOINC+77f63c2457-r0 do_package: libavutil-4.2.1 was registered as shlib provider for libavutil.so.56, changing it to gstreamer1.0-libav-1.17.0.1+git2254+77f63c2 because it was built later
WARNING: gstreamer1.0-libav-1.17.0.1+gitAUTOINC+77f63c2457-r0 do_package: libavformat-4.2.1 was registered as shlib provider for libavformat.so.58, changing it to gstreamer1.0-libav-1.17.0.1+git2254+77f63c2 because it was built later
WARNING: gstreamer1.0-libav-1.17.0.1+gitAUTOINC+77f63c2457-r0 do_package: libavcodec-4.2.1 was registered as shlib provider for libavcodec.so.58, changing it to gstreamer1.0-libav-1.17.0.1+git2254+77f63c2 because it was built later
WARNING: gstreamer1.0-libav-1.17.0.1+gitAUTOINC+77f63c2457-r0 do_package: libswscale-4.2.1 was registered as shlib provider for libswscale.so.5, changing it to gstreamer1.0-libav-1.17.0.1+git2254+77f63c2 because it was built later
WARNING: gstreamer1.0-libav-1.17.0.1+gitAUTOINC+77f63c2457-r0 do_package: libavdevice-4.2.1 was registered as shlib provider for libavdevice.so.58, changing it to gstreamer1.0-libav-1.17.0.1+git2254+77f63c2 because it was built later
WARNING: gstreamer1.0-libav-1.17.0.1+gitAUTOINC+77f63c2457-r0 do_package: libavresample-4.2.1 was registered as shlib provider for libavresample.so.4, changing it to gstreamer1.0-libav-1.17.0.1+git2254+77f63c2 because it was built later
WARNING: gstreamer1.0-libav-1.17.0.1+gitAUTOINC+77f63c2457-r0 do_package: libavfilter-4.2.1 was registered as shlib provider for libavfilter.so.7, changing it to gstreamer1.0-libav-1.17.0.1+git2254+77f63c2 because it was built later

nickersk, 2019-12-08:
so gst replaced ffmpeg libs

68foxboris, 2019-12-08:
Ok.

68foxboris, 2019-12-10:
After this - https://github.com/oe-alliance/oe-alliance core/commit/17e2498a5f0e26bf6d6df506b7e0e5c83718c3ff not work exteplayer3.
root@et1x000:~# exteplayer3
exteplayer3: error while loading shared libraries: libswscale.so.5: cannot open shared object file: No such file or directory
And the image got a little blurry.May be.

nickersk, 2019-12-10:
try to reflash image first

68foxboris, 2019-12-11:
Clean image flash.Install e2iplayer.Not work hlsdl and exteplayer3.
2019-12-11 18:46:36+0200 [-] STS: /usr/bin/exteplayer3: error while loading shared libraries: libswscale.so.5: cannot open shared object file: No such file or directory
2019-12-11 18:46:36+0200 [-] 
2019-12-11 18:46:36+0200 [-] КОД ОШИБКИ: 127)
'Screens.MessageBox.MessageBox'>(Видимо, видео-плеер "Внешний eplayer3" не работает должным образом.
2019-12-11 18:46:36+0200 [-] 
2019-12-11 18:46:36+0200 [-] STS: /usr/bin/exteplayer3: error while loading shared libraries: libswscale.so.5: cannot open shared object file: No such file or directory
2019-12-11 18:46:36+0200 [-] 
2019-12-11 18:46:36+0200 [-] КОД ОШИБКИ: 127)

root@et1x000:~# ffmpeg
ffmpeg: error while loading shared libraries: libavdevice.so.58: cannot open shared object file: No such file or directory
root@et1x000:~# exteplayer3
exteplayer3: error while loading shared libraries: libswscale.so.5: cannot open shared object file: No such file or directory
root@et1x000:~# hlsdl
hlsdl: error while loading shared libraries: libavcodec.so.58: cannot open shared object file: No such file or directory
root@et1x000:~# 
[image]

nickersk, 2019-12-11:
have you noticed exteplayer3 project is closed? https://github.com/e2iplayer/exteplayer3

68foxboris, 2019-12-11:
I know this. I have version 68 and it works well.

nickersk, 2019-12-11:
please try now, if all ok, i will close this.

68foxboris, 2019-12-11:
Ok.

68foxboris, 2019-12-12:
Clean image flash.Install serviceapp-exteplayer3 not install and not work.

nickersk, 2019-12-12:
nothing else changed, how is it possible it worked before?

68foxboris, 2019-12-12:
This is at the previous commit.

nickersk, 2019-12-12:
which ? please paste link here...

68foxboris, 2019-12-12:
This
https://github.com/oe-alliance/oe-alliance-core/commit/17e2498a5f0e26bf6d6df506b7e0e5c83718c3ff

nickersk, 2019-12-12:
ahm ok, thx for feedback.

68foxboris, 2019-12-12:
Yes.

closed 2019-12-12 by nickersk

68foxboris, 2020-01-03:
Only I can’t watch DVD video through MovieSelection?
libdvdread: Couldn't find device name.
libdvdnav: Can't read name block. Probably not a DVD-ROM device.
libdvdnav: vm: dvd_read_name failed
libdvdnav: DVD disk reports itself with Region mask 0x00c00000. Regions: 1 2 3 4 5 6
[eServiceDVD] DDVD_SHOWOSD_TITLESTRING: 
[eServiceDVD] loadCuesheet trying /media/hdd/bluray/Masha i Medved'_22_DVD-5//dvd.cuts
libdreamdvd: soft ac3 decoding is available, liba52.so.0 loaded !
enigma2: ../git/mpegaudio_enc.c:347: ddvd_mpa_compute_scale_factors: Assertion `index >=0 && index <= 63' failed.
Backtrace: