oe-alliance/oe-alliance-core #556: [Plugin] SubsSupport - Crash in some areas (openATV 6.4)

thawtes, 2019-10-19:
hello, 
Subssupport plugin Crashes when, open searchparam suggestion lists !

device: vu+duo2
image: openatv 6.4

log lines below, thx ..
`
20:20:05.6850 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-] KEY: 398 1 KEY_RED RED
20:20:05.6878 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-] [ActionMap] Keymap 'ColorActions' -> Action = 'red'.
20:20:05.6989 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-] [ActionMap] Keymap(s) 'OkCancelActions, ColorActions, DirectionActions' -> Undefined action(s) 'rightUp, downRepeated, upRepeated, downUp, rightRepeated, leftRepeated, leftUp, upUp'.
20:20:05.7014 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-] [ActionMap] Keymap(s) 'OkCancelActions, ColorActions, DirectionActions' -> Undefined action(s) 'rightUp, downRepeated, upRepeated, downUp, rightRepeated, leftRepeated, leftUp, upUp'.
20:20:05.7048 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-] [SKIN] processing screen SubsSearchParamsMenu:
20:20:05.7952 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-] [SKIN] processing screen NumericalTextInputHelpDialog:
20:20:05.8189 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-] [SCREENNAME]  NumericalTextInputHelpDialog
20:20:05.8279 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-] [SKIN] Parsing embedded skin <embedded-in-'SuggestionsListScreen'>
20:20:05.8311 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-] [SKIN] processing screen <embedded-in-'SuggestionsListScreen'>:
20:20:05.8414 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-] [SCREENNAME]  SuggestionsListScreen
20:20:05.8505 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-] [SKIN] Parsing embedded skin <embedded-in-'HistoryListScreen'>
20:20:05.8538 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-] [SKIN] processing screen <embedded-in-'HistoryListScreen'>:
20:20:05.8645 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-] [SCREENNAME]  HistoryListScreen
20:20:05.8677 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-] Traceback (most recent call last):
20:20:05.8694 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-]   File "/usr/lib/enigma2/python/Plugins/Extensions/SubsSupport/e2_utils.py", line 76, in selectionChanged
20:20:05.8713 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-]     return ConfigList.selectionChanged(self)
20:20:05.8732 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-]   File "/usr/lib/enigma2/python/Components/ConfigList.py", line 76, in selectionChanged
20:20:05.8751 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-]   File "/usr/lib/enigma2/python/Plugins/Extensions/SubsSupport/subtitles.py", line 3453, in onSelect
20:20:05.8818 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-]     self.getSuggestions()
20:20:05.8840 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-]   File "/usr/lib/enigma2/python/Plugins/Extensions/SubsSupport/subtitles.py", line 3560, in getSuggestions
20:20:05.8911 { E } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-]     self.__suggestions = self.suggestionsClass().getSuggestions(self.value, self.propagateSuggestions, self.gotSuggestionsError)
20:20:05.8930 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-]   File "/usr/lib/enigma2/python/Plugins/Extensions/SubsSupport/subtitles.py", line 3272, in getSuggestions
20:20:05.8995 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-]     d = self._getSuggestions(queryString)
20:20:05.9012 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-]   File "/usr/lib/enigma2/python/Plugins/Extensions/SubsSupport/subtitles.py", line 3296, in _getSuggestions
20:20:05.9078 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-]     return client.getPage(query, timeout=6)
20:20:05.9112 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-]   File "/usr/lib/python2.7/site-packages/twisted/python/deprecate.py", line 301, in deprecatedFunction
20:20:05.9137 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-]   File "/usr/lib/python2.7/site-packages/twisted/web/client.py", line 795, in getPage
20:20:05.9157 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-]   File "/usr/lib/python2.7/site-packages/twisted/web/client.py", line 742, in _makeGetterFactory
20:20:05.9178 {   } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-]   File "/usr/lib/python2.7/site-packages/twisted/web/_newclient.py", line 648, in _ensureValidURI
20:20:05.9200 { E } /usr/lib/python2.7/site-packages/twisted/python/util.py:815 untilConcludes 2019-10-19 20:20:05+0300 [-] ValueError: Invalid URI 'http://www.opensubtitles.org/libs/suggest.php?format=json2&SubLanguageID=null&MovieName=Rhapsody'
20:20:05.9204 [ E ] python/python.cpp:210 call [ePyObject] (PyObject_CallObject(<bound method MyConfigList.selectionChanged of <Plugins.Extensions.SubsSupport.e2_utils.MyConfigList object at 0x703eb5d0>>,()) failed)
`

thawtes, 2019-10-19:
NOTE:
i back to **python-twisted-web_18.4.0-r1** (manual install)
result : no crash

thawtes, 2019-10-19:
> 
> 
> NOTE:
> i back to **python-twisted-web_18.4.0-r1** (manual install)
> result : no crash

of course this is not the solution

nickersk, 2019-10-20:
just a suggestion: wouldnt it be better to ask there: https://github.com/mx3L/subssupport ?

thawtes, 2019-10-20:
> just a suggestion: wouldnt it be better to ask there:

of course! I asked there too.
but many plugins have similar problems.
I wanted to ask if a backward compatibility patch can be done.

thanks @nickersk

nickersk, 2019-10-20:
well imho problem can be how json.loads is directly used by return

thawtes, 2019-10-20:
> 
> 
> well imho problem can be how json.loads is directly used by return

i'll try to find a way to fix it
thank you for your interest

arn354, 2019-10-20:
https://github.com/mx3L/subssupport/blob/f4a0c6f13748252ca1547bd47bfa46dda589fa00/plugin/searchsubs.py#L29

?

def recieve():
    jsonreceive = json.loads(sys.stdin.read(int(sys.stdin.read(7))))
    return jsonreceive

thawtes, 2019-10-20:
> 
> 
> https://github.com/mx3L/subssupport/blob/f4a0c6f13748252ca1547bd47bfa46dda589fa00/plugin/searchsubs.py#L29
> 
> ?
> 
> def recieve():
> jsonreceive = json.loads(sys.stdin.read(int(sys.stdin.read(7))))
> return jsonreceive

code needs to be edited this way?

closed 2019-10-20 by thawtes