oe-alliance/oe-alliance-core #562: Error compile qtbase to dm920 ?!!

fairbird, 2019-11-04:
> fig.tests/unix/libpng/libpng.cpp: In function 'int main(int, char**)':
| > /home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/git/config.tests/unix/libpng/libpng.cpp:48:17: warning: variable 'png_ptr' set but not used [-Wunused-but-set-variable]
| >    48 |     png_structp png_ptr;
| >       |                 ^~~~~~~
| > arm-oe-linux-gnueabi-g++  -mfpu=neon-vfpv4 -mfloat-abi=hard -mcpu=cortex-a15 --sysroot=/home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed --sysroot=/home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot --sysroot=/home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot -Wl,-O1 -fuse-ld=gold -o libpng libpng.o   -lpng16 -lz
|  => source accepted.
| test config.gui.libraries.libpng succeeded
| Checking for XLib...
| + cd /home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/build/config.tests/x11/xlib && PKG_CONFIG_SYSROOT_DIR=/home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot PKG_CONFIG_LIBDIR=/home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot/usr/lib/pkgconfig /home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot-native/usr/bin/qt5/qmake -qtconf /home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/build/bin/qt.conf "CONFIG -= qt debug_and_release app_bundle lib_bundle" "CONFIG += shared use_gold_linker console" "QMAKE_CFLAGS += --sysroot=/home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot" "QMAKE_CXXFLAGS += --sysroot=/home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot" "QMAKE_LFLAGS += --sysroot=/home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot" /home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/git/config.tests/x11/xlib
| > /home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/git/mkspecs/features/toolchain.prf:51: Extra characters after test expression.
| > /home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/git/mkspecs/features/toolchain.prf:52: Extra characters after test expression.
| + cd /home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/build/config.tests/x11/xlib && MAKEFLAGS= make
| > arm-oe-linux-gnueabi-g++  -mfpu=neon-vfpv4 -mfloat-abi=hard -mcpu=cortex-a15 --sysroot=/home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot -c -pipe  -Os -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0=/usr/src/debug/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0                      -fdebug-prefix-map=/home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0=/usr/src/debug/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0                      -fdebug-prefix-map=/home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden --sysroot=/home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot --sysroot=/home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/recipe-sysroot -O2 -Wall -W -fPIC  -I/home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/git/config.tests/x11/xlib -I. -I/home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/git/mkspecs/linux-oe-g++ -o xlib.o /home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/git/config.tests/x11/xlib/xlib.cpp
| > /home/raed/build_image/OpenATV/build-enviroment/builds/openatv/release/dm920/tmp/work/dm920-oe-linux-gnueabi/qtbase/5.8.0+gitAUTOINC+49dc9aa409-r0/git/config.tests/x11/xlib/xlib.cpp:40:10: fatal error: X11/Xlib.h: No such file or directory
| >    40 | #include <X11/Xlib.h>
| >       |          ^~~~~~~~~~~~
| > compilation terminated.
| > make: *** [Makefile:176: xlib.o] Error 1
| test config.gui.tests.xlib FAILED
| Done running configuration tests.
| 
| Configure summary:
| 
| Building on:  arm
| Building for: arm
| Configuration: use_gold_linker cross_compile compile_examples enable_new_dtags largefile neon silent shared release c++11 c++14 c++1z concurrent dbus mremap no-qml-debug reduce_exports stl
| Build options:
|   Mode ................................... release
|   Building shared libraries .............. yes
|   Using C++ standard ..................... C++1z
|   Using gold linker ...................... yes
|   Using new DTAGS ........................ yes
|   Using precompiled headers .............. no
|   Using LTCG ............................. no
|   Target compiler supports:
|     NEON ................................. yes
|   Build parts ............................ libs tools
| Qt modules and options:
|   Qt Concurrent .......................... yes
|   Qt D-Bus ............................... yes
|   Qt D-Bus directly linked to libdbus .... yes
|   Qt Gui ................................. yes
|   Qt Widgets ............................. yes
| Support enabled for:
|   Accessibility .......................... no
|   Using pkg-config ....................... yes
|   QML debugging .......................... no
|   udev ................................... yes
|   Using system zlib ...................... yes
| Qt Core:
|   DoubleConversion ....................... yes
|     Using system DoubleConversion ........ no
|   GLib ................................... no
|   iconv .................................. no
|   ICU .................................... no
|   Logging backends:
|     journald ............................. no
|     syslog ............................... no
|   Using system PCRE ...................... no
| Qt Network:
|   getaddrinfo() .......................... yes
|   getifaddrs() ........................... yes
|   IPv6 ifname ............................ yes
|   libproxy ............................... no
|   OpenSSL ................................ yes
|     Qt directly linked to OpenSSL ........ no
|   SCTP ................................... no
|   Use system proxies ..................... yes
| Qt Sql:
|   DB2 (IBM) .............................. no
[... 32 lines cut ...]
|   mtdev .................................. no
[... cut ...] g -no-libproxy -make libs -no-linuxfb -no-mtdev -openssl -qt-pcre -no-qml-debug -release -no-sm -no-sql-db2 -no-sql-ibase -no-sql-mysql -no-sql-oci -no-sql-odbc -no-sql-psql -no-sql-sqlite -no-sql-sqlite2 -no-sql-tds -nomake tests -make tools -no-tslib -libudev -widgets -no-xcb -no-xinput2 -no-xkb -no-xkbcommon -no-xkbcommon-evdev -no-xrender -system-zlib'
| 
ERROR: Task (/home/raed/build_image/OpenATV/build-enviroment/meta-qt5/recipes-qt/qt5/qtbase_git.bb:do_configure) failed with exit code '1'

atvcaptain, 2019-11-04:
dont build qt for dm9x00 not needed

fairbird, 2019-11-04:
Yes I know dm9x00 not need ..
But my plane is try to make enigma2-plugin-extensions-browser plugin to work on open source images.
Just I need to build qtbase + qtwebkit and then I can make the plugin work on open source images.

So please help me to fix that error ..
Thank you

atvcaptain, 2019-11-05:
we have hbbtv (browser) for long time working with dm9x0 but not to same time with e2
we hast ask dm info for init the fb but no answer

you must add this to dm920.conf

PACKAGECONFIG_GL_pn-qtbase = "gles2 eglfs linuxfb"
PACKAGECONFIG_FB_pn-qtbase = " "

DISTRO_FEATURES_remove = "x11 wayland"

fairbird, 2019-11-05:
Already I have add that option and get same error 

PACKAGECONFIG_GL_pn-qtbase = "gles2 eglfs linuxfb"
PACKAGECONFIG_FB_pn-qtbase = " "

But I will add also as you advice and test build again ..

DISTRO_FEATURES_remove = "x11 wayland"

fairbird, 2019-11-05:
Still get error ...

> | WARNING: Accessibility disabled. This configuration of Qt is unsupported.
| 
| WARNING: Feature xkbcommon-system is insignificant in this configuration, ignoring related command line option(s).
| 
| ERROR: Feature 'opengles2' was enabled, but the pre-condition 'config.win32 || (!config.watchos && !features.opengl-desktop && libs.opengl_es2)' failed.
| 
| ERROR: Feature 'eglfs' was enabled, but the pre-condition 'features.egl' failed.
| 
| ERROR: The OpenGL functionality tests failed!
| You might need to modify the include and library search paths by editing QMAKE_INCDIR_OPENGL[_ES2],
| QMAKE_LIBDIR_OPENGL[_ES2] and QMAKE_LIBS_OPENGL[_ES2] in the mkspec for your platform.

nickersk, 2019-11-05:
this only enables som packages for x11 distro feature. 5.8.0 is too old. try recent meta-qt5 and it will compile fine (without existing patches in meta-oe-alliance).

fairbird, 2019-11-05:
Hi nickersk, How did you do ?!!
Yes I have already tried with latest version of meta-qt5 and also try version of 5.10 what opendreambox use it ... but I have got a lot of errors..

I try with these core ..
https://github.com/oe-alliance/oe-alliance-core/tree/4.4
and
https://github.com/Hains/openvision-dm920-oe-core
Both using same version openssl 1.1 and gcc9.2

Ok I will try again and then I will post here the error code what I get it ..

Thank you

nickersk, 2019-11-05:
try meta-qt5 master branch with nextp3-ssl111 OEA branch and with changes @atvcaptain wrote sooner.

fairbird, 2019-11-05:
Ok I will do some test now and then I will post the result ...

fairbird, 2019-11-05:
But with which option I need to add it to dm920.conf ?!
Is it same atvcaptain wrote it ?!

PACKAGECONFIG_GL_pn-qtbase = "gles2 eglfs linuxfb"
PACKAGECONFIG_FB_pn-qtbase = " "

DISTRO_FEATURES_remove = "x11 wayland"

fairbird, 2019-11-05:
Result (recent meta-qt5 master + oe-all 4.4 (without existing patches in meta-oe-alliance))

with config in dm920.conf Got it error

PACKAGECONFIG_GL_pn-qtbase = "gles2 eglfs linuxfb"
PACKAGECONFIG_FB_pn-qtbase = " "
DISTRO_FEATURES_remove = "x11 wayland"

qtbase compiled without error ..

> qtbase_5.13.0+git0+fc9ae22c88-r0_cortexa15hf-neon-vfpv4.ipk
qtbase-dbg_5.13.0+git0+fc9ae22c88-r0_cortexa15hf-neon-vfpv4.ipk
qtbase-dev_5.13.0+git0+fc9ae22c88-r0_cortexa15hf-neon-vfpv4.ipk
qtbase-doc_5.13.0+git0+fc9ae22c88-r0_cortexa15hf-neon-vfpv4.ipk
qtbase-mkspecs_5.13.0+git0+fc9ae22c88-r0_cortexa15hf-neon-vfpv4.ipk
qtbase-plugins_5.13.0+git0+fc9ae22c88-r0_cortexa15hf-neon-vfpv4.ipk
qtbase-qmlplugins_5.13.0+git0+fc9ae22c88-r0_cortexa15hf-neon-vfpv4.ipk
qtbase-staticdev_5.13.0+git0+fc9ae22c88-r0_cortexa15hf-neon-vfpv4.ipk
qtbase-tools_5.13.0+git0+fc9ae22c88-r0_cortexa15hf-neon-vfpv4.ipk

Now I will try to compile other packages that enigma2-plugin-extensions-browser need and then seeI what happens ...

Thank you

atvcaptain, 2019-11-05:
better fix kodi for dm, qtbrowser 99% of web not working :)

closed 2019-11-05 by atvcaptain

fairbird, 2019-11-06:
@nickersk 
Please my friend need your help ..
I need also to compile it with this code ..
https://github.com/Hains/openvision-dm920-oe-core
I have test meta-qt (master, Zeus and Zeus-next) but I have get this error

So what can you advice to me ?!!
Thank you

with dm920.conf options

PACKAGECONFIG_GL_pn-qtbase = "gles2 eglfs linuxfb"
PACKAGECONFIG_FB_pn-qtbase = " "
DISTRO_FEATURES_remove = "x11 wayland"

> | Note: -hostbindir is not a subdirectory of -hostprefix.
| Note: -hostlibdir is not a subdirectory of -hostprefix.
| Note: -hostdatadir is not a subdirectory of -hostprefix.
| 
| WARNING: Feature xcb-xinput is insignificant in this configuration, ignoring related command line option(s).
| 
| WARNING: Feature xkb is insignificant in this configuration, ignoring related command line option(s).
| 
| ERROR: Feature 'opengles2' was enabled, but the pre-condition '(config.win32 && !features.opengl-dynamic) || (!config.watchos && !features.opengl-desktop && libs.opengl_es2)' failed.
| 
| ERROR: Feature 'eglfs' was enabled, but the pre-condition '!config.android && !config.darwin && !config.win32 && !config.wasm && features.egl' failed.
| 
| ERROR: The OpenGL functionality tests failed!
| You might need to modify the include and library search paths by editing QMAKE_INCDIR_OPENGL[_ES2],
| QMAKE_LIBDIR_OPENGL[_ES2] and QMAKE_LIBS_OPENGL[_ES2] in the mkspec for your platform.
| WARNING: /home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/qtbase/5.13.0+gitAUTOINC+fc9ae22c88-r0/temp/run.do_configure.15847:1 exit 3 from '/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/qtbase/5.13.0+gitAUTOINC+fc9ae22c88-r0/git/configure -v -opensource -confirm-license -sysroot /home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/qtbase/5.13.0+gitAUTOINC+fc9ae22c88-r0/recipe-sysroot -prefix /usr -bindir /usr/bin -libdir /usr/lib -datadir /usr/share -sysconfdir /etc -docdir /usr/share/doc -headerdir /usr/include -archdatadir /usr/lib -libexecdir /usr/lib/libexec -plugindir /usr/lib/plugins -qmldir /usr/lib/qml -translationdir /usr/share/translations -testsdir /usr/share/tests -examplesdir /usr/share/examples -hostprefix /home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/qtbase/5.13.0+gitAUTOINC+fc9ae22c88-r0/recipe-sysroot-native -hostlibdir /home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/qtbase/5.13.0+gitAUTOINC+fc9ae22c88-r0/recipe-sysroot/usr/lib -hostbindir /usr/bin -external-hostbindir /home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/qtbase/5.13.0+gitAUTOINC+fc9ae22c88-r0/recipe-sysroot-native/usr/bin -hostdatadir /home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/qtbase/5.13.0+gitAUTOINC+fc9ae22c88-r0/recipe-sysroot/usr/lib -platform linux-oe-g++ -xplatform linux-oe-g++ -no-use-gold-linker -shared -silent -no-pch -no-rpath -pkg-config -accessibility -no-cups -dbus -no-directfb -eglfs -evdev -nomake examples -no-fontconfig -system-freetype -no-gbm -no-feature-getentropy -no-gif -opengl es2 -no-glib -no-gtk -qt-harfbuzz -no-ico -no-iconv -no-icu -no-journald -system-libjpeg -no-kms -no-libinput -system-libpng -no-libproxy -make libs -linuxfb -trace no -no-mtdev -openssl -qt-pcre -release -no-sm -no-sql-db2 -no-sql-ibase -no-sql-mysql -no-sql-oci -no-sql-odbc -no-sql-psql -no-sql-sqlite -no-sql-sqlite2 -no-sql-tds -shared -make tests -make tools -no-tslib -libudev -widgets -no-xcb -no-xcb-xinput -no-xkb -no-xkbcommon -no-xkbcommon -system-zlib'
| 
ERROR: Task (/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/meta-qt5/recipes-qt/qt5/qtbase_git.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1588 tasks of which 1586 didn't need to be rerun and 1 failed.

without dm920.conf options

PACKAGECONFIG_GL_pn-qtbase = "gles2 eglfs linuxfb"
PACKAGECONFIG_FB_pn-qtbase = " "
DISTRO_FEATURES_remove = "x11 wayland"

> | compiling /home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/qtbase/5.13.0+gitAUTOINC+fc9ae22c88-r0/git/tests/auto/network/socket/qlocalsocket/tst_qlocalsocket.cpp
| linking tst_qabstractnetworkcache
| linking ../tst_qudpsocket
| linking ../tst_qnetworkreply
| linking ../tst_qlocalsocket
| make[5]: Leaving directory '/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/qtbase/5.13.0+gitAUTOINC+fc9ae22c88-r0/build/tests/auto/network/access/qabstractnetworkcache'
| make[6]: Leaving directory '/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/qtbase/5.13.0+gitAUTOINC+fc9ae22c88-r0/build/tests/auto/network/socket/qlocalsocket/test'
| make[6]: Leaving directory '/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/qtbase/5.13.0+gitAUTOINC+fc9ae22c88-r0/build/tests/auto/network/socket/qudpsocket/test'
[... cut ...]
| make[1]: Leaving directory '/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/qtbase/5.13.0+gitAUTOINC+fc9ae22c88-r0/build/tests'
| make: *** [Makefile:101: sub-tests-make_first] Error 2
| WARNING: /home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/qtbase/5.13.0+gitAUTOINC+fc9ae22c88-r0/temp/run.do_compile.5368:1 exit 1 from 'exit 1'
| 
ERROR: Task (/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/meta-qt5/recipes-qt/qt5/qtbase_git.bb:do_compile) failed with exit code '1'

nickersk, 2019-11-06:
hmm, i have no idea at all i cant see which file is causing error here. try qt5 master-next branch

fairbird, 2019-11-06:
I think there is some packages missing or config option ...
So way I have got this error if I use ...

PACKAGECONFIG_GL_pn-qtbase = "gles2 eglfs linuxfb"
PACKAGECONFIG_FB_pn-qtbase = " "
DISTRO_FEATURES_remove = "x11 wayland"

> | WARNING: Feature xcb-xinput is insignificant in this configuration, ignoring related command line option(s).
|
| WARNING: Feature xkb is insignificant in this configuration, ignoring related command line option(s).
|
| ERROR: Feature 'opengles2' was enabled, but the pre-condition '(config.win32 && !features.opengl-dynamic) || (!config.watchos && !features.opengl-desktop && libs.opengl_es2)' failed.
|
| ERROR: Feature 'eglfs' was enabled, but the pre-condition '!config.android && !config.darwin && !config.win32 && !config.wasm && features.egl' failed.
|
| ERROR: The OpenGL functionality tests failed!
| You might need to modify the include and library search paths by editing QMAKE_INCDIR_OPENGL[_ES2],
| QMAKE_LIBDIR_OPENGL[_ES2] and QMAKE_LIBS_OPENGL[_ES2] in the mkspec for your platform.

Ok I will try also with master-next ...

fairbird, 2019-11-06:
Yes ... with master-next branch compiled just fine ..
but without 

PACKAGECONFIG_GL_pn-qtbase = "gles2 eglfs linuxfb"
PACKAGECONFIG_FB_pn-qtbase = " "
DISTRO_FEATURES_remove = "x11 wayland"

If I use that options I got error 

> | WARNING: Feature xcb-xinput is insignificant in this configuration, ignoring related command line option(s).
|
| WARNING: Feature xkb is insignificant in this configuration, ignoring related command line option(s).
|
| ERROR: Feature 'opengles2' was enabled, but the pre-condition '(config.win32 && !features.opengl-dynamic) || (!config.watchos && !features.opengl-desktop && libs.opengl_es2)' failed.
|
| ERROR: Feature 'eglfs' was enabled, but the pre-condition '!config.android && !config.darwin && !config.win32 && !config.wasm && features.egl' failed.
|
| ERROR: The OpenGL functionality tests failed!
| You might need to modify the include and library search paths by editing QMAKE_INCDIR_OPENGL[_ES2],
| QMAKE_LIBDIR_OPENGL[_ES2] and QMAKE_LIBS_OPENGL[_ES2] in the mkspec for your platform.

I don't know way ...

Thank you

nickersk, 2019-11-06:
looks like problem in meta dm layer then ..

fairbird, 2019-11-06:
Here is layers

https://github.com/Hains/openvision-dm920-oe-core/blob/933305fdb9344f9fd5c3a247f28a0f122bba9c8f/meta-dream/conf/layer.conf

https://github.com/Hains/openvision-dm920-oe-core/blob/933305fdb9344f9fd5c3a247f28a0f122bba9c8f/meta-dream/conf/machine/dm920.conf

https://github.com/Hains/openvision-dm920-oe-core/blob/933305fdb9344f9fd5c3a247f28a0f122bba9c8f/meta-dream/conf/machine/include/dreambox-arm.inc

nickersk, 2019-11-06:
sorry, i really got no more ideas :(

fairbird, 2019-11-06:
No problem my friend ..
Thank you for your time ..

fairbird, 2019-11-07:
@nickersk 
 Do you have any idea about this error from Compile (qtwebkit)
 (found suitable version "1.1.33", minimum required is "1.1.7")

> | -- Found LibXslt: /home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/dm920-oe-linux-gnueabi/qtwebkit/5.13.2+gitAUTOINC+ab1bd15209-r0/recipe-sysroot/usr/lib/libxslt.so (found suitable version "1.1.33", minimum required is "1.1.7")
| -- Could NOT find WebP (missing: WEBP_INCLUDE_DIRS WEBP_LIBRARIES)
| CMake Error at Source/cmake/OptionsQt.cmake:596 (message):
|   Qt with OpenGL support is required for ENABLE_OPENGL
| Call Stack (most recent call first):
|   Source/cmake/WebKitCommon.cmake:53 (include)
|   CMakeLists.txt:137 (include)
| 
| 
| -- Configuring incomplete, errors occurred!
| See also "/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/dm920-oe-linux-gnueabi/qtwebkit/5.13.2+gitAUTOINC+ab1bd15209-r0/build/CMakeFiles/CMakeOutput.log".
| See also "/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/dm920-oe-linux-gnueabi/qtwebkit/5.13.2+gitAUTOINC+ab1bd15209-r0/build/CMakeFiles/CMakeError.log".
| WARNING: /home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/dm920-oe-linux-gnueabi/qtwebkit/5.13.2+gitAUTOINC+ab1bd15209-r0/temp/run.do_configure.29204:1 exit 1 from 'cmake -G 'Ninja' -DCMAKE_MAKE_PROGRAM=ninja $oecmake_sitefile /home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/dm920-oe-linux-gnueabi/qtwebkit/5.13.2+gitAUTOINC+ab1bd15209-r0/git -DCMAKE_INSTALL_PREFIX:PATH=/usr -DCMAKE_INSTALL_BINDIR:PATH=bin -DCMAKE_INSTALL_SBINDIR:PATH=sbin -DCMAKE_INSTALL_LIBEXECDIR:PATH=libexec -DCMAKE_INSTALL_SYSCONFDIR:PATH=/etc -DCMAKE_INSTALL_SHAREDSTATEDIR:PATH=../com -DCMAKE_INSTALL_LOCALSTATEDIR:PATH=/var -DCMAKE_INSTALL_LIBDIR:PATH=lib -DCMAKE_INSTALL_INCLUDEDIR:PATH=include -DCMAKE_INSTALL_DATAROOTDIR:PATH=share -DCMAKE_INSTALL_SO_NO_EXE=0 -DCMAKE_TOOLCHAIN_FILE=/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/dm920-oe-linux-gnueabi/qtwebkit/5.13.2+gitAUTOINC+ab1bd15209-r0/toolchain.cmake -DCMAKE_NO_SYSTEM_FROM_IMPORTED=1 -DOE_QMAKE_PATH_PREFIX=/usr -DOE_QMAKE_PATH_HEADERS=/usr/include -DOE_QMAKE_PATH_LIBS=/usr/lib -DOE_QMAKE_PATH_ARCHDATA=/usr/lib -DOE_QMAKE_PATH_DATA=/usr/share -DOE_QMAKE_PATH_BINS=/usr/bin -DOE_QMAKE_PATH_LIBEXECS=/usr/lib/libexec -DOE_QMAKE_PATH_PLUGINS=/usr/lib/plugins -DOE_QMAKE_PATH_QML=/usr/lib/qml -DOE_QMAKE_PATH_TRANSLATIONS=/usr/share/translations -DOE_QMAKE_PATH_DOCS=/usr/share/doc -DOE_QMAKE_PATH_SETTINGS=/etc -DOE_QMAKE_PATH_EXAMPLES=/usr/share/examples -DOE_QMAKE_PATH_TESTS=/usr/share/tests -DOE_QMAKE_PATH_HOST_PREFIX=/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/dm920-oe-linux-gnueabi/qtwebkit/5.13.2+gitAUTOINC+ab1bd15209-r0/recipe-sysroot-native -DOE_QMAKE_PATH_HOST_BINS=/usr/bin -DOE_QMAKE_PATH_HOST_DATA=/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/dm920-oe-linux-gnueabi/qtwebkit/5.13.2+gitAUTOINC+ab1bd15209-r0/recipe-sysroot/usr/lib -DOE_QMAKE_PATH_HOST_LIBS=/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/dm920-oe-linux-gnueabi/qtwebkit/5.13.2+gitAUTOINC+ab1bd15209-r0/recipe-sysroot/usr/lib -DOE_QMAKE_PATH_EXTERNAL_HOST_BINS=/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/dm920-oe-linux-gnueabi/qtwebkit/5.13.2+gitAUTOINC+ab1bd15209-r0/recipe-sysroot-native/usr/bin -DOE_QMAKE_PATH_QT_HEADERS=/usr/include -DOE_QMAKE_PATH_QT_ARCHDATA=/usr/lib -DOE_QMAKE_PATH_QT_DATA=/usr/share -DOE_QMAKE_PATH_QT_BINS=/usr/bin -DOE_QMAKE_PATH_QT_TRANSLATIONS=/usr/share/translations -DOE_QMAKE_PATH_QT_DOCS=/usr/share/doc -DOE_QMAKE_PATH_QT_SETTINGS=/etc -DOE_QMAKE_PATH_QT_EXAMPLES=/usr/share/examples -DOE_QMAKE_PATH_QT_TESTS=/usr/share/tests -DPORT=Qt -DCROSS_COMPILE=ON -DECM_MKSPECS_INSTALL_DIR=/usr/lib/mkspecs/modules -DQML_INSTALL_DIR=/usr/lib/qml -DENABLE_TEST_SUPPORT=OFF -DUSE_GSTREAMER=OFF -DUSE_LIBHYPHEN=OFF -DENABLE_GEOLOCATION=OFF -DUSE_QT_MULTIMEDIA=OFF -DENABLE_DEVICE_ORIENTATION=OFF -DENABLE_QT_WEBCHANNEL=OFF -DENABLE_X11_TARGET=OFF -Wno-dev'
| 
ERROR: Task (/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/meta-qt5/recipes-qt/qt5/qtwebkit_git.bb:do_configure) failed with exit code '1'

nickersk, 2019-11-07:
error is: CMake Error at Source/cmake/OptionsQt.cmake:596 (message):
| Qt with OpenGL support is required for ENABLE_OPENGL so you have to build qt with opengl support or disable it somehow

fairbird, 2019-11-07:
If I enable it I have got error on config

> | WARNING: Feature xcb-xinput is insignificant in this configuration, ignoring related command line option(s).
|
| WARNING: Feature xkb is insignificant in this configuration, ignoring related command line option(s).
|
| ERROR: Feature 'opengles2' was enabled, but the pre-condition '(config.win32 && !features.opengl-dynamic) || (!config.watchos && !features.opengl-desktop && libs.opengl_es2)' failed.
|
| ERROR: Feature 'eglfs' was enabled, but the pre-condition '!config.android && !config.darwin && !config.win32 && !config.wasm && features.egl' failed.
|
| ERROR: The OpenGL functionality tests failed!
| You might need to modify the include and library search paths by editing QMAKE_INCDIR_OPENGL[_ES2],
| QMAKE_LIBDIR_OPENGL[_ES2] and QMAKE_LIBS_OPENGL[_ES2] in the mkspec for your platform.

fairbird, 2019-11-07:
I have tried also like this option

PACKAGECONFIG_GL_pn-qtbase = "${@bb.utils.contains('DISTRO_FEATURES', 'opengl', 'gles2 eglfs', '', d)}"
PACKAGECONFIG_DISTRO_pn-qtbase = "gif ico icu sql-sqlite"
PACKAGECONFIG_FB_pn-qtbase = "linuxfb"
PACKAGECONFIG_pn-qtdeclarative = ""
PACKAGECONFIG_pn-qtwebkit = ""
DISTRO_FEATURES_remove = "x11 wayland"

But I have got error

nickersk, 2019-11-07:
DISTRO_FEATURES_append = " opengl"

fairbird, 2019-11-07:
Thank you 
I will try latter ...

fairbird, 2019-11-07:
@atvcaptain 
@nickersk 
What you think about this feature remove is it ok or not for dm902 ?!
https://github.com/Hains/openvision-dm920-oe-core/blob/master/meta-dream/conf/machine/include/dreambox-arm.inc#L49

nickersk, 2019-11-07:
depends on real hardware, which i dont have access to

fairbird, 2019-11-07:
with 

DISTRO_FEATURES_append = " opengl"

> | meson.build:427:4: ERROR: Problem encountered: building dri drivers require at least one windowing system or classic osmesa
| 
| A full log can be found at /home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/mesa-gl/2_19.1.6-r0/build/meson-logs/meson-log.txt
| ERROR: meson failed
| WARNING: /home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/mesa-gl/2_19.1.6-r0/temp/run.do_configure.5943:1 exit 1 from 'exit 1'
| ERROR: Execution of '/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/mesa-gl/2_19.1.6-r0/temp/run.do_configure.5943' failed with exit code 1:
| The Meson build system
| Version: 0.52.0
| Source dir: /home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/mesa-gl/2_19.1.6-r0/mesa-19.1.6
| Build dir: /home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/mesa-gl/2_19.1.6-r0/build
| Build type: cross build
| Program python found: YES (/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/hosttools/python)
| Project name: mesa
| Project version: 19.1.6
| C compiler for the build machine: gcc (gcc 8.3.0 "gcc (Ubuntu 8.3.0-6ubuntu1) 8.3.0")
| C linker for the build machine: GNU ld.bfd 2.32
| C++ compiler for the build machine: g++ (gcc 8.3.0 "g++ (Ubuntu 8.3.0-6ubuntu1) 8.3.0")
| C++ linker for the build machine: GNU ld.bfd 2.32
| C compiler for the host machine: arm-oe-linux-gnueabi-gcc -mfpu=neon-vfpv4 -mfloat-abi=hard -mcpu=cortex-a15 --sysroot=/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/mesa-gl/2_19.1.6-r0/recipe-sysroot (gcc 9.2.0 "arm-oe-linux-gnueabi-gcc (GCC) 9.2.0")
| C linker for the host machine: GNU ld.bfd 2.32.0.20190204
| C++ compiler for the host machine: arm-oe-linux-gnueabi-g++ -mfpu=neon-vfpv4 -mfloat-abi=hard -mcpu=cortex-a15 --sysroot=/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/mesa-gl/2_19.1.6-r0/recipe-sysroot (gcc 9.2.0 "arm-oe-linux-gnueabi-g++ (GCC) 9.2.0")
| C++ linker for the host machine: GNU ld.bfd 2.32.0.20190204
| Build machine cpu family: x86_64
| Build machine cpu: x86_64
| Host machine cpu family: arm
| Host machine cpu: arm
| Target machine cpu family: arm
| Target machine cpu: arm
| 
| meson.build:427:4: ERROR: Problem encountered: building dri drivers require at least one windowing system or classic osmesa
| 
| A full log can be found at /home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/mesa-gl/2_19.1.6-r0/build/meson-logs/meson-log.txt
| WARNING: /home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/build/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/mesa-gl/2_19.1.6-r0/temp/run.do_configure.5943:1 exit 1 from 'exit 1'
| 
ERROR: Task (/home/raed/build_image/920-BlackHole/openvision-dm920-oe-core/openembedded-core/meta/recipes-graphics/mesa/mesa-gl_19.1.6.bb:do_configure) failed with exit code '1'

fairbird, 2019-12-10:
> we have hbbtv (browser) for long time working with dm9x0 but not to same time with e2
> we have ask dm info for init the fb but no answer
> 
> you must add this to dm920.conf
> 
> PACKAGECONFIG_GL_pn-qtbase = "gles2 eglfs linuxfb"
> PACKAGECONFIG_FB_pn-qtbase = " "
> 
> DISTRO_FEATURES_remove = "x11 wayland"
@atvcaptain 
Hi my friend
Where can i find the hbbtv (browser) to dm9x0 ?!
I want to test it ?!
Thank you