oe-alliance/oe-alliance-core #564: dags7356 oe-a 4.4 not booting

atvcaptain, 2019-11-09:
dags7356  i check optimussos3plus

VFS: Mounted root (ubifs filesystem) on device 0:12.
devtmpfs: mounted
Freeing unused kernel memory: 208k freed
INIT: version 2.88 booting
INIT: Entering runlevel: 3
INIT: Id "gui" respawning too fast: disabled for 5 minutes
INIT: no more processes left in this runlevel

and box not start

i check oe-a 4.4 kernel with rootfs from 4.3 works
i check kernel from 4.3 and rootfs from 4.4 not booting
i flash image 4.3 and online flash to 4.4 not booting

i see kernel is running and working insert usb i see log output

i remove drivers and flash the rootfs without drivers not working
problem i see no more infos after

INIT: Entering runlevel: 3
INIT: Id "gui" respawning too fast: disabled for 5 minutes
INIT: no more processes left in this runlevel

all output stops

nickersk, 2019-11-10:
is there any 'gui' in /etc/init.d in rootfs?

atvcaptain, 2019-11-10:
INIT: version 2.88 booting
INIT: Reading inittab
INIT: Checking for children to kill
INIT: Checking for children to start
INIT: chld_handler: marked 58 as zombie
INIT: got SIGCHLD
INIT: Child died, PID= 58
INIT: Checking for children to start
INIT: SYSINIT -> BOOT
INIT: init_main: waiting..
INIT: Checking for children to start
INIT: BOOT -> NORMAL
INIT: Entering runlevel: 3
INIT: init_main: waiting..
INIT: Checking for children to start
INIT: Started id l3 (pid 61)
INIT: init_main: waiting..
INIT: chld_handler: marked 61 as zombie
INIT: got SIGCHLD
INIT: Child died, PID= 61
INIT: Checking for children to start
INIT: Started id gui (pid 62)
INIT: init_main: waiting..
INIT: chld_handler: marked 62 as zombie
INIT: got SIGCHLD
INIT: Child died, PID= 62
INIT: Checking for children to start
INIT: Started id gui (pid 63)
INIT: init_main: waiting..
INIT: chld_handler: marked 63 as zombie
INIT: got SIGCHLD
INIT: Child died, PID= 63
INIT: Checking for children to start
INIT: Started id gui (pid 64)
INIT: init_main: waiting..
INIT: chld_handler: marked 64 as zombie
INIT: got SIGCHLD
INIT: Child died, PID= 64
INIT: Checking for children to start
INIT: Started id gui (pid 65)
INIT: init_main: waiting..
INIT: chld_handler: marked 65 as zombie
INIT: got SIGCHLD
INIT: Child died, PID= 65
INIT: Checking for children to start
INIT: Started id gui (pid 66)
INIT: init_main: waiting..
INIT: chld_handler: marked 66 as zombie
INIT: got SIGCHLD
INIT: Child died, PID= 66
INIT: Checking for children to start
INIT: Started id gui (pid 67)
INIT: init_main: waiting..
INIT: chld_handler: marked 67 as zombie
INIT: got SIGCHLD
INIT: Child died, PID= 67
INIT: Checking for children to start
INIT: Started id gui (pid 68)
INIT: init_main: waiting..
INIT: chld_handler: marked 68 as zombie
INIT: got SIGCHLD
INIT: Child died, PID= 68
INIT: Checking for children to start
INIT: Started id gui (pid 69)
INIT: init_main: waiting..
INIT: chld_handler: marked 69 as zombie
INIT: got SIGCHLD
INIT: Child died, PID= 69
INIT: Checking for children to start
INIT: Started id gui (pid 70)
INIT: init_main: waiting..
INIT: chld_handler: marked 70 as zombie
INIT: got SIGCHLD
INIT: Child died, PID= 70

nickersk, 2019-11-10:
more /etc/inittab ... gui:3:respawn:/usr/bin/enigma2.sh so problem is with e2 sh ..

nickersk, 2019-11-10:
try to boot without it in inittab and then run manually e2 sh ...

BliNeR-KeY, 2019-12-11:
Maybe CFE limit .. @atvcaptain if you try to install same image (4.4) in multboot i sure works.

Schimmelreiter, 2019-12-11:
> Maybe CFE limit .. @atvcaptain if you try to install same image (4.4) in multboot i sure works.

The CFE bug is exactly what should have been fixed now ...

AbuBaniaz, 2019-12-11:
If you flash a 6.3 image by USB, it does not boot.

Flash using "Flash Online" it boots. I'll see if I can get serial log later

a4tech, 2019-12-11:
meta-ini boxes same issue. Looks like rootfs size is to big.

AbuBaniaz, 2019-12-19:
6.3 flashed by USB does not boot
6.3 flashed by ofgwrite boots

6.4 by USB or ofgwrite does not boot

Serial  logs attached

6.3 OFGWrite
[6.3 with ofgwrite boot OK.txt](https://github.com/oe-alliance/oe-alliance-core/files/3986006/6.3.with.ofgwrite.boot.OK.txt)

6.4 OFGWrite
[6.4 with ofgwrite not boot.txt](https://github.com/oe-alliance/oe-alliance-core/files/3986007/6.4.with.ofgwrite.not.boot.txt)

Vix 5.2 USB
[Flash Vix 5.2 045 by USB Boot OK.txt](https://github.com/oe-alliance/oe-alliance-core/files/3986008/Flash.Vix.5.2.045.by.USB.Boot.OK.txt)

6.3 USB
[Flash 6.3 by USB not boot.txt](https://github.com/oe-alliance/oe-alliance-core/files/3986009/Flash.6.3.by.USB.not.boot.txt)

6.4 USB
[Flash 6.4 by USB not boot.txt](https://github.com/oe-alliance/oe-alliance-core/files/3986010/Flash.6.4.by.USB.not.boot.txt)

AbuBaniaz, 2019-12-19:
I tried ATV 6.4 on a TM NanoSE M2, that did not boot either.  Flashed by USB and OFGwrite, both the same. No serial port so could not get log

nickersk, 2019-12-20:
the difference between 6.3 with ofgwrite boot OK.txt and 6.4 with ofgwrite not boot.txt is missing: usb 1-1: new high-speed USB device number 2 using ehci-brcm scsi3 : usb-storage 1-1:1.0 scsi 3:0:0:0: Direct-Access     Generic  Flash Disk       8.07 PQ: 0 ANSI: 4
so it looks like 6.4 dont see USB stick ?

AbuBaniaz, 2019-12-22:
You had posted earlier about missing USB devices, I tried with two USB devices connected to the TM Nano 3T Combo. It still did not boot.

Sorry about delay in response.

AbuBaniaz, 2019-12-30:
I will check tomorrows build unless you can build one for me to test

AbuBaniaz, 2019-12-31:
Flashed using offwrite and working.
Many thanks

AbuBaniaz, 2019-12-31:
Flashed TM Nano Se M2 using ofgwrite, also fine.

nickersk, 2019-12-31:
thx for reporting.

closed 2020-01-04 by atvcaptain