oe-alliance/oe-alliance-core #568: zerotier_1.4.6.bb:do_compile (branch 4.4)

atvcaptain, 2019-11-12:
arm-oe-linux-gnueabi-g++  -mfpu=neon-vfpv4 -mfloat-abi=hard -mcpu=cortex-a15 --sysroot=/home/oe1/atv64arm/build-enviroment/builds/openatv/release/vuultimo4k/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/zerotier/1.4.6+gitAUTOINC+e1e27eb19d-r0/recipe-sysroot  -Os -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/oe1/atv64arm/build-enviroment/builds/openatv/release/vuultimo4k/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/zerotier/1.4.6+gitAUTOINC+e1e27eb19d-r0=/usr/src/debug/zerotier/1.4.6+gitAUTOINC+e1e27eb19d-r0                      -fdebug-prefix-map=/home/oe1/atv64arm/build-enviroment/builds/openatv/release/vuultimo4k/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/zerotier/1.4.6+gitAUTOINC+e1e27eb19d-r0=/usr/src/debug/zerotier/1.4.6+gitAUTOINC+e1e27eb19d-r0                      -fdebug-prefix-map=/home/oe1/atv64arm/build-enviroment/builds/openatv/release/vuultimo4k/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/zerotier/1.4.6+gitAUTOINC+e1e27eb19d-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/oe1/atv64arm/build-enviroment/builds/openatv/release/vuultimo4k/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/zerotier/1.4.6+gitAUTOINC+e1e27eb19d-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden -Wall -Wno-deprecated -std=c++11 -pthread  -DNDEBUG  -DZT_USE_MINIUPNPC -DMINIUPNP_STATICLIB -DMINIUPNPC_SET_SOCKET_TIMEOUT -DMINIUPNPC_GET_SRC_ADDR -D_BSD_SOURCE -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DOS_STRING=\"Linux\" -DMINIUPNPC_VERSION_STRING=\"2.0\" -DUPNP_VERSION_STRING=\"UPnP/1.1\" -DENABLE_STRNATPMPERR -DZT_NO_TYPE_PUNNING -DZT_BUILD_PLATFORM=1 -DZT_BUILD_ARCHITECTURE=3 -DZT_SOFTWARE_UPDATE_DEFAULT="\"disable\"" -D_MT_ALLOCATOR_H -D_POOL_ALLOCATOR_H -D_EXTPTR_ALLOCATOR_H -D_DEBUG_ALLOCATOR_H -DZT_USE_ARM32_NEON_ASM_SALSA2012   -c -o service/OneService.o service/OneService.cpp
service/OneService.cpp:697: error: unterminated #ifdef
  697 | #ifdef ZT_USE_MINIUPNPC

nickersk, 2019-11-12:
will take a look on this

nickersk, 2019-11-12:
https://github.com/oe-alliance/oe-alliance-core/commit/b685fdb4e86de8143b8a858c6a0d701dd541dcbe

atvcaptain, 2019-11-12:
thx i am not at home cant not test self :)

closed 2019-11-12 by atvcaptain

teamblue-e2, 2019-11-12:
works, thx

nickersk, 2019-11-12:
https://github.com/zerotier/ZeroTierOne/commit/b8e70d866a16ef26d782e4a0b074a124c2cb4212