oe-alliance/oe-alliance-core #574: gst1.0 - meson

68foxboris, 2020-01-03:
Only I can not watch DVD video through MovieSelection?
`libdvdnav: Using dvdnav version 6.0.1
[eServiceDVD] DVD_SIZE_CHANGED 0: 720x576 2
[eServiceDVD] DVD_FRAMERATE_CHANGED 0: 29970
[eServiceDVD] DVD_PROGRESSIVE_CHANGED 0: 0
libdvdread: Couldn't find device name.
libdvdnav: Can't read name block. Probably not a DVD-ROM device.
libdvdnav: vm: dvd_read_name failed
libdvdnav: DVD disk reports itself with Region mask 0x00c00000. Regions: 1 2 3 4 5 6
[eServiceDVD] DDVD_SHOWOSD_TITLESTRING: 
[eServiceDVD] loadCuesheet trying /media/hdd/bluray/Masha i Medved'_22_DVD-5//dvd.cuts
.....
enigma2: ../git/mpegaudio_enc.c:347: ddvd_mpa_compute_scale_factors: Assertion `index >=0 && index <= 63' failed.
Backtrace:
`

nickersk, 2020-01-04:
have you tried EMC?

68foxboris, 2020-01-04:
No, but it should reproduce here.

nickersk, 2020-01-04:
you mean non-meson gst version? or some of previous meson ported version?

68foxboris, 2020-01-04:
non-meson

nickersk, 2020-01-04:
well this transition is bit complicated, will need much more time ...

68foxboris, 2020-01-04:
Ok.

nickersk, 2020-03-07:
done, now uses ff 4.2.2 (hisi 3.4.2)

closed 2020-03-07 by nickersk