oe-alliance/oe-alliance-core #578: platforms

Power535, 2020-02-07:
What list platforms mips32el supported in user mode ?

nickersk, 2020-02-07:
i am sorry i dont understand your question. you mean what mips32el platform is? mips32el = mips 32bit little endian

Power535, 2020-02-07:
mipsel for bchip

Power535, 2020-02-07:
mips 32 little endian for bchip

nickersk, 2020-02-07:
you can find SOC models in meta-brands layer under meta-machine/conf/machine/machine.conf

Power535, 2020-02-07:
There is no information
kernel or user mode

nickersk, 2020-02-07:
its not that easy,you have to search for this: https://github.com/oe-alliance/oe-alliance-core/blob/8b6580eb6aca8bc3a50b6b2fb2e2e4908fe03516/meta-brands/meta-airdigital/conf/machine/include/airdigital-arm.inc#L14

Power535, 2020-02-07:
Meant bchip modules in user mode

nickersk, 2020-02-07:
it is part of manufacturers drivers - closed source.

Power535, 2020-02-07:
Meant bchip modules in user mode
I tried to search for a long time, but I fall into kernel mode
I have modules only in user mode i want adding platform for compile

nickersk, 2020-02-07:
maybe you can write recipe for kernel-module-abc and then install it to image IMAGE_INSTALL_append += " \ 

kernel-module-abc\
"

Power535, 2020-02-07:
I am not a programmer
Possible i send for you tomorrow modules ?
To see how to add
I do not know as configure
I have night here now

nickersk, 2020-02-07:
i am really busy these days, maybe you can try it yourself ..look at some other recipes to find the logic ...

Power535, 2020-02-07:
I tried fews year and no results
Will wait any time
Any help from any users

nickersk, 2020-02-07:
ok, so i will let this thread open for some time, maybe someone with more time will take look at this.

atvcaptain, 2020-02-10:
e2 drivers use kernel mode, user mode drivers not support with e2 code

atvcaptain, 2020-02-10:
i read you "I am not a programmer" i think forget your bcm 7231 meta

closed 2020-02-10 by atvcaptain