oe-alliance/oe-alliance-core #579: qtwebkit

kueken, 2020-02-08:
It seems that in some cases the qtwebkit makes construction mistakes because of 'historical' cflags.
'-mieee'
Is it possible to deactivate it, or to remove it immediately?
Will probably no longer be interesting for sh4 ;-)

You can find these lines
git

/source/WTF/WTF.pro
/source/JavaScriptCore/Target.pri
/source/autotools/SetupCompilerFlags.m4

nickersk, 2020-02-08:
which recipe (we have more qt versions)?

kueken, 2020-02-08:
https://github.com/qt/qtwebkit/blob/5.9/Source/WTF/WTF.pro#L272-L275

Just as an example, but that's not the git source from here.
Do not know if the other two git have these 'historical' flags.
However, they lead to possible construction errors even though one does not build for sh4, but for poor / hisi.

nickersk, 2020-02-08:
thx, will take a look @ these

68foxboris, 2020-02-11:
Good evening. Please tell me how to solve this problem during compilation.
`In file included from /home/foxbob/zeus/build/tmp/work/et1x000-oe-linux-gnueabi/qtwebkit/5.14.1+gitAUTOINC+ab1bd15209-r0/recipe-sysroot/usr/include/EGL/egl.h:39,
                 from /home/foxbob/zeus/build/tmp/work/et1x000-oe-linux-gnueabi/qtwebkit/5.14.1+gitAUTOINC+ab1bd15209-r0/git/Source/WebCore/platform/graphics/opengl/Extensions3DOpenGLES.cpp:36:
/home/foxbob/zeus/build/tmp/work/et1x000-oe-linux-gnueabi/qtwebkit/5.14.1+gitAUTOINC+ab1bd15209-r0/recipe-sysroot/usr/include/EGL/eglplatform.h:113:10: fatal error: X11/Xlib.h: No such file or directory
  113 | #include <X11/Xlib.h>
      |          ^~~~~~~~~~~~
compilation terminated.`

nickersk, 2020-02-11:
there is include requesting x11 feature in EGL which afaik is part of drivers ...

68foxboris, 2020-02-11:
I add DISTRO_FEATURES_EXTRA = "opengl",but there is a mistake.

nickersk, 2020-02-11:
search for Xlib.h

68foxboris, 2020-02-11:
foxbob@foxbob-System:~$ find /usr/include/X11 -name Xlib.h
/usr/include/X11/Xlib.h

nickersk, 2020-02-11:
it must be in work tree, not host root

nickersk, 2020-02-11:
u can try to add x11 to or libx11 to depends and try, but imho it is just waste of time

68foxboris, 2020-02-11:
And how to add it or do you need to make a symlink?

nickersk, 2020-02-11:
start with studying openembedded manuals, thats the correct way, like all of us do...

68foxboris, 2020-02-11:
Ok.

nickersk, 2020-02-11:
@68foxboris any progress ?

68foxboris, 2020-02-11:
It doesn’t work with me. And most likely that there is no support for gles.

68foxboris, 2020-02-11:
There is webkit-classic,hbbtv works.And I wanted to add something else.Compilation is not received with kodi 18.

nickersk, 2020-02-24:
@kueken pls see https://github.com/oe-alliance/build-enviroment/commit/27c54e3e202bb902a2614cfd38dcea0c92d78f44

closed 2020-02-24 by nickersk

68foxboris, 2020-03-09:
I managed to collect qtwebkit_5.14.0+git0+ab1bd15209-r0_et1x000.ipk,zeus branch,so wrong,but i just added a folder X11 system in usr/include.After start qtstalker dark screen and throws it into the channel view.