oe-alliance/oe-alliance-core #580: Openspa 7.4 build error

ahmedmoselhi, 2020-02-10:
openspabuilder@ip-172-31-22-227:~/openspa/build-enviroment$ MACHINE=amiko8900 DISTRO=openspa DISTRO_TYPE=release make image
check internet connection: Online
Loading cache: 100% |#########################################| Time: 0:00:01Loaded 4856 entries from dependency cache.
Parsing recipes: 100% |#######################################| Time: 0:02:28Parsing of 4148 .bb files complete (3774 cached, 374 parsed). 5230 targets, 630 skipped, 21 masked, 0 errors.
WARNING: No bb files matched BBFILE_PATTERN_local-layer '^/home/openspabuilder/openspa/build-enviroment/meta-local/'
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION = "1.40.0"
BUILD_SYS = "x86_64-linux"
NATIVELSBSTRING = "ubuntu-16.04"
TARGET_SYS = "sh4-oe-linux"
MACHINE = "spark"
DISTRO = "openspa"
DISTRO_VERSION = "7.4"
TUNE_FEATURES = "sh4"
meta = "HEAD:04469ab5b7f0446404b4cb55a15595678581ab26"
meta-oe
meta-multimedia
meta-networking
meta-filesystems
meta-python
meta-webserver = "HEAD:c26b10b155cf7873ffc8e966f95fd4abed8230c6"
meta-oe = "HEAD:ef66e22b6e75d7c64358cbbbe79012eeca379022"
meta-qt5 = "HEAD:22f3362cfc08453a97a01eedeeac6b03eee1b71d"
meta-airdigital
meta-amiko
meta-azbox
meta-ax
meta-blackbox
meta-beyonwiz
meta-ceryon
meta-clap
meta-cube
meta-dags
meta-dinobot
meta-dream
meta-ebox
meta-edision
meta-entwopia
meta-formuler
meta-fulan
meta-gfutures
meta-gigablue
meta-ini
meta-ixuss
meta-linkdroid
meta-maxytec
meta-broadmedia
meta-odin
meta-odroid
meta-octagon
meta-protek
meta-skylake
meta-tiviar
meta-tripledot
meta-uclan
meta-ultramini
meta-vuplus
meta-wetek
meta-xp
meta-xtrend
meta-xcore = "HEAD:ef66e22b6e75d7c64358cbbbe79012eeca379022"
meta-local = "4.3:14aa416c623104a80a2460bb0b5a9df4817cc328"

Initialising tasks: 100% |####################################| Time: 0:00:29Sstate summary: Wanted 3780 Found 222 Missed 3558 Current 779 (5% match, 21% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks ERROR: enigma2-7.4+gitAUTOINC+130f6a0a6d-r0 do_compile: oe_runmake failed ERROR: enigma2-7.4+gitAUTOINC+130f6a0a6d-r0 do_compile: Function failed: do_compile (log file is located at /home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/amiko8900-oe-linux/enigma2/7.4+gitAUTOINC+130f6a0a6d-r0/temp/log.do_compile.18677) ERROR: Logfile of failure stored in: /home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/amiko8900-oe-linux/enigma2/7.4+gitAUTOINC+130f6a0a6d-r0/temp/log.do_compile.18677
Log data follows:
| DEBUG: SITE files ['endian-little', 'bit-32', 'sh-common', 'common-linux', 'common-glibc', 'sh4-linux', 'common']
[... 446 lines cut ...]
| memcpy(dest, &data[pos+19], slen>35 ? 35 : slen);
| ^
| make[4]: Leaving directory '/home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/amiko8900-oe-linux/enigma2/7.4+gitAUTOINC+130f6a0a6d-r0/git/lib'
| Makefile:2648: recipe for target 'all-recursive' failed
| make[3]: *** [all-recursive] Error 1
| make[3]: Leaving directory '/home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/amiko8900-oe-linux/enigma2/7.4+gitAUTOINC+130f6a0a6d-r0/git/lib'
| ERROR: oe_runmake failed
| Makefile:1163: recipe for target 'all' failed
| make[2]: *** [all] Error 2
| make[2]: Leaving directory '/home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/amiko8900-oe-linux/enigma2/7.4+gitAUTOINC+130f6a0a6d-r0/git/lib'
| Makefile:601: recipe for target 'all-recursive' failed
| make[1]: *** [all-recursive] Error 1
| make[1]: Leaving directory '/home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/amiko8900-oe-linux/enigma2/7.4+gitAUTOINC+130f6a0a6d-r0/git'
| Makefile:444: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: /home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/amiko8900-oe-linux/enigma2/7.4+gitAUTOINC+130f6a0a6d-r0/temp/run.do_compile.18677:1 exit 1 from 'exit 1'
| ERROR: Function failed: do_compile (log file is located at /home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/amiko8900-oe-linux/enigma2/7.4+gitAUTOINC+130f6a0a6d-r0/temp/log.do_compile.18677)
ERROR: Task (/home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2/enigma2.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3419 tasks of which 3178 didn't need to be rerun and 1 failed.
Summary: 1 task failed:
/home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2/enigma2.bb:do_compile
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
Makefile:1026: recipe for target 'image' failed
make: *** [image] Error 1

nickersk, 2020-02-10:
try build from 4.4 branch

Schimmelreiter, 2020-02-10:
It has never been possible to build OpenSPA with just the public portion of the source code.
They must be using local patches or additions to build it.

ahmedmoselhi, 2020-02-10:
Ohk but there is no openspa builds for sh4

nickersk, 2020-02-10:
so the best would be to contact openspa team

ahmedmoselhi, 2020-02-10:
> so the best would be to contact openspa team 
i tried 4.4 branch now got this error:
NOTE: Setscene tasks completed
ERROR: bitstream-1.4+gitAUTOINC+e05f189970-r0 do_compile: oe_runmake failed
ERROR: bitstream-1.4+gitAUTOINC+e05f189970-r0 do_compile: Execution of '/home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/sh4-oe-linux/bitstream/1.4+gitAUTOINC+e05f189970-r0/temp/run.do_compile.16455' failed with exit code 1:
ln -nsf .. examples/bitstream
make -C examples
make[1]: Entering directory '/home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/sh4-oe-linux/bitstream/1.4+gitAUTOINC+e05f189970-r0/git/examples'
sh4-oe-linux-gcc  -ml -m4 --sysroot=/home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/sh4-oe-linux/bitstream/1.4+gitAUTOINC+e05f189970-r0/recipe-sysroot -I. -I.. -I../.. -Wall -Wextra -Wno-unused-parameter -Wno-sign-compare -O2 -g  -Os -pipe  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -lrt -o dvb_print_si dvb_print_si.c
sh4-oe-linux-gcc  -ml -m4 --sysroot=/home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/sh4-oe-linux/bitstream/1.4+gitAUTOINC+e05f189970-r0/recipe-sysroot -I. -I.. -I../.. -Wall -Wextra -Wno-unused-parameter -Wno-sign-compare -O2 -g  -Os -pipe  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -lrt -o dvb_gen_si dvb_gen_si.c
sh4-oe-linux-gcc  -ml -m4 --sysroot=/home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/sh4-oe-linux/bitstream/1.4+gitAUTOINC+e05f189970-r0/recipe-sysroot -I. -I.. -I../.. -Wall -Wextra -Wno-unused-parameter -Wno-sign-compare -O2 -g  -Os -pipe  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -lrt -o mpeg_print_pcr mpeg_print_pcr.c
sh4-oe-linux-gcc  -ml -m4 --sysroot=/home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/sh4-oe-linux/bitstream/1.4+gitAUTOINC+e05f189970-r0/recipe-sysroot -I. -I.. -I../.. -Wall -Wextra -Wno-unused-parameter -Wno-sign-compare -O2 -g  -Os -pipe  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -lrt -o rtp_check_seqnum rtp_check_seqnum.c
In file included from ./bitstream/atsc/si/descs_list.h:31:0,
                 from ./bitstream/mpeg/psi/descs_print.h:35,
                 from ./bitstream/dvb/si_print.h:31,
                 from dvb_print_si.c:42:
./bitstream/atsc/si/desc_81.h: In function 'desc81_bit_rate_code_from_octetrate':
./bitstream/atsc/si/desc_81.h:83:5: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
     for (unsigned i = 0; i < 19; i++) {
     ^
./bitstream/atsc/si/desc_81.h:83:5: note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 to compile your code
Makefile:18: recipe for target 'dvb_print_si' failed
make[1]: *** [dvb_print_si] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from ./bitstream/atsc/si/descs_list.h:31:0,
                 from ./bitstream/mpeg/psi/descs_print.h:35,
                 from ./bitstream/dvb/si_print.h:31,
                 from dvb_gen_si.c:43:
./bitstream/atsc/si/desc_81.h: In function 'desc81_bit_rate_code_from_octetrate':
./bitstream/atsc/si/desc_81.h:83:5: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
     for (unsigned i = 0; i < 19; i++) {
     ^
./bitstream/atsc/si/desc_81.h:83:5: note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 to compile your code
Makefile:18: recipe for target 'dvb_gen_si' failed
make[1]: *** [dvb_gen_si] Error 1
make[1]: Leaving directory '/home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/sh4-oe-linux/bitstream/1.4+gitAUTOINC+e05f189970-r0/git/examples'
Makefile:10: recipe for target 'all' failed
make: *** [all] Error 2
WARNING: /home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/sh4-oe-linux/bitstream/1.4+gitAUTOINC+e05f189970-r0/temp/run.do_compile.16455:1 exit 1 from 'exit 1'

ERROR: Logfile of failure stored in: /home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/sh4-oe-linux/bitstream/1.4+gitAUTOINC+e05f189970-r0/temp/log.do_compile.16455
Log data follows:
| DEBUG: Executing python function autotools_aclocals
| DEBUG: SITE files ['endian-little', 'bit-32', 'sh-common', 'common-linux', 'common-glibc', 'sh4-linux', 'common']
| DEBUG: Python function autotools_aclocals finished
| DEBUG: Executing shell function do_compile
| NOTE: make -j 4
| ln -nsf .. examples/bitstream
| make -C examples
| make[1]: Entering directory '/home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/sh4-oe-linux/bitstream/1.4+gitAUTOINC+e05f189970-r0/git/examples'
| sh4-oe-linux-gcc  -ml -m4 --sysroot=/home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/sh4-oe-linux/bitstream/1.4+gitAUTOINC+e05f189970-r0/recipe-sysroot -I. -I.. -I../.. -Wall -Wextra -Wno-unused-parameter -Wno-sign-compare -O2 -g  -Os -pipe  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -lrt -o dvb_print_si dvb_print_si.c
| sh4-oe-linux-gcc  -ml -m4 --sysroot=/home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/sh4-oe-linux/bitstream/1.4+gitAUTOINC+e05f189970-r0/recipe-sysroot -I. -I.. -I../.. -Wall -Wextra -Wno-unused-parameter -Wno-sign-compare -O2 -g  -Os -pipe  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -lrt -o dvb_gen_si dvb_gen_si.c
[... cut ...]
ERROR: Task (/home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-oe/recipes-support/bitstream/bitstream_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 6245 tasks of which 5870 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-oe/recipes-support/bitstream/bitstream_git.bb:do_compile
Summary: There were 2 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
Makefile:982: recipe for target 'image' failed
make: *** [image] Error 1

nickersk, 2020-02-11:
see https://github.com/oe-alliance/oe-alliance-core/commit/afbcf095aecab06b24b5d5a76a9f1f23ab63e383 and https://github.com/oe-alliance/oe-alliance-core/commit/17e3c278cbd0707ad85aee32e6c10f3a3eb6db13 openspa image should build ok now

ahmedmoselhi, 2020-02-11:
> see [afbcf09](https://github.com/oe-alliance/oe-alliance-core/commit/afbcf095aecab06b24b5d5a76a9f1f23ab63e383) and [17e3c27](https://github.com/oe-alliance/oe-alliance-core/commit/17e3c278cbd0707ad85aee32e6c10f3a3eb6db13) openspa image should build ok now
Initialising tasks: 100% |#######################################################################################################################################################| Time: 0:01:09
Sstate summary: Wanted 1208 Found 860 Missed 348 Current 3351 (71% match, 92% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: stslave-fw-0.1-r7 do_fetch: Failed to fetch URL file:///data/stslave_fw/spark/audio.elf, attempting MIRRORS if available
ERROR: stslave-fw-0.1-r7 do_fetch: Fetcher failure: Unable to find file file:///data/stslave_fw/spark/audio.elf anywhere. The paths that were searched were:
    /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/stslave-fw-0.1/openspa
    /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/stslave-fw/openspa
    /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/files/openspa
    /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/stslave-fw-0.1/spark
    /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/stslave-fw/spark
    /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/files/spark
    /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/stslave-fw-0.1/sti7111
    /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/stslave-fw/sti7111
    /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/files/sti7111
    /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/stslave-fw-0.1/amiko8900
    /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/stslave-fw/amiko8900
    /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/files/amiko8900
    /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/stslave-fw-0.1/allarch
    /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/stslave-fw/allarch
    /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/files/allarch
    /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/stslave-fw-0.1/
    /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/stslave-fw/
    /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/files/
    /home/openspabuilder/openspa/build-enviroment/sources
ERROR: stslave-fw-0.1-r7 do_fetch: Fetcher failure for URL: 'file:///data/stslave_fw/spark/audio.elf'. Unable to fetch URL from any source.
ERROR: stslave-fw-0.1-r7 do_fetch: Function failed: base_do_fetch
ERROR: Logfile of failure stored in: /home/openspabuilder/openspa/build-enviroment/builds/openspa/release/spark/tmp/work/all-oe-linux/stslave-fw/0.1-r7/temp/log.do_fetch.8921
ERROR: Task (/home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/stslave-fw_0.1.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 12864 tasks of which 12681 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/openspabuilder/openspa/build-enviroment/meta-oe-alliance/meta-brands/meta-fulan/recipes-bsp/stslave-fw_0.1.bb:do_fetch
Summary: There were 2 WARNING messages shown.
Summary: There were 3 ERROR messages shown, returning a non-zero exit code.
Makefile:1026: recipe for target 'image' failed
make: *** [image] Error 1
now i got this!!!

nickersk, 2020-02-11:
https://www.opena.tv/howtos/8187-howto-build-own-openatv-sh4-image.html

ahmedmoselhi, 2020-02-12:
compiled sucess!!!!!

closed 2020-02-12 by ahmedmoselhi