oe-alliance/oe-alliance-core #581: Error compile edid-decode.

68foxboris, 2020-02-12:
Error compile edid-decode.

ERROR: edid-decode-1.0+git${SRCPV}-r0 do_compile: Function failed: do_compile (log file is located at /home/foxbob/et/build/tmp/work/armv7ahf-neon-oe-linux-gnueabi/edid-decode/1.0+git${SRCPV}-r0/temp/log.do_compile.8811)
ERROR: Logfile of failure stored in: /home/foxbob/et/build/tmp/work/armv7ahf-neon-oe-linux-gnueabi/edid-decode/1.0+git${SRCPV}-r0/temp/log.do_compile.8811
Log data follows:
| DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
| DEBUG: Executing shell function do_compile
| NOTE: make -j 5
| make: *** No targets specified and no makefile found.  Stop.
| ERROR: oe_runmake failed

nickersk, 2020-02-12:
branch?

68foxboris, 2020-02-12:
Sorry,4.3.4.4 compile ok.

nickersk, 2020-02-12:
there have been some changes upstream... thx for reporting anyway.

closed 2020-02-12 by nickersk