oe-alliance/oe-alliance-core #585: build error opennfr

ahmedmoselhi, 2020-02-21:
Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "sh4-oe-linux"
MACHINE              = "spark"
DISTRO               = "opennfr"
DISTRO_VERSION       = "6.4"
TUNE_FEATURES        = "sh4"
meta                 = "HEAD:ffa91ff19f73c3ed4060a6ecd3b6da7f46732cb3"
meta-oe
meta-multimedia
meta-networking
meta-filesystems
meta-python
meta-webserver       = "HEAD:696408ce3661b663a7895f75a01613e6cfeb8186"
meta-python2         = "HEAD:2c4413b62b7283def2edcda2e1bd0be36bbd5096"
meta-oe              = "HEAD:967087010cc14e117d38371382c1b956db05e66d"
meta-qt5             = "HEAD:a68a39320eca9ca1c7bab60d788436d07cdf827e"
meta-airdigital
meta-amiko
meta-ax
meta-blackbox
meta-beyonwiz
meta-ceryon
meta-cube
meta-dags
meta-dinobot
meta-dream
meta-edision
meta-entwopia
meta-formuler
meta-fulan
meta-gfutures
meta-gigablue
meta-ini
meta-ixuss
meta-maxytec
meta-broadmedia
meta-odin
meta-octagon
meta-protek
meta-skylake
meta-tiviar
meta-tripledot
meta-uclan
meta-ultramini
meta-vuplus
meta-xp
meta-xtrend
meta-xcore           = "HEAD:967087010cc14e117d38371382c1b956db05e66d"
meta-local           = "nextp3:474d3dc5a8e488eb93cc3882d3eb0c1884a40edd"

Initialising tasks: 100% |########################################################################################################################################################################################| Time: 0:00:22
Sstate summary: Wanted 2931 Found 278 Missed 2653 Current 1921 (9% match, 45% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
 /home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sparkreloaded-oe-linux/enigma2/6.4+gitAUTOINC+d89d1ad62c-r0/recipe-sysroot/usr/include/sigc++-2.0/sigc++/signal_base.h: At global scope:
| /home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sparkreloaded-oe-linux/enigma2/6.4+gitAUTOINC+d89d1ad62c-r0/recipe-sysroot/usr/include/sigc++-2.0/sigc++/signal_base.h:198:10: warning: inline function 'sigc::internal::signal_exec::signal_exec(const sigc::internal::signal_impl*)' used but never defined
|    inline signal_exec(const signal_impl* sig) noexcept
|           ^
| Makefile:518: recipe for target 'socket_mmi.lo' failed
| make[8]: *** [socket_mmi.lo] Error 1
| make[8]: Leaving directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sparkreloaded-oe-linux/enigma2/6.4+gitAUTOINC+d89d1ad62c-r0/git/lib/python/Plugins/Extensions/SocketMMI/src'
| Makefile:494: recipe for target 'all-recursive' failed
| make[7]: *** [all-recursive] Error 1
| make[7]: Leaving directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sparkreloaded-oe-linux/enigma2/6.4+gitAUTOINC+d89d1ad62c-r0/git/lib/python/Plugins/Extensions/SocketMMI'
| Makefile:502: recipe for target 'all-recursive' failed
| make[6]: *** [all-recursive] Error 1
| make[6]: Leaving directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sparkreloaded-oe-linux/enigma2/6.4+gitAUTOINC+d89d1ad62c-r0/git/lib/python/Plugins/Extensions'
| Makefile:492: recipe for target 'all-recursive' failed
| make[5]: *** [all-recursive] Error 1
| make[5]: Leaving directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sparkreloaded-oe-linux/enigma2/6.4+gitAUTOINC+d89d1ad62c-r0/git/lib/python/Plugins'
| Makefile:412: recipe for target 'all-recursive' failed
| make[4]: *** [all-recursive] Error 1
| make[4]: Leaving directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sparkreloaded-oe-linux/enigma2/6.4+gitAUTOINC+d89d1ad62c-r0/git/lib/python'
| Makefile:2634: recipe for target 'all-recursive' failed
| make[3]: *** [all-recursive] Error 1
| make[3]: Leaving directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sparkreloaded-oe-linux/enigma2/6.4+gitAUTOINC+d89d1ad62c-r0/git/lib'
| Makefile:1155: recipe for target 'all' failed
| make[2]: *** [all] Error 2
| make[2]: Leaving directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sparkreloaded-oe-linux/enigma2/6.4+gitAUTOINC+d89d1ad62c-r0/git/lib'
| Makefile:597: recipe for target 'all-recursive' failed
| make[1]: *** [all-recursive] Error 1
| make[1]: Leaving directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sparkreloaded-oe-linux/enigma2/6.4+gitAUTOINC+d89d1ad62c-r0/git'
| Makefile:440: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: /home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sparkreloaded-oe-linux/enigma2/6.4+gitAUTOINC+d89d1ad62c-r0/temp/run.do_compile.1348:1 exit 1 from 'exit 1'
|
ERROR: Task (/home/openspabuilder/opennfr/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2/enigma2.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4929 tasks of which 3841 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/openspabuilder/opennfr/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2/enigma2.bb:do_compile
Summary: There were 27 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
Makefile:983: recipe for target 'image' failed
make: *** [image] Error 1

ahmedmoselhi, 2020-02-22:
please help

nickersk, 2020-02-22:
try to set CXXFLAGS_append_sh4 = " -std=c++11" in recipe

ahmedmoselhi, 2020-02-22:
> try to set CXXFLAGS_append_sh4 = " -std=c++11" in recipe

worked!!! thank you

ahmedmoselhi, 2020-02-22:
ERROR: enigma2-3rdparty-plugins-6.4+gitrAUTOINC+5f59d5f93b-r259 do_packagedata: The recipe enigma2-3rdparty-plugins is trying to install files into a shared area when those files already exist. Those files and their manifest location are:
  /home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/pkgdata/spark/runtime/enigma2-plugin-extensions-customsubservices
    (matched in manifest-spark-enigma2.packagedata)
  /home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/pkgdata/spark/runtime/enigma2-plugin-extensions-customsubservices.packaged
    (matched in manifest-spark-enigma2.packagedata)
Please verify which recipe should provide the above files.

The build has stopped, as continuing in this scenario WILL break things - if not now, possibly in the future (we've seen builds fail several months later). If the system knew how to recover from this automatically it would, however there are several different scenarios which can result in this and we don't know which one this is. It may be you have switched providers of something like virtual/kernel (e.g. from linux-yocto to linux-yocto-dev), in that case you need to execute the clean task for both recipes and it will resolve this error. It may be you changed DISTRO_FEATURES from systemd to udev or vice versa. Cleaning those recipes should again resolve this error, however switching DISTRO_FEATURES on an existing build directory is not supported - you should really clean out tmp and rebuild (reusing sstate should be safe). It could be the overlapping files detected are harmless in which case adding them to SSTATE_DUPWHITELIST may be the correct solution. It could also be your build is including two different conflicting versions of things (e.g. bluez 4 and bluez 5 and the correct solution for that would be to resolve the conflict. If in doubt, please ask on the mailing list, sharing the error and filelist above.
ERROR: enigma2-3rdparty-plugins-6.4+gitrAUTOINC+5f59d5f93b-r259 do_packagedata: If the above message is too much, the simpler version is you're advised to wipe out tmp and rebuild (reusing sstate is fine). That will likely fix things in most (but not all) cases.
ERROR: Logfile of failure stored in: /home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/spark-oe-linux/enigma2-3rdparty-plugins/6.4+gitrAUTOINC+5f59d5f93b-r259/temp/log.do_packagedata.6283
ERROR: Task (/home/openspabuilder/opennfr/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-3rdparty-plugins.bb:do_packagedata) failed with exit code '1'
NOTE: Tasks Summary: Attempted 9732 tasks of which 7688 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/openspabuilder/opennfr/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-3rdparty-plugins.bb:do_packagedata
Summary: There were 3 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
Makefile:983: recipe for target 'image' failed
make: *** [image] Error 1
now i got this error @nickersk

ahmedmoselhi, 2020-02-22:
ERROR: cifs-utils-6.10-r0 do_compile: oe_runmake failed
ERROR: cifs-utils-6.10-r0 do_compile: Execution of '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sh4-oe-linux/cifs-utils/6.10-r0/temp/run.do_compile.11703' failed with exit code 1:
make  all-recursive
make[1]: Entering directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sh4-oe-linux/cifs-utils/6.10-r0/build'
Making all in contrib
make[2]: Entering directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sh4-oe-linux/cifs-utils/6.10-r0/build/contrib'
Making all in request-key.d
make[3]: Entering directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sh4-oe-linux/cifs-utils/6.10-r0/build/contrib/request-key.d'
sed 's,[@]sbindir@,/usr/sbin,' ../../../git/contrib/request-key.d/cifs.idmap.conf.in > cifs.idmap.conf-t && mv cifs.idmap.conf-t cifs.idmap.conf
sed 's,[@]sbindir@,/usr/sbin,' ../../../git/contrib/request-key.d/cifs.spnego.conf.in > cifs.spnego.conf-t && mv cifs.spnego.conf-t cifs.spnego.conf
make[3]: Leaving directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sh4-oe-linux/cifs-utils/6.10-r0/build/contrib/request-key.d'
make[3]: Entering directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sh4-oe-linux/cifs-utils/6.10-r0/build/contrib'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sh4-oe-linux/cifs-utils/6.10-r0/build/contrib'
make[2]: Leaving directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sh4-oe-linux/cifs-utils/6.10-r0/build/contrib'
make[2]: Entering directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sh4-oe-linux/cifs-utils/6.10-r0/build'
sh4-oe-linux-gcc  -ml -m4 --sysroot=/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sh4-oe-linux/cifs-utils/6.10-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../git    -Wall -Wextra -D_FORTIFY_SOURCE=2 -fpie -pie -Wl,-z,relro,-z,now -Os -pipe  -c -o smbinfo.o ../git/smbinfo.c
sh4-oe-linux-gcc  -ml -m4 --sysroot=/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sh4-oe-linux/cifs-utils/6.10-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../git    -Wall -Wextra -D_FORTIFY_SOURCE=2 -fpie -pie -Wl,-z,relro,-z,now -Os -pipe  -c -o mount.cifs.o ../git/mount.cifs.c
sh4-oe-linux-gcc  -ml -m4 --sysroot=/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sh4-oe-linux/cifs-utils/6.10-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../git    -Wall -Wextra -D_FORTIFY_SOURCE=2 -fpie -pie -Wl,-z,relro,-z,now -Os -pipe  -c -o mtab.o ../git/mtab.c
sh4-oe-linux-gcc  -ml -m4 --sysroot=/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sh4-oe-linux/cifs-utils/6.10-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../git    -Wall -Wextra -D_FORTIFY_SOURCE=2 -fpie -pie -Wl,-z,relro,-z,now -Os -pipe  -c -o resolve_host.o ../git/resolve_host.c
sh4-oe-linux-gcc  -ml -m4 --sysroot=/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sh4-oe-linux/cifs-utils/6.10-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../git    -Wall -Wextra -D_FORTIFY_SOURCE=2 -fpie -pie -Wl,-z,relro,-z,now -Os -pipe  -c -o util.o ../git/util.c
../git/smbinfo.c: In function 'dump_keys':
../git/smbinfo.c:1134:2: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
  for (int i = 0; i < 8; i++)
  ^
../git/smbinfo.c:1134:2: note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 to compile your code
../git/smbinfo.c:1138:11: error: redefinition of 'i'
  for (int i = 0; i < 16; i++)
           ^
../git/smbinfo.c:1134:11: note: previous definition of 'i' was here
  for (int i = 0; i < 8; i++)
           ^
../git/smbinfo.c:1138:2: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
  for (int i = 0; i < 16; i++)
  ^
../git/smbinfo.c:1141:11: error: redefinition of 'i'
  for (int i = 0; i < SMB3_SIGN_KEY_SIZE; i++)
           ^
../git/smbinfo.c:1138:11: note: previous definition of 'i' was here
  for (int i = 0; i < 16; i++)
           ^
../git/smbinfo.c:1141:2: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
  for (int i = 0; i < SMB3_SIGN_KEY_SIZE; i++)
  ^
../git/smbinfo.c:1144:11: error: redefinition of 'i'
  for (int i = 0; i < SMB3_SIGN_KEY_SIZE; i++)
           ^
../git/smbinfo.c:1141:11: note: previous definition of 'i' was here
  for (int i = 0; i < SMB3_SIGN_KEY_SIZE; i++)
           ^
../git/smbinfo.c:1144:2: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
  for (int i = 0; i < SMB3_SIGN_KEY_SIZE; i++)
  ^
Makefile:871: recipe for target 'smbinfo.o' failed
make[2]: *** [smbinfo.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sh4-oe-linux/cifs-utils/6.10-r0/build'
Makefile:994: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sh4-oe-linux/cifs-utils/6.10-r0/build'
Makefile:502: recipe for target 'all' failed
[... cut ...]
| make[1]: Leaving directory '/home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sh4-oe-linux/cifs-utils/6.10-r0/build'
| Makefile:502: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: /home/openspabuilder/opennfr/build-enviroment/builds/opennfr/release/spark/tmp/work/sh4-oe-linux/cifs-utils/6.10-r0/temp/run.do_compile.11703:1 exit 1 from 'exit 1'
|
ERROR: Task (/home/openspabuilder/opennfr/build-enviroment/meta-openembedded/meta-networking/recipes-support/cifs/cifs-utils_6.10.bb:do_compile) failed with exit code '1'

ahmedmoselhi, 2020-02-22:
cleared temp and rebuilded now i got this error

nickersk, 2020-02-22:
its clearly written in the log, what needs to be done. out of my time for this game.

ahmedmoselhi, 2020-02-23:
build completed successfully after resolving all errors

closed 2020-02-23 by ahmedmoselhi