oe-alliance/oe-alliance-core #586: include e2iStream into plugins list instead of or together with e2iPlayer

rorso, 2020-02-21:
e2IPlayer as it is in the current plugin repository is essentially dead and hardly functional.

A working spin off with active development [is available as "e2IStream"](https://www.sat-universe.com/showthread.php?t=276136&page=100).

While there seems valid interest for some to keep the old plugin running, it might still be a good idea to include e2iStream. Keep unaware casual users from frustrating over an already gone bad plugin by offering a working alternative.

nickersk, 2020-02-22:
we've been discussing this topic sooner already and decided to leave as is now. because original author stopped developing after somebody stolen his private sources...i dont know if he reacted somehow on question if is it ok to use other sources...

ahmedmoselhi, 2020-02-22:
> we've been discussing this topic sooner already and decided to leave as is. because original author stopped developing after somebody stolen his private sources...i dont know if he reacted somehow on question if is it ok to use other sources... what is your opinion guys?

in my opinion u should drop it for now as it may take time for developper to resume plugin development

BarviX6, 2020-02-24:
A working spin off with active development -- https://gitlab.com/zadmario/e2iplayer

kueken, 2020-02-26:
All of this is not really sensible to solve here in the OE core.
e2istream is closed and, according to the information, is being built with a stolen code.
Nobody can judge whether anything has been stolen.
It is definitely the case that the creator in the e2istream was simply overwritten in the code.
A cheek in my eyes.
e2istream also only causes problems, since internally the plugin also uses e2player engine.
This leads to conflicts in software.
Switch to zadmario would work, but is not welcome due to the OpenAtv forum and some obviously important users active there.

nickersk, 2020-02-26:
best would be to write it whole new from scratch imho, but im afraid no much interest by devs exists on this topic.

atvcaptain, 2021-12-24:
new version in branch 5.0

closed 2021-12-24 by atvcaptain