oe-alliance/oe-alliance-core #587: openvix 5.3 build error

ahmedmoselhi, 2020-02-26:
oealliancebuilder@ip-172-31-5-218:~/oealliance/build-enviroment$ MACHINE=sparkreloaded DISTRO=openvix make image
Generating /home/oealliancebuilder/oealliance/build-enviroment/builds/openvix/release/spark/env.source
Generating /home/oealliancebuilder/oealliance/build-enviroment/builds/openvix/release/spark/conf/openvix.conf
Generating /home/oealliancebuilder/oealliance/build-enviroment/builds/openvix/release/spark/conf/bblayers.conf
Generating /home/oealliancebuilder/oealliance/build-enviroment/builds/openvix/release/spark/conf/local.conf
check internet connection: Online
NOTE: Your conf/bblayers.conf has been automatically updated.
NOTE: Your conf/bblayers.conf has been automatically updated.
NOTE: Your conf/bblayers.conf has been automatically updated.
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-wetek/recipes-bsp/wetek-shutdown.bb: Unable to get checksum for wetek-shutdown SRC_URI entry wetek-shutdown.sh: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-wetek/recipes-bsp/wetek-amlsetfb_1.0.bb: Unable to get checksum for wetek-amlsetfb SRC_URI entry amlsetfb: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-wetek/recipes-bsp/wetek-amlsetfb_1.0.bb: Unable to get checksum for wetek-amlsetfb SRC_URI entry aaa: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-vuplus/recipes-bsp/vuplus-shutdown.bb: Unable to get checksum for vuplus-shutdown SRC_URI entry turnoff_power: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-odroid/recipes-drivers/odroid-e2-procfs-odroidc2.bb: Unable to get checksum for odroid-e2-procfs-odroidc2 SRC_URI entry odroid-e2-procfs-3.14.79-20161023.zip: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-odroid/recipes-drivers/odroid-dvb-modules-odroidc2.bb: Unable to get checksum for odroid-dvb-modules-odroidc2 SRC_URI entry odroid-dvb-modules-3.14.79-20161126.zip: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-odroid/recipes-bsp/odroid-shutdown.bb: Unable to get checksum for odroid-shutdown SRC_URI entry spark-shutdown.sh: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-odin/recipes-bsp/odin-backup.bb: Unable to get checksum for odin-backup SRC_URI entry odin-backup.sh: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-linkdroid/recipes-drivers/linkdroid-stb-alien5.bb: Unable to get checksum for linkdroid-stb-alien5 SRC_URI entry spark-stb-20181205.zip: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-linkdroid/recipes-drivers/linkdroid-nftl-dev-alien5.bb: Unable to get checksum for linkdroid-nftl-dev-alien5 SRC_URI entry spark-nftl_dev-20180513.zip: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-linkdroid/recipes-drivers/linkdroid-extra-alien5.bb: Unable to get checksum for linkdroid-extra-alien5 SRC_URI entry spark-extra-20180314.zip: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-linkdroid/recipes-bsp/linkdroid-shutdown.bb: Unable to get checksum for linkdroid-shutdown SRC_URI entry shutdown.sh: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-linkdroid/recipes-bsp/linkdroid-shutdown.bb: Unable to get checksum for linkdroid-shutdown SRC_URI entry amlreboot: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-linkdroid/recipes-bsp/linkdroid-shutdown.bb: Unable to get checksum for linkdroid-shutdown SRC_URI entry amlhalt: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-linkdroid/recipes-bsp/linkdroid-amlsetfb_1.0.bb: Unable to get checksum for linkdroid-amlsetfb SRC_URI entry amlsetfb: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-linkdroid/recipes-bsp/linkdroid-amlsetfb_1.0.bb: Unable to get checksum for linkdroid-amlsetfb SRC_URI entry aaa: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-gigablue/recipes-gb/gigablue-ethwol.bb: Unable to get checksum for gigablue-ethwol SRC_URI entry ethwol.sh: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-gfutures/recipes-graphics/libgles/hd-qteglfs-platform.bb: Unable to get checksum for hd-qteglfs-platform SRC_URI entry libqeglfs-nxpl-integration.so: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-dream/recipes-enigma2/dreambox-enigma2-config.bb: Unable to get checksum for dreambox-enigma2-config SRC_URI entry skin_box.xml: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-dream/recipes-drivers/dreambox-dvb-modules-dm800sev2_3.2-20151201.bb: Unable to get checksum for dreambox-dvb-modules-dm800sev2 SRC_URI entry modules: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-dream/recipes-drivers/dreambox-dvb-modules-dm800se_3.2-20151201.bb: Unable to get checksum for dreambox-dvb-modules-dm800se SRC_URI entry modules: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-dream/recipes-drivers/dreambox-dvb-modules-dm800_2.6.18-7.4-20131228a.bb: Unable to get checksum for dreambox-dvb-modules-dm800 SRC_URI entry modules: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-dream/recipes-drivers/dreambox-dvb-modules-dm8000_3.2-20140604a.bb: Unable to get checksum for dreambox-dvb-modules-dm8000 SRC_URI entry modules: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-dream/recipes-drivers/dreambox-dvb-modules-dm7020hd_3.2-20161019.bb: Unable to get checksum for dreambox-dvb-modules-dm7020hd SRC_URI entry modules: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-dream/recipes-drivers/dreambox-dvb-modules-dm500hd_3.2-20140616.bb: Unable to get checksum for dreambox-dvb-modules-dm500hd SRC_URI entry modules: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-dream/recipes-drivers/dreambox-dvb-modules-dm7020hdv2_3.2-20161019.bb: Unable to get checksum for dreambox-dvb-modules-dm7020hdv2 SRC_URI entry modules: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-dream/recipes-drivers/dreambox-dvb-modules-dm500hdv2_3.2-20140616.bb: Unable to get checksum for dreambox-dvb-modules-dm500hdv2 SRC_URI entry modules: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-brands/meta-dags/recipes-bsp/dags-shutdown.bb: Unable to get checksum for dags-shutdown SRC_URI entry turnoff_power: file could not be found
WARNING: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-oe/recipes-multimedia/gstreamer1.0/gstreamer1.0-libav_git.bb: Unable to get checksum for gstreamer1.0-libav SRC_URI entry libav-fix-sh4-compile-gcc48.patch: file could not be found
NOTE: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-oe/recipes-core/glibc/glibc_2.23.bb: glibc doesn't build cleanly with -Os, adding -Wno-error to SELECTED_OPTIMIZATION
NOTE: /home/oealliancebuilder/oealliance/build-enviroment/meta-oe-alliance/meta-oe/recipes-core/glibc/glibc_2.23.bb: nativesdk-glibc doesn't build cleanly with -Os, adding -Wno-error to SELECTED_OPTIMIZATION
[... cut ...] ilder/oealliance/build-enviroment/builds/openvix/release/spark/tmp/work/sparkreloaded-oe-linux/enigma2/enigma2-5.3+gitAUTOINC+2b1ade23b7-r0/recipe-sysroot/usr/include/orc-0.4 -I/home/oealliancebuilder/oealliance/build-enviroment/builds/openvix/release/spark/tmp/work/sparkreloaded-oe-linux/enigma2/enigma2-5.3+gitAUTOINC+2b1ade23b7-r0/recipe-sysroot/usr/include/gstreamer-1.0 -I/home/oealliancebuilder/oealliance/build-enviroment/builds/openvix/release/spark/tmp/work/sparkreloaded-oe-linux/enigma2/e

nickersk, 2020-02-26:
search for "error:" in your spam, i am sure you'll find soon where the problem is ;-)

carlo0815, 2020-02-26:
Why do you think the Teams don't build self for sh4?
Many things for sh4 build be missing in teamGit so you must fixing the missing Code before you build it.

ahmedmoselhi, 2020-02-27:
now i got this
ERROR: Task (virtual:native:/home/oealliancebuilder/build-enviroment/openembedded-core/meta/recipes-graphics/fontconfig/fontconfig_2.13.1.bb:do_compile) failed with exit code '1'

ahmedmoselhi, 2020-03-01:
giving up!!!!!!

closed 2020-03-01 by ahmedmoselhi