oe-alliance/oe-alliance-core #588: Error compile dvdbackup.

68foxboris, 2020-03-04:
Error compile dvdbackup after this https://code.videolan.org/videolan/libdvdread/-/commits/master

`
| ../../dvdbackup-0.4.2/src/dvdbackup.c: In function 'DVDWriteCells':
| ../../dvdbackup-0.4.2/src/dvdbackup.c:172:22: warning: unused parameter 'title_set_info' [-Wunused-parameter]
|   172 |   title_set_info_t * title_set_info, titles_info_t * titles_info,
|       |   ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
| ../../dvdbackup-0.4.2/src/dvdbackup.c: In function 'DVDCopyIfoBup':
| ../../dvdbackup-0.4.2/src/dvdbackup.c:1212:29: error: 'ifo_handle_t' {aka 'struct <anonymous>'} has no member named 'file'
|  1212 |  size = DVDFileSize(ifo_file->file) * DVD_VIDEO_LB_LEN;
|       |                             ^~
| ../../dvdbackup-0.4.2/src/dvdbackup.c:1223:22: error: 'ifo_handle_t' {aka 'struct <anonymous>'} has no member named 'file'
|  1223 |  DVDFileSeek(ifo_file->file, 0);
|       |                      ^~
| ../../dvdbackup-0.4.2/src/dvdbackup.c:1225:27: error: 'ifo_handle_t' {aka 'struct <anonymous>'} has no member named 'file'
|  1225 |  if (DVDReadBytes(ifo_file->file,buffer,size) != size) {
|       |                           ^~
| make[2]: *** [Makefile:387: dvdbackup.o] Error 1
| make[2]: *** Waiting for unfinished jobs....
`

nickersk, 2020-03-04:
branch, machine?

nickersk, 2020-03-04:
branch/machine/distro?

68foxboris, 2020-03-04:
4.4,et11000.It seems to me it does not depend on it.On 4.3 the same error.

68foxboris, 2020-03-04:
Latest stable release 0.4.2 (2012-06-24)-Maybe it also depends on this?

nickersk, 2020-03-04:
what is your buildsystem ?

68foxboris, 2020-03-04:
Ok.

closed 2020-03-04 by 68foxboris