oe-alliance/oe-alliance-core #589: Problem with USB DVB-T2 dongle Hauppauge WinTv DualHD

PetrDubi, 2020-03-05:
Hi,
I bought new USB dongle and I try it in OpenATV 6.3 on Formuler F4 Turbo (kernel 4.10.6).
First problem is with USB ID. In drivers is USB ID 2040:0265 and this dongle has USB ID 2040:8265 (bulk id). If I was plug dongle, then driver was not recognized tuner (plugin ct2-dvb-usb-dualhd is installed).  I was try this trick:
modprobe em28xx card=99
echo "2040 8268" > /sys/bus/usb/drivers/em28xx/new_id
Ater that driver recognized tuner, but something is wrong.
After Binding DVB extension is:
no endpoint for DVB mode and transfer type 1
failed to pre-allocate USB transfer buffers for DVB
Not Added multiplexed i2c bus and not recognized si2168 and si2157.
At end of dmesg is:
"formuler4turbo_dvb" "formuler4turbo_dvb".0: dtv_property_legacy_params_sync: doesn't know how to handle a DVBv3 call to delivery system 0

I put log here

Best regards Petr

[Dmesg.txt](https://github.com/oe-alliance/oe-alliance-core/files/4292106/Dmesg.txt)

JJussi, 2020-11-17:
https://hauppauge.com/pages/support/support_linux.html

atvcaptain, 2022-01-08:
the kernel to old, from kernel 4.12 or newer we need a backport for the new version from the tuner, but the box was long time eol

closed 2022-01-08 by atvcaptain