oe-alliance/oe-alliance-core #601: gst1.0 - meson

68foxboris, 2020-04-05:
Good afternoon. Now what ffmpeg 4.2.2 i think it did not solve the problem with dvdvideo and other.
I add in ffmpeg- --enable-libaom \ --m_errorInfo.missing_codec = video/x-av1(maybe I'm doing something wrong).
I also can not watch the video through MovieSelection.
And now the file *.ape - [eServiceMP3] m_errorInfo.missing_codec = video/x-ape.If through ffmpeg-meson file in visible,but reproduces with errors.

68foxboris, 2020-05-06:
Now used in libav dependencies ffmpeg,but gst-inspect says different.
root@et1x000:~# gst-inspect-1.0 | grep mpeg2
mpeg2dec:  mpeg2dec: mpeg1 and mpeg2 video decoder
libav:  avdec_mpeg2video: libav MPEG-2 video decoder
libav:  avenc_mpeg2video: libav MPEG-2 video encoder
root@et1x000:~# gst-inspect-1.0 | grep av1
aom:  av1dec: AV1 Decoder
aom:  av1enc: AV1 Encoder
libav:  avdec_wmav1: libav Windows Media Audio 1 decoder
libav:  avenc_wmav1: libav Windows Media Audio 1 encoder

nickersk, 2020-05-06:
https://cgit.freedesktop.org/gstreamer/gst-plugins-bad/commit/?id=13d55627f0be18c52dd1019c1f464acfe2da8b98

68foxboris, 2020-05-06:
[Ok,](https://drive.google.com/file/d/19fGURecmw-EINXq2VsedM3Y7FtiUiv6N/view?usp=sharing)
[Navigation] playing:  4097:0:0:0:0:0:0:0:0:0:/media/hdd/4kvideo/BeachDrone-AV1.webm
[eServiceFactoryMP3] ****new play service total services played is 3****
[eServiceMP3] construct!
[eServiceMP3] playbin uri=file:///media/hdd/4kvideo/BeachDrone-AV1.webm
[eServiceMP3] *** starting pipeline ****
[eServiceMP3] ****STATE TRANSITION NULL -> READY ****
[eServiceMP3] PLAYBIN WITH BLOCK READY TO PAUSED state:PAUSED pending:VOID_PENDING ret:SUCCESS
[eServiceMP3] m_errorInfo.missing_codec = video/x-av1, colorimetry=(string)2:0:0:0
[eServiceMP3] ****STATE TRANSITION READY -> PAUSED ****
4097:0:0:0:0:0:0:0:0:0:/MEDIA/HDD/4KVIDEO/BEACHDRONE-AV1WEBM
4097:0:0:0:0:0:0:0:0:0:/MEDIA/HDD/4KVIDEO/BEACHDRONE-AV1WEBM

nickersk, 2020-05-06:
thats ok, no av1 codec enabled in ff.

68foxboris, 2020-05-06:
Nothing happens with ape
[Navigation] playing:  4097:0:0:0:0:0:0:0:0:0:/media/hdd/music/Mylene Farmer - Interstellaires (2015)/Mylene Farmer - Interstellaires.ape
[eServiceFactoryMP3] ****new play service total services played is 2****
[eServiceMP3] construct!
[eServiceMP3] playbin uri=file:///media/hdd/music/Mylene%20Farmer%20-%20Interstellaires%20(2015)/Mylene%20Farmer%20-%20Interstellaires.ape
CRITICAL **: 16:51:31.135: GstAudioMeta properties would cause out-of-bounds memory access on the buffer: max_offset 9216, samples 4608, bps 2, buffer size 5292
[eServiceMP3] PLAYBIN WITH BLOCK PLAYSTART state:PLAYING pending:VOID_PENDING ret:SUCCESS
0:00:26.075175169  3400 0xadf81350 ERROR             GST_BUFFER gstbuffer.c:687:gst_buffer_copy_into: failed to copy meta 0x1311da8 of API type GstAudioMetaAPI
[eServiceMP3] GST_MESSAGE_ASYNC_DONE before evUpdatedInfo
root@et1x000:~# gst-inspect-1.0 | grep ape
taglib:  apev2mux: TagLib-based APEv2 Muxer
apetag:  apedemux: APE tag demuxer
typefindfunctions: application/x-apetag: mp3, ape, mpc, wv
typefindfunctions: application/x-ape: ape
libav:  avdec_ape: libav Monkey's Audio decoder
libav:  avdec_escape124: libav Escape 124 decoder
libav:  avdec_escape130: libav Escape 130 decoder
libav:  avdemux_ape: libav Monkey's Audio demuxer

68foxboris, 2020-05-06:
I aff ffmpeg PACKAGECONFIG[dav1d] = "--enable-libdav1d,--disable-libdav1d,libdav1d",but there are no errors during assembly, I’m probably doing something wrong.

nickersk, 2020-05-06:
add dav1d to PACKAGECONFIG to enable it.

68foxboris, 2020-05-06:
PACKAGECONFIG_append = " libass libbluray libfreetype librtmp libvorbis \
                        mp3lame openjpeg openssl wavpack fdk-aac \
                        libbluray dav1d"

68foxboris, 2020-05-09:
Here are some logs.
`
root@et1x000:~# gst-launch-1.0 --gst-debug-level=2 playbin uri=file:///media/hdd/4kvideo/BeachDrone-AV1.webm
Setting pipeline to PAUSED ...
0:00:00.053165259  1473   0x167e60 WARN                 basesrc gstbasesrc.c:3631:gst_base_src_start_complete:<source> pad not activated yet
0:00:00.055640555  1473   0x167e60 WARN                 basesrc gstbasesrc.c:3631:gst_base_src_start_complete:<source> pad not activated yet
Pipeline is PREROLLING ...
Missing element: AV1 decoder
0:00:00.137012185  1473 0xb5801af0 WARN            uridecodebin gsturidecodebin.c:957:unknown_type_cb:<uridecodebin0> warning: No decoder available for type 'video/x-av1, codec_data=(buffer)811f0d000a0b000000fa6355ef8842b068, width=(int)854, height=(int)480, framerate=(fraction)25/1, colorimetry=(string)2:0:0:0'.
WARNING: from element /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: No decoder available for type 'video/x-av1, codec_data=(buffer)811f0d000a0b000000fa6355ef8842b068, width=(int)854, height=(int)480, framerate=(fraction)25/1, colorimetry=(string)2:0:0:0'.
Additional debug info:
../git/gst/playback/gsturidecodebin.c(957): unknown_type_cb (): /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0
0:00:00.176626111  1473 0xb5801af0 WARN      matroskareadcommon matroska-read-common.c:711:gst_matroska_read_common_parse_skip:<matroskademux0:sink> Unknown CueTrackPositions subelement 0xf0 - ignoring
0:00:00.176744222  1473 0xb5801af0 WARN      matroskareadcommon matroska-read-common.c:711:gst_matroska_read_common_parse_skip:<matroskademux0:sink> Unknown CueTrackPositions subelement 0xf0 - ignoring
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
0:00:00.201339777  1473   0x188f50 WARN         audio-resampler audio-resampler.c:274:convert_taps_gint16_c: can't find exact taps
Got EOS from element "playbin0".
Execution ended after 0:00:14.299138177
Setting pipeline to NULL ...
Freeing pipeline ...
`

`
root@et1x000:~# gst-launch-1.0 --gst-debug-level=2 playbin uri=file:///media/hdd/music/Interstellaires.ape
Setting pipeline to PAUSED ...
0:00:00.053775407  1434   0x167e60 WARN                 basesrc gstbasesrc.c:3631:gst_base_src_start_complete:<source> pad not activated yet
0:00:00.056115148  1434   0x167e60 WARN                 basesrc gstbasesrc.c:3631:gst_base_src_start_complete:<source> pad not activated yet
Pipeline is PREROLLING ...
0:00:00.175999629  1434 0xb5801ac0 WARN                   libav gstavdemux.c:1484:gst_ffmpegdemux_loop:<avdemux_ape0> invalid buffer duration, setting to NONE
0:00:00.177510370  1434 0xb5801ac0 WARN                   libav gstavdemux.c:1484:gst_ffmpegdemux_loop:<avdemux_ape0> invalid buffer duration, setting to NONE
0:00:00.178726963  1434 0xb5801ac0 WARN                   libav gstavdemux.c:1484:gst_ffmpegdemux_loop:<avdemux_ape0> invalid buffer duration, setting to NONE
0:00:00.179795629  1434 0xb5801ac0 WARN                   libav gstavdemux.c:1484:gst_ffmpegdemux_loop:<avdemux_ape0> invalid buffer duration, setting to NONE
0:00:00.181074777  1434 0xb5801ac0 WARN                   libav gstavdemux.c:1484:gst_ffmpegdemux_loop:<avdemux_ape0> invalid buffer duration, setting to NONE
0:00:00.182423629  1434 0xb5801ac0 WARN                   libav gstavdemux.c:1484:gst_ffmpegdemux_loop:<avdemux_ape0> invalid buffer duration, setting to NONE
0:00:00.183949370  1434 0xb5801ac0 WARN                   libav gstavdemux.c:1484:gst_ffmpegdemux_loop:<avdemux_ape0> invalid buffer duration, setting to NONE
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
0:00:22.040716579  1434 0xb5801ac0 WARN                   libav gstavdemux.c:1484:gst_ffmpegdemux_loop:<avdemux_ape0> invalid buffer duration, setting to NONE
0:00:22.047242468  1434 0xb1802d20 WARN                   libav gstavauddec.c:749:gst_ffmpegauddec_handle_frame:<avdec_ape0> decoding error
0:00:22.047901024  1434 0xb1802d20 WARN                   libav gstavauddec.c:749:gst_ffmpegauddec_handle_frame:<avdec_ape0> decoding error
0:00:22.048484246  1434 0xb1802d20 WARN                   libav gstavauddec.c:749:gst_ffmpegauddec_handle_frame:<avdec_ape0> decoding error
0:00:22.049125875  1434 0xb1802d20 WARN                   libav gstavauddec.c:749:gst_ffmpegauddec_handle_frame:<avdec_ape0> decoding error
0:00:22.049665061  1434 0xb1802d20 WARN                   libav gstavauddec.c:749:gst_ffmpegauddec_handle_frame:<avdec_ape0> decoding error
0:00:22.050307690  1434 0xb1802d20 WARN                   libav gstavauddec.c:749:gst_ffmpegauddec_handle_frame:<avdec_ape0> decoding error
0:00:22.051308431  1434 0xb5801ac0 WARN                   libav gstavdemux.c:1484:gst_ffmpegdemux_loop:<avdemux_ape0> invalid buffer duration, setting to NONE
0:00:22.051623283  1434 0xb5801ac0 WARN                   libav gstavdemux.c:1578:gst_ffmpegdemux_loop:<avdemux_ape0> stream_movi flow: flushing / flushing
Freeing pipeline ...
`

`
root@et1x000:~# gst-launch-1.0 --gst-debug-level=2 playbin uri=file:///media/hdd/4kvideo/BBB-AVC3.mp4
Setting pipeline to PAUSED ...
0:00:00.053113148  1480   0x166e60 WARN                 basesrc gstbasesrc.c:3631:gst_base_src_start_complete:<source> pad not activated yet
0:00:00.055608333  1480   0x166e60 WARN                 basesrc gstbasesrc.c:3631:gst_base_src_start_complete:<source> pad not activated yet
Pipeline is PREROLLING ...
[... cut ...]
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
0:00:00.133223852  1480 0xb4109260 WARN            dvbvideosink gstdvbvideosink.c:1556:gst_dvbvideosink_set_caps:<dvbvideosink0> codec_data too short(1)
0:00:02.809557517  1480 0xb5801e00 WARN                 qtdemux qtdemux.c:796:gst_qtdemux_pull_atom:<qtdemux0> short read: 538 < 539
^Chandling interrupt. (9.4 %)
Interrupt: Stopping pipeline ...
Execution ended after 0:00:10.510411883
Setting pipeline to NULL ...
Freeing pipeline ...
`

68foxboris, 2020-05-09:
`
root@et1x000:~# gst-launch-1.0 -v --gst-debug-level=3 playbin uri=file:///media/hdd/music/Interstellaires.ape
Setting pipeline to PAUSED ...
0:00:00.054383222 [331m 1542[00m   0x167460 [33;01mWARN   [00m [00m             basesrc gstbasesrc.c:3631:gst_base_src_start_complete:<source>[00m pad not activated yet
0:00:00.056241518 [331m 1542[00m   0x167460 [33;01mWARN   [00m [00m             basesrc gstbasesrc.c:3631:gst_base_src_start_complete:<source>[00m pad not activated yet
Pipeline is PREROLLING ...
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: ring-buffer-max-size = 0
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: buffer-size = -1
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: buffer-duration = -1
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: force-sw-decoders = false
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: use-buffering = false
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: download = false
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: uri = file:///media/hdd/music/Interstellaires.ape
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: connection-speed = 0
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: source = "\(GstFileSrc\)\ source"
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstTypeFindElement:typefind.GstPad:src: caps = application/x-ape
0:00:00.158216407 [331m 1542[00m   0x167460 [32;01mFIXME  [00m [00;01m                 bin gstbin.c:4359:gst_bin_query:[00m implement duration caching in GstBin again
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstTypeFindElement:typefind.GstPad:src: caps = NULL
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMultiQueue:multiqueue0.GstMultiQueuePad:src_0: caps = audio/x-ffmpeg-parsed-ape, rate=(int)44100, channels=(int)2, channel-mask=(bitmask)0x0000000000000003, depth=(int)16, codec_data=(buffer)960fd0070000
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/avdec_ape:avdec_ape0.GstPad:sink: caps = audio/x-ffmpeg-parsed-ape, rate=(int)44100, channels=(int)2, channel-mask=(bitmask)0x0000000000000003, depth=(int)16, codec_data=(buffer)960fd0070000
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMultiQueue:multiqueue0: max-size-buffers = 5
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMultiQueue:multiqueue0: max-size-time = 0
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMultiQueue:multiqueue0: max-size-bytes = 2097152
0:00:00.175792074  1542 0xb5801ac0 WARN                   libav gstavdemux.c:1484:gst_ffmpegdemux_loop:<avdemux_ape0> invalid buffer duration, setting to NONE
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMultiQueue:multiqueue0.GstMultiQueuePad:sink_0: caps = audio/x-ffmpeg-parsed-ape, rate=(int)44100, channels=(int)2, channel-mask=(bitmask)0x0000000000000003, depth=(int)16, codec_data=(buffer)960fd0070000
0:00:00.177137037  1542 0xb5801ac0 WARN                   libav gstavdemux.c:1484:gst_ffmpegdemux_loop:<avdemux_ape0> invalid buffer duration, setting to NONE
0:00:00.178354148  1542 0xb5801ac0 WARN                   libav gstavdemux.c:1484:gst_ffmpegdemux_loop:<avdemux_ape0> invalid buffer duration, setting to NONE
0:00:00.179283703  1542 0xb5801ac0 WARN                   libav gstavdemux.c:1484:gst_ffmpegdemux_loop:<avdemux_ape0> invalid buffer duration, setting to NONE
0:00:00.180277926  1542 0xb5801ac0 WARN                   libav gstavdemux.c:1484:gst_ffmpegdemux_loop:<avdemux_ape0> invalid buffer duration, setting to NONE
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/avdec_ape:avdec_ape0.GstPad:src: caps = audio/x-raw, format=(string)S16LE, layout=(string)non-interleaved, rate=(int)44100, channels=(int)2, channel-mask=(bitmask)0x0000000000000003
0:00:00.181442444  1542 0xb5801ac0 WARN                   libav gstavdemux.c:1484:gst_ffmpegdemux_loop:<avdemux_ape0> invalid buffer duration, setting to NONE
0:00:00.183826629  1542 0xb5801ac0 WARN                   libav gstavdemux.c:1484:gst_ffmpegdemux_loop:<avdemux_ape0> invalid buffer duration, setting to NONE
/GstPlayBin:playbin0/GstInputSelector:inputselector0.GstSelectorPad:sink_0: always-ok = false
/GstPlayBin:playbin0/GstInputSelector:inputselector0.GstSelectorPad:sink_0: active = true
/GstPlayBin:playbin0/GstInputSelector:inputselector0: active-pad = "\(GstSelectorPad\)\ sink_0"
/GstPlayBin:playbin0/GstPlaySink:playsink: volume = 1
/GstPlayBin:playbin0/GstPlaySink:playsink: mute = false
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPlayBin:playbin0/GstInputSelector:inputselector0.GstPad:src: caps = audio/x-raw, format=(string)S16LE, layout=(string)non-interleaved, rate=(int)44100, channels=(int)2, channel-mask=(bitmask)0x0000000000000003
/GstPlayBin:playbin0/GstPlaySink:playsink.GstGhostPad:audio_sink.GstProxyPad:proxypad4: caps = audio/x-raw, format=(string)S16LE, layout=(string)non-interleaved, rate=(int)44100, channels=(int)2, channel-mask=(bitmask)0x0000000000000003
/GstPlayBin:playbin0/GstPlaySink:playsink/GstTee:audiotee.GstTeePad:src_0: caps = audio/x-raw, format=(string)S16LE, layout=(string)non-interleaved, rate=(int)44100, channels=(int)2, channel-mask=(bitmask)0x0000000000000003
[... cut ...]
Setting pipeline to NULL ...
0:00:07.888044958  1542 0xb5801ac0 WARN                   libav gstavdemux.c:1484:gst_ffmpegdemux_loop:<avdemux_ape0> invalid buffer duration, setting to NONE
0:00:07.892575254  1542 0xb5801ac0 WARN                   libav gstavdemux.c:1484:gst_ffmpegdemux_loop:<avdemux_ape0> invalid buffer duration, setting to NONE
0:00:07.892823217  1542 0xb5801ac0 WARN                   libav gstavdemux.c:1578:gst_ffmpegdemux_loop:<avdemux_ape0> stream_movi flow: flushing / flushing
Freeing pipeline ...
`

closed 2020-08-02 by 68foxboris