oe-alliance/oe-alliance-core #607: How to extract the kernel config file for Zgemma H7S

FlexMcMurphy, 2020-04-30:
I want to confirm if CONFIG_EFI_PARTITION is supported on my Zgemma H7S.

What command do I run on a Zgemma H7S to get the Kernel config file?

Cheers,

Flex

atvcaptain, 2020-04-30:
https://github.com/oe-alliance/oe-alliance-core/blob/4.4/meta-brands/meta-airdigital/recipes-linux/linux-airdigital-4.10.12/h7/defconfig

FlexMcMurphy, 2020-04-30:
Thanks a lot for this.

But out of interest. Is it possible to run a command from the command line that gives me the kernel config file from the Zgemma H7S?

On an Amiko Alien2 I can do this:
```
zcat /proc/config.gz > /tmp/Kconfig
cat /tmp/Kconfig | grep "CONFIG_EFI_PARTITION"
```

On a Raspberry Pi Raspbian OS I can do this:
```
modprobe configs
zcat /proc/config.gz > /tmp/Kconfig
cat /tmp/Kconfig | grep "CONFIG_EFI_PARTITION"
```

and also this:
`cat /lib/modules/$(uname -r)/build/.config`

On OSMC I can do this:
`cat /boot/config-4.19.55-6-osmc | grep "CONFIG_EFI_PARTITION"`

But I don't think any of these efforts work on OpenATV on Zgemma H7S. It doesn't recognise the configs module and there is no config file already in /proc or /boot

Am I doing something wrong? Thank you for that link but I would still like to know is there a command that can get the kernel config file from my system?

Flex

atvcaptain, 2020-05-01:
stb have limit flash size we dont add this, you find all files in the git

FlexMcMurphy, 2020-05-03:
When I installed OpenATV on an Amiko Alien2 the kernel config file is in /proc/config.gz
It's only 12KB. Many set top boxes (stb) these days have at least 1GB flash memory. I think that config file should be available on the host stb so anyone can confirm what its kernel config is using their own system and not going to GitHub that is disconnected from your own stb.

Actually, in OpenATV you can run the command:
`# modprobe configs`

And this creates /proc/config.gz
Maybe I can try this on a Zgemma H7S?

Cheers,

Flex

atvcaptain, 2022-01-08:
add the support with image 7.1  

https://github.com/oe-alliance/oe-alliance-core/commit/72986ec59f226a4700d6d43d69e808d54062b91e

closed 2022-01-08 by atvcaptain