oe-alliance/oe-alliance-core #611: openvix build error

ahmedmoselhi, 2020-05-14:
| dvb/frontend.cpp: In member function 'int eDVBFrontend::openFrontend()':
| dvb/frontend.cpp:730:70: error: 'SYS_DVBC_ANNEX_A' was not declared in this scope
|   m_multitype = m_delsys[SYS_DVBS] && (m_delsys[SYS_DVBT] || m_delsys[SYS_DVBC_ANNEX_A]);
|                                                                       ^
| dvb/frontend.cpp: In member function 'void eDVBFrontend::setFrontend(bool)':
| dvb/frontend.cpp:2184:92: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
|      p[cmdseq.num].cmd = DTV_ISDBT_SB_SEGMENT_IDX, p[cmdseq.num].u.data = (parm.t2mi_plp_id == eDVBFrontendParametersSatellite::No_T2MI_PLP_Id ? 0 : (0x80000000 | (parm.t2mi_pid << 16) | parm.t2mi_plp_id)), cmdseq.num++;
|                                                                                             ^
| dvb/frontend.cpp: In member function 'virtual bool eDVBFrontend::setDeliverySystem(const char*)':
| dvb/frontend.cpp:3023:17: error: 'SYS_DVBC_ANNEX_A' was not declared in this scope
|    p[0].u.data = SYS_DVBC_ANNEX_A;
|                  ^
| dvb/frontend.cpp: In member function 'std::string eDVBFrontend::getCapabilities()':
| dvb/frontend.cpp:3159:8: error: 'SYS_DVBC_ANNEX_A' was not declared in this scope
|    case SYS_DVBC_ANNEX_A: ss << " DVBC_ANNEX_A"; break;
|         ^
| dvb/frontend.cpp:3160:8: error: 'SYS_DVBC_ANNEX_C' was not declared in this scope
|    case SYS_DVBC_ANNEX_C: ss << " DVBC_ANNEX_C"; break;
|         ^
| dvb/frontend.cpp:3162:8: error: 'SYS_TURBO' was not declared in this scope
|    case SYS_TURBO:  ss << " TURBO"; break;
|         ^
| dvb/frontend.cpp:3143:10: warning: enumeration value 'SYS_DVBC_ANNEX_AC' not handled in switch [-Wswitch]
|    switch (it->first)
|           ^
| dvb/frontend.cpp:3143:10: warning: enumeration value 'SYS_DMBTH' not handled in switch [-Wswitch]
| Makefile:2621: recipe for target 'dvb/frontend.o' failed

ahmedmoselhi, 2020-05-14:
using master branch of openvix

AbuBaniaz, 2020-05-14:
Are you building for a receiver we support?

ahmedmoselhi, 2020-05-14:
> Are you building for a receiver we support?

i build for sh4 receiver , vusolo2

ahmedmoselhi, 2020-05-14:
booth same error

nickersk, 2020-05-14:
he is missing defs from kernel headers, its not vix problem.

ahmedmoselhi, 2020-05-14:
i fixed it by removing that code

closed 2020-05-14 by ahmedmoselhi