oe-alliance/oe-alliance-core #620: [dinobot] u533 no audio/video.

badproject, 2020-08-03:
Last E2 images haven't audio and video at DVB channels. Search and tune without problem but cannot play.
Also, can not play any content from USB media stick.

STB Model: Ferguson Ariva ATV TT 
OEM Model: u533
Drivers: 12.12.2019

[dinobot] u533

atvcaptain, 2020-08-03:
see you at boot the openatv bootlogo not the first splash image,

badproject, 2020-08-03:
Show openatv bootlogo and GUI.
Also cannot take screenshot, it's seems have a driver problem.

`root@arivatwin:~# grab -o -p /tmp/grab.png`
write reg:119 [0]error! res:-1 No such device or address
`root@arivatwin:~# grab -d -p /tmp/grab.png`
write reg:119 [0]error! res:-1 No such device or address

badproject, 2020-08-08:
All dinobot u53x models have same problem.

- u53
- u532
- u533

atvcaptain, 2020-08-09:
u53 here all ok

[image]

badproject, 2020-11-22:
Seems this part have problem

> 23:42:56.7287 [   ] base/ebase.cpp:280 processOneEvent [eMainloop::processOneEvent] unhandled POLLERR/HUP/NVAL for fd 37(17)
23:42:56.7288 [   ] base/ebase.cpp:280 processOneEvent [eMainloop::processOneEvent] unhandled POLLERR/HUP/NVAL for fd 40(17)
23:42:56.7289 [   ] base/ebase.cpp:280 processOneEvent [eMainloop::processOneEvent] unhandled POLLERR/HUP/NVAL for fd 43(17)
23:42:56.7289 [   ] base/ebase.cpp:280 processOneEvent [eMainloop::processOneEvent] unhandled POLLERR/HUP/NVAL for fd 46(17)
23:42:56.7289 [   ] base/ebase.cpp:280 processOneEvent [eMainloop::processOneEvent] unhandled POLLERR/HUP/NVAL for fd 47(16)
23:42:56.7290 [   ] base/ebase.cpp:280 processOneEvent [eMainloop::processOneEvent] unhandled POLLERR/HUP/NVAL for fd 49(17)
23:42:56.7290 [   ] base/ebase.cpp:280 processOneEvent [eMainloop::processOneEvent] unhandled POLLERR/HUP/NVAL for fd 52(17)

[Enigma2-debug-20201122_23-42-19.log](https://github.com/oe-alliance/oe-alliance-core/files/5580389/Enigma2-debug-20201122_23-42-19.log)

badproject, 2021-02-07:
Problem not solved

atvcaptain, 2022-01-08:
you must check serial log, old atv working ?

closed 2022-01-08 by atvcaptain

badproject, 2022-01-30:
> you must check serial log, old atv working ?

This device haven't serial port.
Old ATV not working.
Android side working without problem.

atvcaptain, 2022-01-30:
box have serial internal the 3 pin header

badproject, 2022-01-31:
Do you know location at mainboard?