oe-alliance/oe-alliance-core #631: Octagon SF8008 drivers outdate and buggy

jeepcook, 2020-10-05:
Currently the drivers installed for octagon sf8008 are dated from 13 june 2020. But these drivers are buggy. New drivers exist dated from 03/07/2020 and solved all bugs. I don't know exactly where to find source but I have an ipk tested and installed on openatv 6.4 & 6.5 and solved zapping bug. You have a discussion about this and solved here with link to ipk (compiled drivers) https://www.opena.tv/openatv-6-5-python3-entwicklung/52858-octagon-sf8008-6-5-compilation-erros-question-post452382.html#post452382

Hope I post this issue at the good place.

Thanks

nickersk, 2020-10-06:
Well, we need download URL of drivers archive provided by Octagon to update bb recipe. You may contact Octagon to provide it and paste it here. We can update drivers recipe then.

jeepcook, 2020-10-06:
In fact I can paste the new driver here because they are committed in another image from another team (openvision), but I have to know where to find them in the repository directories. Could you help me for that ?
Thanks

nickersk, 2020-10-06:
https://github.com/OpenVisionE2/meta-octagon/blob/cd79fa1d7ca92a525365067f181e09a93ad6a11f/recipes-bsp/drivers/octagon-dvb-modules-hisi.inc#L12

nickersk, 2020-10-06:
seems like they are pli specific, if you can test them we can update URL.

atvcaptain, 2020-10-06:
yes its only for pli not for other teams dont use this drivers with oe-a

nickersk, 2020-10-06:
can we request updated drivers from octagon to use in oe-a?

atvcaptain, 2020-10-06:
i have send info to china ask for changes

jeepcook, 2020-10-06:
Hello, I can affirm that using the package containing the dvb modules drivers I have posted in the link in my first post here, solve the low speed zapping bug introduced by previous drivers. This part is not only for Pli. We have tested it on open at 6.4, 6.5 and 6.3. For the rest of the drivers we don't test them (lan, wifi....). I think we can commit this part, for the other parts we have to test.

jeepcook, 2020-10-06:
I compared 20200613 & 20200703 drivers, the only difference (size and date) is on a file named hisi-ir.ko. These file has no impact on the dvb modules, it concerns  an improvement of the IR Infrared receiving sensitivity. All other files are identical, the dvb module hi-dvb.ko has been update the 01 june 2020 and is on the 20200613 update (and 20200703). 
So I want to conclude that 20200613 drivers are in fact correct because identical as the ones dated from 03 july 2020 (except ir module) .
So I think the problem is not on the files content but perhaps on the load method, I don't know. Files /etc/modules-load.d/_sf8008.conf contains booth the correct module hi_media

The content of the 20200703 control file:
Package: octagon-dvb-modules-sf8008
Version: 4.4.35+20200703-r0
Description: Hardware drivers for sf8008
 Hardware drivers for sf8008.
Section: base
Priority: required
Maintainer: OE-Core Developers <openembedded-core@lists.openembedded.org>
License: Proprietary
Architecture: sf8008
OE: octagon-dvb-modules-sf8008
Source: octagon-dvb-modules-sf8008.bb

So I concluded that this package is integrated to oe-a. And I found in meta-oe-alliance/meta-brands/meta-octagon-recipes-drivers the file octagon-dvb-modules-sf8008.bb which depends to octagon-dvb-modules-hisi.inc. In this last file we have a correct url line 12 SRC_URI = "http://source.mynonpublic.com/octagon/${MACHINE}-hiko-${SRCDATE}.zip \

http://source.mynonpublic.com/octagon/sf8008-hiko-20200613.zip exists and http://source.mynonpublic.com/octagon/sf8008-hiko-20200703.zip doesn't exist.

So these package is not for Pli, but for all oe-a distros using this bb file.

Sorry a lot of blabla but these infos can help you, because I don't understand why 20200613 is buggy but identical to 20200703 (ir execpt). Finally perhaps the bug comes from remote speed command reception..

Regards.

jeepcook, 2020-10-09:
Do you think we can use/merge definesat os drivers to oe-a? Some are recents as the oe-a ones. Here is a link with drivers extracted from 20200927 DefineSat OS: http://www.mediafire.com/file/vvqdt3w9tu31qqz/sf8008_drivers_definesatos_20200927.zip/file

jeepcook, 2020-11-27:
This issue can be closed. Last drivers have been committed (https://github.com/oe-alliance/oe-alliance-core/blob/4.4/meta-brands/meta-octagon/recipes-drivers/octagon-dvb-modules-sf8008.bb) and no bug found with them, details of tests here: https://www.opena.tv/octagon-sf8008-4k-uhd/42924-status-treiber-und-features-stand-2-4-2019-a-58.html#post457297

closed 2020-11-27 by jeepcook