oe-alliance/oe-alliance-core #641: Error compile breach 4.4 ?!!

RR1717, 2020-12-11:
gcc (Ubuntu 10.2.0-5ubuntu1~20.04) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

5.4.0-56-generic

> | /home/build_image/OBH/build-enviroment/builds/openbh/release/dm920/tmp/hosttools/ld: .libs/libtirpc_la-netnamer.o:(.bss+0x8): multiple definition of `__svc_xports'; .libs/libtirpc_la-binddynport.o:(.bss+0x8): first defined here
| collect2: error: ld returned 1 exit status
| make[2]: *** [Makefile:548: libtirpc.la] Error 1
| make[2]: Leaving directory '/home/build_image/OBH/build-enviroment/builds/openbh/release/dm920/tmp/work/x86_64-linux/libtirpc-native/libtirpc-native-1.1.4-r0/build/src'
| make[1]: *** [Makefile:517: all-recursive] Error 1
| make[1]: Leaving directory '/home/build_image/OBH/build-enviroment/builds/openbh/release/dm920/tmp/work/x86_64-linux/libtirpc-native/libtirpc-native-1.1.4-r0/build'
| make: *** [Makefile:402: all] Error 2
| WARNING: /home/build_image/OBH/build-enviroment/builds/openbh/release/dm920/tmp/work/x86_64-linux/libtirpc-native/libtirpc-native-1.1.4-r0/temp/run.do_compile.155468:1 exit 1 from 'exit 1'
| 
ERROR: Task (virtual:native:/home/build_image/OBH/build-enviroment/openembedded-core/meta/recipes-extended/libtirpc/libtirpc_1.1.4.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 810 tasks of which 0 didn't need to be rerun and 1 failed.

>Summary: 1 task failed:
  virtual:native:/home/build_image/OBH/build-enviroment/openembedded-core/meta/recipes-extended/libtirpc/libtirpc_1.1.4.bb:do_compile
Summary: There were 41 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
make: *** [Makefile:1022: image] Error 1

nickersk, 2020-12-11:
https://github.com/openatv/enigma2/issues/1873

RR1717, 2020-12-11:
Already I try with gcc9.3 and same error ... I think I need to clean /tmp directory (I will try and give feedback) 

> $ gcc --version
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

> $ g++ --version
g++ (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

nickersk, 2020-12-11:
sure, you need to rebuild ...

RR1717, 2020-12-11:
I think it is ok ... build arrived to
> gcc-cross-arm-9.2.0-r0 do_compile
and built libtirpc-native just fine ..
Thank you

nickersk, 2020-12-11:
closing as solved.

closed 2020-12-11 by nickersk