oe-alliance/oe-alliance-core #642: samba config error branch 4.4 ?!!

RR1717, 2020-12-11:
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions. There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

> | Checking for library z                                                            : not found
| Checking for zlibVersion                                                          : not found
| Checking for library z                                                            : not found
| Checking for zlibVersion                                                          : not found
| ERROR: System library z of version 1.2.3 not found, and bundling disabled
| WARNING: /home/build_image/OBH/build-enviroment/builds/openbh/release/dm920/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/samba/samba-4.9.15-r3/temp/run.do_configure.3232541:1 exit 1 from '${CONFIG_CMD} --cross-answers="${CROSS_ANSWERS}"'
| 
ERROR: Task (/home/build_image/OBH/build-enviroment/meta-oe-alliance/meta-oe/recipes-connectivity/samba/samba.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 5247 tasks of which 4956 didn't need to be rerun and 1 failed.

nickersk, 2020-12-11:
have you installed all openembedded prerequisities?

RR1717, 2020-12-11:
Of course as It has been suggested here ...
https://github.com/openatv/enigma2

> $ sudo apt-get install -y autoconf automake bison bzip2 chrpath coreutils cpio curl cvs debianutils default-jre default-jre-headless diffstat flex g++ gawk gcc gcc-8 gettext git git-core gzip help2man info iputils-ping java-common libc6-dev libegl1-mesa libglib2.0-dev libncurses5-dev libperl4-corelibs-perl libproc-processtable-perl libsdl1.2-dev libserf-dev libtool libxml2-utils make ncurses-bin patch perl pkg-config psmisc python3 python3-git python3-jinja2 python3-pexpect python3-pip python-setuptools qemu quilt socat sshpass subversion tar texi2html texinfo unzip wget xsltproc xterm xz-utils zip zlib1g-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'git' instead of 'git-core'
autoconf is already the newest version (2.69-11.1).
automake is already the newest version (1:1.16.1-4ubuntu6).
bison is already the newest version (2:3.5.1+dfsg-1).
bzip2 is already the newest version (1.0.8-2).
coreutils is already the newest version (8.30-3ubuntu2).
cpio is already the newest version (2.13+dfsg-2).
debianutils is already the newest version (4.9.1).
default-jre is already the newest version (2:1.11-72).
default-jre-headless is already the newest version (2:1.11-72).
diffstat is already the newest version (1.63-1).
flex is already the newest version (2.6.4-6.2).
g++ is already the newest version (4:9.3.0-1ubuntu2).
gawk is already the newest version (1:5.0.1+dfsg-1).
gcc is already the newest version (4:9.3.0-1ubuntu2).
gettext is already the newest version (0.19.8.1-10build1).
git is already the newest version (1:2.25.1-1ubuntu3).
gzip is already the newest version (1.10-0ubuntu4).
info is already the newest version (6.7.0.dfsg.2-5).
iputils-ping is already the newest version (3:20190709-3).
java-common is already the newest version (0.72).
libncurses5-dev is already the newest version (6.2-0ubuntu2).
libperl4-corelibs-perl is already the newest version (0.004-2).
libtool is already the newest version (2.4.6-14).
libxml2-utils is already the newest version (2.9.10+dfsg-5).
make is already the newest version (4.2.1-1.2).
ncurses-bin is already the newest version (6.2-0ubuntu2).
patch is already the newest version (2.7.6-6).
pkg-config is already the newest version (0.29.1-0ubuntu4).
psmisc is already the newest version (23.3-1).
python3 is already the newest version (3.8.2-0ubuntu2).
python3-jinja2 is already the newest version (2.10.1-2).
python3-pexpect is already the newest version (4.6.0-1build1).
socat is already the newest version (1.7.3.3-2).
tar is already the newest version (1.30+dfsg-7).
unzip is already the newest version (6.0-25ubuntu1).
wget is already the newest version (1.20.3-1ubuntu1).
xsltproc is already the newest version (1.1.34-4).
zip is already the newest version (3.0-11build1).
chrpath is already the newest version (0.16-2).
cvs is already the newest version (2:1.12.13+real-27build1).
gcc-8 is already the newest version (8.4.0-3ubuntu2).
help2man is already the newest version (1.47.13).
libproc-processtable-perl is already the newest version (0.59-2).
libsdl1.2-dev is already the newest version (1.2.15+dfsg2-5).
libserf-dev is already the newest version (1.3.9-8build1).
python-setuptools is already the newest version (44.0.0-2).
python3-git is already the newest version (3.0.7-1).
quilt is already the newest version (0.65-3).
sshpass is already the newest version (1.06-1).
subversion is already the newest version (1.13.0-3).
texi2html is already the newest version (1.82+dfsg1-5).
texinfo is already the newest version (6.7.0.dfsg.2-5).
xterm is already the newest version (353-1ubuntu1).
curl is already the newest version (7.68.0-1ubuntu2.4).
libc6-dev is already the newest version (2.31-0ubuntu9.1).
libglib2.0-dev is already the newest version (2.64.3-1~ubuntu20.04.1).
perl is already the newest version (5.30.0-9ubuntu0.2).
qemu is already the newest version (1:4.2-3ubuntu6.10).
xz-utils is already the newest version (5.2.4-1ubuntu1).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-2ubuntu1.2).
libegl1-mesa is already the newest version (20.0.8-0ubuntu1~20.04.1).
python3-pip is already the newest version (20.0.2-5ubuntu1.1).
0 upgraded, 0 newly installed, 0 to remove and 9 not upgraded.

nickersk, 2020-12-11:
enigma2 is gui for stb, openembedded is build system https://docs.yoctoproject.org/brief-yoctoprojectqs/brief-yoctoprojectqs.html#build-host-packages you also need much ram, fast cpus and diskspace to prevent build being killed...
closing because of too few infos.

closed 2020-12-11 by nickersk