oe-alliance/oe-alliance-core #643: Build OverlayHD skin from the git

persianpros, 2020-12-20:
Hello,

Here's the new home of @IanSav skin: https://github.com/IanSav/OverlayHD

We have a ready bitbake recpie for it: https://github.com/OpenVisionE2/openvision-development-platform/blob/develop/meta-openvision/recipes-openvision/enigma2-skins/enigma2-plugin-skins-overlayhd.bb

Which you can adapt for your OE.

Just to let you know :+1: 

Regards,
Persian Prince

Huevos, 2020-12-20:
If Ian wants to add this he is welcome to make a pull request.

persianpros, 2020-12-20:
You have it as a pre-compiled ipk in https://github.com/oe-alliance/3rdparty-plugins and I think Ian isn't so familiar with bitbake so better to add the recipe yourself as you have the build and could check what's needed in OE-A

jbleyel, 2021-06-08:
https://github.com/OpenVisionE2/openvision-development-platform/blob/develop/meta-openvision/recipes-openvision/enigma2-skins/enigma2-plugin-skins-overlayhd.bb -> Error 404

closed 2021-06-08 by jbleyel