oe-alliance/oe-alliance-core #644: kodiext from OpenSPA team

rdcalle, 2020-12-30:
How about using the https://github.com/OpenSPA/kodiext instead the one got from https://github.com/mx3L/kodiext?
The OpenSPA team is doing a great job with it.

atvcaptain, 2020-12-30:
but openspa not share the infos and send back the updates to oe-a,

rdcalle, 2020-12-30:
Well, they shared all about the Kodi stuff a few time ago. And kodiext is a
plugin to be used by Kodi.

Wouldn't that be a good idea if it's more updated that the other one?

atvcaptain, 2020-12-30:
need kodi a update too for this change , openspa kodi git not public

OpenSPA, 2021-01-02:
Hello, our kodiext git is public and everybody can use it. All kodi development is shared as well, but kodiext and kodi are not the same.

GIT for kodiext : https://github.com/OpenSPA/kodiext

and this converter is needed for kodiext: https://github.com/OpenSPA/dvbapp2/blob/master/lib/python/Components/Converter/TrackInfo.py

ahmedmoselhi, 2021-01-15:
> Hello, our kodiext git is public and everybody can use it. All kodi development is shared as well, but kodiext and kodi are not the same.
>
> GIT for kodiext : https://github.com/OpenSPA/kodiext
>
> and [...]

why not sharing spaze menu and plugins too?? @OpenSPA

atvcaptain, 2021-04-05:
now merge to oe-a thx for the info

closed 2021-04-05 by atvcaptain