oe-alliance/oe-alliance-core #652: libssl and libcrypto missing shortcuts

jeepcook, 2021-02-27:
Hello,

On some programs, libssl.so.0.9.8 and libcrypto.so.0.9.8 are needed and unable to mod this value into the program.

1.0.2 versions for this 2 libraries are linked to 0.9.7 and 1.0.0 versions. So I think 0.9.8 is frequently used, so a shortcut to this version has to be added.

```
cd /usr/lib
ln -sf libssl.so.1.0.2 libssl.so.0.9.8
ln -sf libcrypto.so.1.0.2 libcrypto.so.0.9.8
```

Thanks.

Regards

AbuBaniaz, 2021-02-27:
This was brought up a few years ago. There are a few odd occasions when the 0.98 itself is required, this is why the symlink was not added.

Ideally, the people who made the oscam-emu patched binary for tntsat can amend it so it does not depend on the old version of libssl existing or a symlink. It obviously works on newer versions anyway. Better still add the patch to the oscam-emu directly.

jeepcook, 2021-02-27:
Thanks for the answer. I'm ok with you.

AbuBaniaz, 2021-02-27:
Also, I think there is a libcrypto compat package defined which will add the 0.98 symlink.

AbuBaniaz, 2021-02-27:
https://github.com/oe-alliance/oe-alliance-core/blob/4.4/meta-oe/recipes-connectivity/openssl/libcrypto-compat-0.9.8.bb

Not too sure if it is built.

jeepcook, 2021-02-27:
Yes probably, I will look at the content of compat package and find the good version

jeepcook, 2021-02-27:
No I's not build. Test with my own oatv 6.4 compilation and the official one, this link isn't build

AbuBaniaz, 2021-02-27:
Try libxcrypt-compat

Otherwise 0.98 will need to be added here if it does not clash with anything
https://github.com/oe-alliance/oe-alliance-core/blob/4.4/meta-oe/recipes-oe-alliance/image/oe-alliance-base.bb#L43:L45

jeepcook, 2021-02-27:
libxcrypt-compat is not a package, we only have these already installed packages:

libcrypto-compat-0.9.7 - 1.0-r0 - Compatibility for packages that link to libcrypto or libssl 0.9.7
libcrypto-compat-1.0.0 - 1.1-r0 - Compatibility for packages that link to libcrypto or libssl 1.0.0

jeepcook, 2021-02-28:
Sorry made a mistake in my previous post, I edited it.

jeepcook, 2021-03-02:
Hello, 
To implement this mod, I have to use nextp3 as dev 4.4 branch ?
Thanks

atvcaptain, 2022-01-08:
openssl-old_0.9.8x  build correct  libs we cant add the link to default image, better add depends to cams ipk

closed 2022-01-08 by atvcaptain

jeepcook, 2022-01-08:
OK with closure. It's an old demand.