oe-alliance/oe-alliance-core #661: hd51 Bad mounts on new image startup

jeepcook, 2021-04-08:
Hello,
When I install a new image on a mutant hd51, the default mount are bad:

internal hdd --> /media/hdd
external hdd (usb3 port) --> /media/hdd
external usb drive (usb2 rear port) --> /media/usb

So we have 2 mounts on /media/hdd. My settings backups are on internal hdd (/media/hdd) and when I install a new image with fast restore option the system can't fount backups because instead of pointing to internal hdd, the /media/hdd mount is pointing to external hdd.

The fix is to automount drives but only one with /media/hdd mount point and logically for the internal hdd.

atvcaptain, 2022-01-08:
this script mount the hdd

you must enable log in the script  and check the output

https://github.com/oe-alliance/oe-alliance-core/blob/5.0/meta-oe/recipes-core/base-files/base-files/mount-helper.sh

remove the # from line 3 and 11-28

closed 2023-05-16 by atvcaptain

jeepcook, 2023-08-08:
Hello,
Really sorry I never answered to you. 
In fact now I don't have anymore a hd51 box but an Octagon SF8008.
I opened an issue in oatv github with a similar case but I'm not sure if it's an issue for oa-tv team or oe-a team?
[https://github.com/openatv/enigma2/issues/2989](url)
What do you thing about?
Thanks