oe-alliance/oe-alliance-core #674: [OSMIO4K] Kernel Update through package manager fails

Fighter19, 2021-05-03:
This night I tried updating my osmio4k through the UI.

The Web interface doesn't come up, but I can still use SSH.
When using opkg to reinstall the new kernel, I get notified that zImage can not be found.

```
Reinstalling kernel-image (5.12-4) on root.
Installing kernel-image (5.12) on root.
Configuring kernel-image.
/bin/sh: udevadm: command not found
dd: can't open '/tmp/zImage': No such file or directory
Collected errors:
 * pkg_run_script: package "kernel-image" postinst script returned status 1.
 * opkg_configure: kernel-image.postinst returned 1.
```

The postinstall script inside the ipk looks different from the one eidision is having in their repo.

https://github.com/oe-alliance/oe-alliance-core/blob/cb6ff7f271e57d872e7ecba1bdb37ba89f791ded/meta-brands/meta-edision/recipes-linux/linux-edision_5.12.bb#L62

https://github.com/edision-open/meta-edision/blob/9a4ede619ab90cafe6de210fb81bdd287a135f14/recipes-kernel/linux/linux-edision_5.12.bb#L37

I suspect the line is supposed to read 
```
dd if=/tmp/zImage of=/dev/kernel
```
instead of
```
dd if=/tmp/zImage-5.12.0 of=/dev/kernel 
```
Also I have no /dev/kernel device so this is probably the wrong routine to begin with.

Huevos, 2021-05-04:
What image are you using?

Fighter19, 2021-05-04:
OpenATV 6.15 I don't know the exact that, but I think it was from January.

Huevos, 2021-05-05:
If OpenWebInterface is not loading you should see some output in the debug log.

atvcaptain, 2022-01-08:
fixed with 7.0 
https://github.com/oe-alliance/oe-alliance-core/commit/29d12c21ccb7184a3e7d0bcf45a6d4e10fcc9267

closed 2022-01-08 by atvcaptain