oe-alliance/oe-alliance-core #677: Unify Vu+ support for kodi

andrea-adami, 2021-05-28:
Hi,
please check out the new Vu+ packages providing kodi support.

https://github.com/vu-plus/meta-vuplus/commit/fff776af1459b3968092b0c3b99ef9f8d2237d14
https://github.com/OpenPLi/openpli-oe-core/pull/989

Please catch up so that we can finally have an unified base for these out-of-sale (now  obsolete?) devices.
Thanks
A.A.

P.S.
It would be wondefulif while unifying the branding stuff we could have a shared defconfig for the kernels of these outdated machines.

andrea-adami, 2021-09-26:
Update: support package added for meta-gigablue as well:
https://github.com/zhtq/meta-gigablue/pull/10

I am trying to clean the kodi recipes and have just in the BSP something like
 EXTRA_OECMAKE_append_pn-kodi = " -DWITH_V3D=nxpl"

You did it differently, with many recipes spread in the BSP's. Your choice.

At the moment I have grouped all this stuff in PLi's kodi_18.inc [1] but will try to put all in the BSP's.
I think we should have matching definitions, not vuplus-mipsel vs vumips...and so on.
Your patchset supports more devices so the one in PLi will be a sub-ensemble.

Cheers
A.A.

[1] https://github.com/OpenPLi/openpli-oe-core/blob/develop/meta-openpli/recipes-mediacenter/kodi/kodi_18.inc

andrea-adami, 2021-10-06:
Update:
I have sent these renaming-patches to the BSP maintainers.
Some BSP layers are already updated, some do not respond yet.

I hope this can help to unify the external layers and maybe the branding.
Cheers
A.A.

atvcaptain, 2021-10-07:
at the moment i am busy no free time, i have test all models with kodi and dont forget we have more models as pli and have kodi 19 and 20 ready

atvcaptain, 2023-05-16:
oe-a support more models not possible unify the support

closed 2023-05-16 by atvcaptain