oe-alliance/oe-alliance-core #681: building openatv error...

aouamed, 2021-06-05:
build openatv branch nextp3-ssl111 
ubuntu 20.04
vuuno4kse
Error :

DEBUG: Executing python function autotools_aclocals
DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
DEBUG: Python function autotools_aclocals finished
DEBUG: Executing shell function do_compile
NOTE: make -j 8
make: *** No targets specified and no makefile found.  Stop.
ERROR: oe_runmake failed
WARNING: exit code 1 from a shell command.
ERROR: Execution of '/home/mido/Bureau/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/vuuno4kse-oe-linux-gnueabi/oe-alliance-branding/7.0+gitAUTOINC+844a061ece-r20210605-vuuno4kse/temp/run.do_compile.1111011' failed with exit code 1:
make: *** No targets specified and no makefile found.  Stop.
WARNING: exit code 1 from a shell command.

nickersk, 2021-06-05:
hi, remove tmp/ from build dir and rerun build.

aouamed, 2021-06-05:
ok..
Thank you for your answer

jbleyel, 2021-06-05:
Please close this issue if the build runs again.

aouamed, 2021-06-05:
other error samba : up to 99/100 

ERROR: QA Issue: /usr/bin/regtree contained in package registry-tools requires libcmdline-credentials-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_registry-tools? [file-rdeps]
ERROR: QA Issue: /usr/bin/regtree contained in package registry-tools requires libregistry-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_registry-tools? [file-rdeps]
ERROR: QA Issue: /usr/bin/regtree contained in package registry-tools requires libevents-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_registry-tools? [file-rdeps]
ERROR: QA Issue: /usr/bin/regtree contained in package registry-tools requires libsamba-debug-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_registry-tools? [file-rdeps]
ERROR: QA Issue: /usr/bin/regtree contained in package registry-tools requires libregistry-samba4.so, but no providers found in RDEPENDS_registry-tools? [file-rdeps]
ERROR: QA Issue: /usr/bin/regtree contained in package registry-tools requires libcmdline-credentials-samba4.so, but no providers found in RDEPENDS_registry-tools? [file-rdeps]
ERROR: QA Issue: /usr/bin/regtree contained in package registry-tools requires libevents-samba4.so, but no providers found in RDEPENDS_registry-tools? [file-rdeps]
ERROR: QA Issue: /usr/bin/regtree contained in package registry-tools requires libsamba-debug-samba4.so, but no providers found in RDEPENDS_registry-tools? [file-rdeps]
ERROR: QA Issue: /usr/bin/regshell contained in package registry-tools requires libreplace-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_registry-tools? [file-rdeps]
ERROR: QA Issue: /usr/bin/regshell contained in package registry-tools requires libsamba-security-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_registry-tools? [file-rdeps]
ERROR: QA Issue: /usr/bin/regshell contained in package registry-tools requires libutil-reg-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_registry-tools? [file-rdeps]
ERROR: QA Issue: /usr/bin/regshell contained in package registry-tools requires libsamba-security-samba4.so, but no providers found in RDEPENDS_registry-tools? [file-rdeps]
ERROR: QA Issue: /usr/bin/regshell contained in package registry-tools requires libreplace-samba4.so, but no providers found in RDEPENDS_registry-tools? [file-rdeps]
ERROR: QA Issue: /usr/bin/regshell contained in package registry-tools requires libutil-reg-samba4.so, but no providers found in RDEPENDS_registry-tools? [file-rdeps]
NOTE: Package libndr-krb5pac skipping QA tests: {'dev-so'}
NOTE: Checking Package: libndr-krb5pac
NOTE: arm-oe-linux-gnueabi-objdump -p /home/mido/Bureau/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/samba/4.14.4-r2/packages-split/libndr-krb5pac/usr/lib/libndr-krb5pac.so.0.0.1
ERROR: QA Issue: /usr/lib/libndr-krb5pac.so.0.0.1 contained in package libndr-krb5pac requires libsamba-security-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_libndr-krb5pac? [file-rdeps]
ERROR: QA Issue: /usr/lib/libndr-krb5pac.so.0.0.1 contained in package libndr-krb5pac requires libsamba-security-samba4.so, but no providers found in RDEPENDS_libndr-krb5pac? [file-rdeps]
NOTE: Package samba-doc skipping QA tests: {'dev-so'}
NOTE: Checking Package: samba-doc
NOTE: Package libsamba-errors skipping QA tests: {'dev-so'}
NOTE: Checking Package: libsamba-errors
NOTE: arm-oe-linux-gnueabi-objdump -p /home/mido/Bureau/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/samba/4.14.4-r2/packages-split/libsamba-errors/usr/lib/libsamba-errors.so.1
NOTE: Package pam-pluginwinbind skipping QA tests: {'dev-so'}
NOTE: Checking Package: pam-pluginwinbind
NOTE: arm-oe-linux-gnueabi-objdump -p /home/mido/Bureau/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/samba/4.14.4-r2/packages-split/pam-pluginwinbind/lib/security/pam_winbind.so
ERROR: QA Issue: /lib/security/pam_winbind.so contained in package pam-pluginwinbind requires libreplace-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_pam-pluginwinbind? [file-rdeps]
ERROR: QA Issue: /lib/security/pam_winbind.so contained in package pam-pluginwinbind requires libreplace-samba4.so, but no providers found in RDEPENDS_pam-pluginwinbind? [file-rdeps]
NOTE: Package samba-ctdb-tests skipping QA tests: {'dev-so'}
NOTE: Checking Package: samba-ctdb-tests

ERROR: QA Issue: /usr/lib/libsamba-util.so.0.0.1 contained in package libsamba-util requires libtevent.so.0, but no providers found in RDEPENDS_libsamba-util? [file-rdeps]
ERROR: QA Issue: /usr/lib/libsamba-util.so.0.0.1 contained in package libsamba-util requires libsamba-debug-samba4.so, but no providers found in RDEPENDS_libsamba-util? [file-rdeps]
ERROR: QA Issue: /usr/lib/libsamba-util.so.0.0.1 contained in package libsamba-util requires libsys-rw-samba4.so, but no providers found in RDEPENDS_libsamba-util? [file-rdeps]
ERROR: QA Issue: /usr/lib/libsamba-util.so.0.0.1 contained in package libsamba-util requires libtime-basic-samba4.so, but no providers found in RDEPENDS_libsamba-util? [file-rdeps]
ERROR: QA Issue: /usr/lib/libsamba-util.so.0.0.1 contained in package libsamba-util requires libreplace-samba4.so, but no providers found in RDEPENDS_libsamba-util? [file-rdeps]
NOTE: Package samba-python3 skipping QA tests: {'dev-so'}
NOTE: Checking Package: samba-python3
NOTE: Package samba-pidl skipping QA tests: {'dev-so'}
NOTE: Checking Package: samba-pidl
ERROR: QA run found fatal errors. Please consider fixing them.
DEBUG: Python function do_package_qa finished

nickersk, 2021-06-05:
everything ok here ...

aouamed, 2021-06-05:
the build Do not finish...
So where does the problem come from... !?

nickersk, 2021-06-05:
finished build for 4 machines ok here. wipe out tmp and retry...

aouamed, 2021-06-05:
Ok I delete tmp and try again ...
thanks

aouamed, 2021-06-05:
same error : 

DEBUG: Executing python function sstate_task_prefunc
DEBUG: Python function sstate_task_prefunc finished
DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies are ['/home/mido/Bureau/build-enviroment/openembedded-core/meta/recipes-devtools/binutils/binutils-cross_2.36.bb:do_populate_sysroot', 'virtual:native:/home/mido/Bureau/build-enviroment/openembedded-core/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot']
NOTE: Installed into sysroot: []
NOTE: Skipping as already exists in sysroot: ['binutils-cross-arm', 'pseudo-native', 'gnu-config-native', 'quilt-native', 'zlib-native', 'autoconf-native', 'flex-native', 'texinfo-dummy-native', 'xz-native', 'm4-native', 'gettext-minimal-native', 'automake-native', 'libtool-native']
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing python function do_package_qa
NOTE: DO PACKAGE QA
DEBUG: Executing python function read_subpackage_metadata
DEBUG: Python function read_subpackage_metadata finished
NOTE: Package samba-utils skipping QA tests: {'dev-so'}
NOTE: Checking Package: samba-utils
NOTE: arm-oe-linux-gnueabi-objdump -p /home/mido/Bureau/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/samba/4.14.4-r2/packages-split/samba-utils/usr/bin/smbspool
NOTE: arm-oe-linux-gnueabi-objdump -p /home/mido/Bureau/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/samba/4.14.4-r2/packages-split/samba-utils/usr/bin/smbtree
NOTE: arm-oe-linux-gnueabi-objdump -p /home/mido/Bureau/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/samba/4.14.4-r2/packages-split/samba-utils/usr/bin/mvxattr
NOTE: arm-oe-linux-gnueabi-objdump -p /home/mido/Bureau/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/samba/4.14.4-r2/packages-split/samba-utils/usr/bin/nmblookup
NOTE: arm-oe-linux-gnueabi-objdump -p /home/mido/Bureau/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/samba/4.14.4-r2/packages-split/samba-utils/usr/bin/cifsdd
NOTE: arm-oe-linux-gnueabi-objdump -p /home/mido/Bureau/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/samba/4.14.4-r2/packages-split/samba-utils/usr/bin/smbget
ERROR: QA Issue: /usr/bin/cifsdd contained in package samba-utils requires libcli-ldap-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_samba-utils? [file-rdeps]
ERROR: QA Issue: /usr/bin/cifsdd contained in package samba-utils requires libcliauth-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_samba-utils? [file-rdeps]
ERROR: QA Issue: /usr/bin/cifsdd contained in package samba-utils requires libcli-smb-common-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_samba-utils? [file-rdeps]
ERROR: QA Issue: /usr/bin/smbtree contained in package samba-utils requires libsamba-debug-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_samba-utils? [file-rdeps]
ERROR: QA Issue: /usr/bin/cifsdd contained in package samba-utils requires libcmdline-credentials-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_samba-utils? [file-rdeps]
ERROR: QA Issue: /usr/bin/cifsdd contained in package samba-utils requires libcli-cldap-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_samba-utils? [file-rdeps]
ERROR: QA Issue: /usr/bin/cifsdd contained in package samba-utils requires libgensec-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_samba-utils? [file-rdeps]
ERROR: QA Issue: /usr/bin/cifsdd contained in package samba-utils requires libsamba-security-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_samba-utils? [file-rdeps]
ERROR: QA Issue: /usr/bin/cifsdd contained in package samba-utils requires libevents-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_samba-utils? [file-rdeps]
ERROR: QA Issue: /usr/bin/nmblookup contained in package samba-utils requires libtevent.so.0(TEVENT_0.9.9), but no providers found in RDEPENDS_samba-utils? [file-rdeps]
ERROR: QA Issue: /usr/bin/smbtree contained in package samba-utils requires libreplace-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_samba-utils? [file-rdeps]
ERROR: QA Issue: /usr/bin/nmblookup contained in package samba-utils requires libsamba-sockets-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_samba-utils? [file-rdeps]
ERROR: QA Issue: /usr/bin/nmblookup contained in package samba-utils requires libtevent-util.so.0(SAMBA_4.14.4), but no providers found in RDEPENDS_samba-utils? [file-rdeps]
ERROR: QA Issue: /usr/bin/cifsdd contained in package samba-utils requires libsmbclient-raw-samba4.so(SAMBA_4.14.4), but no providers found in RDEPENDS_samba-utils? [file-rdeps]
ERROR: QA Issue: /usr/bin/cifsdd contained in package samba-utils requires libevents-samba4.so, but no providers found in RDEPENDS_samba-utils? [file-rdeps]
ERROR: QA Issue: /usr/bin/cifsdd contained in package samba-utils requires libcliauth-samba4.so, but no providers found in RDEPENDS_samba-utils? [file-rdeps]
ERROR: QA Issue: /usr/bin/nmblookup contained in package samba-utils requires libtevent.so.0, but no providers found in RDEPENDS_samba-utils? [file-rdeps]
ERROR: QA Issue: /usr/bin/cifsdd contained in package samba-utils requires libsamba-security-samba4.so, but no providers found in RDEPENDS_samba-utils? [file-rdeps]
[... cut ...]
ERROR: QA Issue: /usr/bin/ndrdump contained in package samba-testsuite requires libndr-samba4.so, but no providers found in RDEPENDS_samba-testsuite? [file-rdeps]
NOTE: Package libdcerpc-samr skipping QA tests: {'dev-so'}
NOTE: Checking Package: libdcerpc-samr
NOTE: arm-oe-linux-gnueabi-objdump -p /home/mido/Bureau/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/samba/4.14.4-r2/packages-split/libdcerpc-samr/usr/lib/libdcerpc-samr.so.0.0.1
ERROR: QA run found fatal errors. Please consider fixing them.
DEBUG: Python function do_package_qa finished

jbleyel, 2021-06-05:
I will check build vuuno4kse on my build machine.

aouamed, 2021-06-05:
you check build vuuno4kse and you give me results ...
thanks

jbleyel, 2021-06-05:
samba-4.14.4-r2 
do_compile -> Success
do_install -> Success

jbleyel, 2021-06-05:
please zip this folder 

../openatv/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/samba/4.14.4-r2/temp

and put the zip here

aouamed, 2021-06-05:
my floder is : 
build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/cortexa15hf-neon-vfpv4-oe-linux-gnueabi/samba/4.14.4-r2/temp

https://www.mediafire.com/file/h9tblyp6s0a894s/temp.zip/file

jbleyel, 2021-06-06:
Do you use local bb's / classes / configs?
What's your build server linux version?

aouamed, 2021-06-06:
No class configs of local bb ...I do not change the file of the OpenATV image 

ubuntu 20.04 desktop (64-bit) latest update

aouamed, 2021-06-06:
BB_VERSION           = "1.51.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "arm-oe-linux-gnueabi"
MACHINE              = "vuuno4kse"
DISTRO               = "openatv"
DISTRO_VERSION       = "7.0"
TUNE_FEATURES        = "arm vfp cortexa15 neon vfpv4 callconvention-hard"
TARGET_FPU           = "hard"
meta                 = "HEAD:edaf8ff278fc96b122c4fc3266b63856e3350f4c"
meta-oe              
meta-multimedia      
meta-networking      
meta-filesystems     
meta-python          
meta-webserver       = "HEAD:f3c21a3fe2f1026acb67cd0f904acb5a911eb7b2"
meta-oe              
meta-vuplus          = "HEAD:f4474421975c3abca42128f46d7124b239a843f2"

aouamed, 2021-06-09:
same probleme with last update branch 5.0 samba 4.14.5...

atvcaptain, 2021-06-10:
you have follow the readme to Prepare your Build Host
https://github.com/openatv/enigma2/tree/7.0#readme

in 2-3h vuuno4kse is ready i build now too

i have start clean and first builds vu ultimo4k and vusolo4l was ok

andrea-adami, 2021-06-10:
There is something wrong in the OE-A samba recipe, it failed on my build and even before.
Now another report.

If I can I'll give it again one try here on Ubuntu 18.04.
Cheers
A.A.

https://github.com/oe-alliance/build-enviroment/issues/46
https://github.com/oe-alliance/oe-alliance-core/issues/592

nickersk, 2021-06-10:
put file-rdeps to insane skip

aouamed, 2021-06-10:
I followed all the instruction and other mistake ... 

DEBUG: Executing python function sstate_task_prefunc
DEBUG: Removing manifest: /home/mido/Documents/build-enviroment/builds/openatv/release/vuuno4kse/tmp/deploy/ipk/vuuno4kse/oe-alliance-branding_7.0+git1014+844a061-r20210610-vuuno4kse_vuuno4kse.ipk
DEBUG: Removing manifest: /home/mido/Documents/build-enviroment/builds/openatv/release/vuuno4kse/tmp/deploy/ipk/vuuno4kse/oe-alliance-branding-dbg_7.0+git1014+844a061-r20210610-vuuno4kse_vuuno4kse.ipk
DEBUG: Removing manifest: /home/mido/Documents/build-enviroment/builds/openatv/release/vuuno4kse/tmp/deploy/ipk/vuuno4kse/oe-alliance-branding-dev_7.0+git1014+844a061-r20210610-vuuno4kse_vuuno4kse.ipk
DEBUG: Removing manifest: /home/mido/Documents/build-enviroment/builds/openatv/release/vuuno4kse/tmp/deploy/ipk/vuuno4kse/
DEBUG: Python function sstate_task_prefunc finished
DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies are ['virtual:native:/home/mido/Documents/build-enviroment/openembedded-core/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/mido/Documents/build-enviroment/openembedded-core/meta/recipes-extended/xz/xz_5.2.5.bb:do_populate_sysroot', 'virtual:native:/home/mido/Documents/build-enviroment/openembedded-core/meta/recipes-devtools/opkg-utils/opkg-utils_0.4.3.bb:do_populate_sysroot']
NOTE: Installed into sysroot: []
NOTE: Skipping as already exists in sysroot: ['pseudo-native', 'xz-native', 'opkg-utils-native', 'quilt-native', 'libtool-native', 'autoconf-native', 'gettext-minimal-native', 'automake-native', 'perl-native', 'texinfo-dummy-native', 'gnu-config-native', 'm4-native', 'perlcross-native', 'db-native', 'zlib-native', 'gdbm-native']
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing python function do_package_write_ipk
DEBUG: Executing python function read_subpackage_metadata
DEBUG: Python function read_subpackage_metadata finished
DEBUG: Executing python function do_package_ipk
NOTE: package name mapping done: glibc -> libc6
NOTE: Not creating empty archive for oe-alliance-branding-locale-7.0+git1014+844a061-r20210610-vuuno4kse
NOTE: Not creating empty archive for oe-alliance-branding-doc-7.0+git1014+844a061-r20210610-vuuno4kse
NOTE: libc6-dbg rprovides glibc-dbg, not replacing the latter
NOTE: Not creating empty archive for oe-alliance-branding-src-7.0+git1014+844a061-r20210610-vuuno4kse
NOTE: libc6-dev rprovides glibc-dev, not replacing the latter
NOTE: Not creating empty archive for oe-alliance-branding-staticdev-7.0+git1014+844a061-r20210610-vuuno4kse
DEBUG: Python function do_package_ipk finished
DEBUG: Python function do_package_write_ipk finished
DEBUG: Executing python function sstate_task_postfunc
NOTE: Using umask 0o002 (not 22) for sstate packaging
DEBUG: Preparing tree /home/mido/Documents/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/vuuno4kse-oe-linux-gnueabi/oe-alliance-branding/7.0+gitAUTOINC+844a061ece-r20210610-vuuno4kse/deploy-ipks for packaging at /home/mido/Documents/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/vuuno4kse-oe-linux-gnueabi/oe-alliance-branding/7.0+gitAUTOINC+844a061ece-r20210610-vuuno4kse/sstate-build-package_write_ipk/deploy-ipks
DEBUG: Executing python function sstate_hardcode_path
NOTE: Removing hardcoded paths from sstate package: 'find /home/mido/Documents/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/vuuno4kse-oe-linux-gnueabi/oe-alliance-branding/7.0+gitAUTOINC+844a061ece-r20210610-vuuno4kse/sstate-build-package_write_ipk/ \( -name "*.la" -o -name "*-config" -o -name "*_config" -o -name "postinst-*" \) -type f | xargs grep -l -e '/home/mido/Documents/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/vuuno4kse-oe-linux-gnueabi/oe-alliance-branding/7.0+gitAUTOINC+844a061ece-r20210610-vuuno4kse/recipe-sysroot' -e '/home/mido/Documents/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/vuuno4kse-oe-linux-gnueabi/oe-alliance-branding/7.0+gitAUTOINC+844a061ece-r20210610-vuuno4kse/recipe-sysroot-native' -e '/home/mido/Documents/build-enviroment/builds/openatv/release/vuuno4kse/tmp/hosttools' | tee /home/mido/Documents/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/vuuno4kse-oe-linux-gnueabi/oe-alliance-branding/7.0+gitAUTOINC+844a061ece-r20210610-vuuno4kse/sstate-build-package_write_ipk/fixmepath | xargs --no-run-if-empty sed -i -e 's:/home/mido/Documents/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/vuuno4kse-oe-linux-gnueabi/oe-alliance-branding/7.0+gitAUTOINC+844a061ece-r20210610-vuuno4kse/recipe-sysroot-native:FIXMESTAGINGDIRHOST:g' -e 's:/home/mido/Documents/build-enviroment/builds/openatv/release/vuuno4kse/tmp/work/vuuno4kse-oe-linux-gnueabi/oe-alliance-branding/7.0+gitAUTOINC+844a061ece-r20210610-vuuno4kse/recipe-sysroot:FIXMESTAGINGDIRTARGET:g' -e 's:/home/mido/Documents/build-enviroment/builds/openatv/release/vuuno4kse/tmp/hosttools:FIXME_HOSTTOOLS_DIR:g''
DEBUG: Python function sstate_hardcode_path finished
DEBUG: Executing python function sstate_report_unihash
DEBUG: Python function sstate_report_unihash finished
DEBUG: Executing shell function sstate_create_package
DEBUG: Shell function sstate_create_package finished
ERROR: Taskhash mismatch ae415ab22ac2212d9eb3f686a6e0e78c8a4c3d1ef37a8868bb04cb1ec395963a versus 3bfb4789235f4314f067c198fa28c2c112c24c6b3c5eff983581f2970a13e569 for /home/mido/Documents/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/ima [... cut ...] kse/deploy-ipks -p --no-recursion --files-from - | tar --xattrs --xattrs-include='*' -xhf - -C /home/mido/Documents/build-enviroment/builds/openatv/release/vuuno4kse/tmp/deploy/ipk' returned non-zero exit status 2.

Subprocess output:
tar: ./vuuno4kse: Cannot change ownership to uid 0, gid 0: Operation not permitted
tar: .: Cannot change ownership to uid 0, gid 0: Operation not permitted
tar: Exiting with failure status due to previous errors

DEBUG: Python function sstate_task_postfunc finished

aouamed, 2021-06-10:
nickersk 
Want to say :
INSANE_SKIP_${PN} += "file-rdeps"
i put... but the same problem

nickersk, 2021-06-10:
wipe out tmp and feed dir and run again

aouamed, 2021-06-10:
https://i.postimg.cc/8PjPz4hP/Capture-d-cran-du-2021-06-10-13-23-11.png
https://i.postimg.cc/5yGJzcw9/Capture-d-cran-du-2021-06-10-13-25-12.png
https://i.postimg.cc/QMs31JW7/Capture-d-cran-du-2021-06-10-13-33-05.png
https://i.postimg.cc/3wBHB0gc/Capture-d-cran-du-2021-06-10-13-37-37.png
https://i.postimg.cc/QxTGzyQ5/Capture-d-cran-du-2021-06-10-13-41-53.png

aouamed, 2021-06-10:
> wipe out build tmp dir and run again

wipe out build tmp dir...
I started from the beginning ... up 
MACHINE=vuuno4kse DISTRO=openatv DISTRO_TYPE=dev make image
same error for samba

nickersk, 2021-06-11:
one of last ideas:

add this to samba.bb:

do_package_qa() {
}

aouamed, 2021-06-11:
> one of last ideas:
> 
> add this to samba.bb:
> 
> do_package_qa() {
> }

Ok I'm trying and I answer you ... 
thanks

aouamed, 2021-06-11:
add this to samba.bb:

do_package_qa() {
}

problem is solved
thanks for all....

> one of last ideas:
> 
> add this to samba.bb:
> 
> do_package_qa() {
> }

other error : openatv-image.bb:do_rootfs

DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies are ['/home/mido/Documents/build-enviroment/openembedded-core/meta/recipes-kernel/kmod/depmodwrapper-cross_1.0.bb:do_populate_sysroot', '/home/mido/Documents/build-enviroment/openembedded-core/meta/recipes-core/glibc/cross-localedef-native_2.33.bb:do_populate_sysroot', '/home/mido/Documents/build-enviroment/openembedded-core/meta/recipes-core/glibc/ldconfig-native_2.12.1.bb:do_populate_sysroot', 'virtual:native:/home/mido/Documents/build-enviroment/openembedded-core/meta/recipes-devtools/makedevs/makedevs_1.0.1.bb:do_populate_sysroot', 'virtual:native:/home/mido/Documents/build-enviroment/openembedded-core/meta/recipes-core/update-rc.d/update-rc.d_0.8.bb:do_populate_sysroot', 'virtual:native:/home/mido/Documents/build-enviroment/openembedded-core/meta/recipes-devtools/opkg/opkg_0.4.4.bb:do_populate_sysroot', 'virtual:native:/home/mido/Documents/build-enviroment/openembedded-core/meta/recipes-extended/pigz/pigz_2.6.bb:do_populate_sysroot', '/home/mido/Documents/build-enviroment/openembedded-core/meta/recipes-devtools/qemu/qemuwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/home/mido/Documents/build-enviroment/openembedded-core/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/mido/Documents/build-enviroment/openembedded-core/meta/recipes-extended/pbzip2/pbzip2_1.1.13.bb:do_populate_sysroot', 'virtual:native:/home/mido/Documents/build-enviroment/openembedded-core/meta/recipes-devtools/opkg-utils/opkg-utils_0.4.3.bb:do_populate_sysroot']
NOTE: Installed into sysroot: ['ldconfig-native', 'makedevs-native', 'update-rc.d-native', 'opkg-native', 'pigz-native', 'pbzip2-native', 'opkg-utils-native', 'gdk-pixbuf-native', 'shadow-native', 'shared-mime-info-native', 'debianutils-native', 'libarchive-native', 'nss-native', 'perl-native', 'gobject-introspection-native', 'libjpeg-turbo-native', 'libpng-native', 'xmlto-native', 'itstool-native', 'libxml2-native', 'lzo-native', 'e2fsprogs-native', 'zstd-native', 'nspr-native', 'perlcross-native', 'db-native', 'flex-native', 'cmake-native', 'docbook-xml-dtd4-native', 'libxslt-native', 'docbook-xsl-stylesheets-native', 'curl-native']
NOTE: Skipping as already exists in sysroot: ['depmodwrapper-cross', 'cross-localedef-native', 'qemuwrapper-cross', 'pseudo-native', 'xz-native', 'patch-native', 'quilt-native', 'kmod-native', 'qemu-native', 'libtool-native', 'autoconf-native', 'automake-native', 'openssl-native', 'pkgconfig-native', 'zlib-native', 'glib-2.0-native', 'bzip2-native', 'gettext-minimal-native', 'attr-native', 'python3-native', 'gtk-doc-native', 'ninja-native', 'meson-native', 'texinfo-dummy-native', 'gnu-config-native', 'm4-native', 'upx-native', 'sqlite3-native', 'libffi-native', 'libpcre-native', 'gettext-native', 'util-linux-native', 'gdbm-native', 'libtirpc-native', 'readline-native', 'libnsl2-native', 're2c-native', 'python3-setuptools-native', 'ucl-native', 'util-linux-libuuid-native', 'ncurses-native', 'libcap-ng-native', 'libpcre2-native']
DEBUG: sed -e 's:^[^/]*/:/home/mido/Documents/build-enviroment/builds/openatv/dev/vuuno4kse/tmp/work/vuuno4kse-oe-linux-gnueabi/openatv-image/7.0/recipe-sysroot-native/:g' /home/mido/Documents/build-enviroment/builds/openatv/dev/vuuno4kse/tmp/sysroots-components/x86_64/gdk-pixbuf-native/fixmepath /home/mido/Documents/build-enviroment/builds/openatv/dev/vuuno4kse/tmp/sysroots-components/x86_64/perl-native/fixmepath /home/mido/Documents/build-enviroment/builds/openatv/dev/vuuno4kse/tmp/sysroots-components/x86_64/gobject-introspection-native/fixmepath /home/mido/Documents/build-enviroment/builds/openatv/dev/vuuno4kse/tmp/sysroots-components/x86_64/libpng-native/fixmepath /home/mido/Documents/build-enviroment/builds/openatv/dev/vuuno4kse/tmp/sysroots-components/x86_64/xmlto-native/fixmepath /home/mido/Documents/build-enviroment/builds/openatv/dev/vuuno4kse/tmp/sysroots-components/x86_64/itstool-native/fixmepath /home/mido/Documents/build-enviroment/builds/openatv/dev/vuuno4kse/tmp/sysroots-components/x86_64/libxml2-native/fixmepath /home/mido/Documents/build-enviroment/builds/openatv/dev/vuuno4kse/tmp/sysroots-components/x86_64/e2fsprogs-native/fixmepath /home/mido/Documents/build-enviroment/builds/openatv/dev/vuuno4kse/tmp/sysroots-components/x86_64/nspr-native/fixmepath /home/mido/Documents/build-enviroment/builds/openatv/dev/vuuno4kse/tmp/sysroots-components/x86_64/docbook-xml-dtd4-native/fixmepath /home/mido/Documents/build-enviroment/builds/openatv/dev/vuuno4kse/tmp/sysroots-components/x86_64/libxslt-native/fixmepath /home/mido/Documents/build-enviroment/builds/openatv/dev/vuuno4kse/tmp/sysroots-components/x86_64/docbook-xsl-stylesheets-native/fixmepath /home/mido/Documents/build-enviroment/builds/openatv/dev/vuuno4kse/tmp/sysroots-components/x86_64/curl-native/fixmepath | xargs sed -i -e 's:FIXMESTAGINGDIRTARGET:/home/mido/Documents/build-enviroment/builds/openatv/dev/vuuno4kse/tmp/work/vuuno4kse-oe-linux-gnueabi/openatv-image/7.0/recipe-sysroot:g; s:FIXMESTAGINGDIRHOST:/home/mido/Documents/build-enviroment/builds/openatv/dev/vuuno4kse/tmp/work/vuuno4kse-oe-linux-gnueabi/openatv-image/7.0/recipe-sysroot-native:g' -e 's:FIXME_PSEUDO_SYSROOT:/home/mido/Documents/build-enviroment/builds/ope [... cut ...]
Configuring pam-plugin-shells.
Configuring util-linux-sulogin.
Configuring shadow.
Configuring update-rc.d.
Configuring rpcbind.
Configuring nfs-utils-client.
Configuring nfs-utils.
Configuring packagegroup-base-nfs.
Configuring mc-fish.
Configuring libdcerpc-binding0.
Configuring libdcerpc-server-core0.
Configuring wsdd.
Configuring samba.
Configuring packagegroup-base-smbfs-server.
Configuring enigma2-plugin-settings-defaultsat.
Configuring run-postinsts.

DEBUG: Python function do_rootfs finished

aouamed, 2021-06-11:
rebuild openatv image ...now is ok thanks for all...

https://i.postimg.cc/GmLNCRCV/1-7-1-0-0-0-0-0-0-0-20210611230311.jpg

aouamed, 2021-06-11:
probleme is sloved

closed 2021-06-11 by aouamed

geeeeeeeeeee, 2022-02-05:
I think, i had this error too.
Create: samba.bbappend ->
FILES:${PN} += "${libdir}/*"

geeeeeeeeeee, 2022-02-06:
I think, i had this bug too.
Create -> samba.bbappend:
FILES:${PN} += "${libdir}/samba/*"

geeeeeeeeeee, 2022-07-16:
I did not have a code block and did not see that it was required. Edited my GH comment now!