oe-alliance/oe-alliance-core #745: unable to build image for sf8008 do_mmc failed

enigma-openatv, 2022-01-18:
oe branch : 5.0
image : openatv 7

`| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing python function set_image_size
| DEBUG: 520390.000000 = 400300 * 1.300000
| DEBUG: 524288.000000 = max(520390.000000, 524288)[524288.000000] + 0
| DEBUG: 524288.000000 = int(524288.000000)
| DEBUG: 524288 = aligned(524288)
| DEBUG: returning 524288
| DEBUG: Python function set_image_size finished
| DEBUG: Executing shell function do_image_octagonemmc
| 0+0 records in
| 0+0 records out
| 0 bytes copied, 0.000270517 s, 0.0 kB/s
| mke2fs 1.46.4 (18-Aug-2021)
| Discarding device blocks: done         
| Creating filesystem with 131072 4k blocks and 131072 inodes
| Filesystem UUID: 7c1c6e3c-9a7a-4ac4-910f-5f41567dc5f3
| Superblock backups stored on blocks:
| 	32768, 98304
| 
| Allocating group tables: done
| Writing inode tables: done
| Creating journal (4096 blocks): done
| Copying files into the device: done
| Writing superblocks and filesystem accounting information: done
| 
| mv: cannot move '/home/ziko/build/openatv7/build-enviroment/builds/openatv/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/openatv-enigma2-image/7.0/deploy-openatv-enigma2-image-image-complete/openatv-7.0-sf8008-20220117.rootfs.ext4' to '/home/ziko/build/openatv7/build-enviroment/builds/openatv/release/sf8008/tmp/deploy/images/sf8008/sf8008-partitions/rootfs.ext4': No such file or directory
| WARNING: /home/ziko/build/openatv7/build-enviroment/builds/openatv/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/openatv-enigma2-image/7.0/temp/run.do_image_octagonemmc.29171:163 exit 1 from 'mv /home/ziko/build/openatv7/build-enviroment/builds/openatv/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/openatv-enigma2-image/7.0/deploy-openatv-enigma2-image-image-complete/openatv-7.0-sf8008-20220117.rootfs.ext4 /home/ziko/build/openatv7/build-enviroment/builds/openatv/release/sf8008/tmp/deploy/images/sf8008/sf8008-partitions/rootfs.ext4'
| WARNING: Backtrace (BB generated script):
| 	#1: do_image_octagonemmc, /home/ziko/build/openatv7/build-enviroment/builds/openatv/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/openatv-enigma2-image/7.0/temp/run.do_image_octagonemmc.29171, line 163
| 	#2: main, /home/ziko/build/openatv7/build-enviroment/builds/openatv/release/sf8008/tmp/work/sf8008-oe-linux-gnueabi/openatv-enigma2-image/7.0/temp/run.do_image_octagonemmc.29171, line 167
ERROR: Task (/home/ziko/build/openatv7/build-enviroment/meta-oe-alliance/meta-oe/recipes-distros/openatv/image/openatv-enigma2-image.bb:do_image_octagonemmc) failed with exit code '1`

atvcaptain, 2022-01-18:
i have see the issue, i commit soon the fix

atvcaptain, 2022-01-21:
fix include now in branch 5.0 and 5.1

closed 2022-01-21 by atvcaptain