oe-alliance/oe-alliance-core #750: Compile error (QA issues in samba package)

vpont, 2022-02-07:
Sice a few weeks, I can't compile enigma2-image for dm800se:
`MACHINE=dm800se DISTRO=openatv DISTRO_TYPE=release make enigma2-image`

Everything goes smoothly until it reaches the samba package:

> Summary: 1 task failed:
>  /home/openatv/build-enviroment/meta-oe-alliance/meta-oe/recipes-connectivity/samba/samba.bb:do_package_qa

Full log: 
[compile.log](https://github.com/oe-alliance/oe-alliance-core/files/8018988/compile.log)

nickersk, 2022-02-08:
add INSANE_SKIP:${PN} += "file-rdeps" to samba bb and rebuild

vpont, 2022-02-08:
I've added that line right at the end, which now looks like this:
```
...
require samba-pkgs.inc

INSANE_SKIP += "dev-so"
INSANE_SKIP:${PN} += "file-rdeps"
```

But I get the same errors.

vpont, 2022-02-08:
Tried as well in `meta-oe/recipes-connectivity/samba/samba-pkgs.inc`:
```
# The Samba server
FILES:${PN} = " \
               ${sbindir}/smbd \
               ${sbindir}/nmbd \
               ${bindir}/smbcontrol \
               ${bindir}/smbstatus \
               ${sysconfdir}/init.d \
               ${systemd_system_unitdir}/nmb.service \
               ${systemd_system_unitdir}/smb.service \
               ${libdir}/charset/*.so \
               ${libdir}/*.dat \
               ${libdir}/auth/*.so \
               ${libdir}/vfs/widelinks.so \
               ${bindir}/smbpasswd \
               ${bindir}/testparm \
               ${libexecdir}/samba/samba-bgqd \
               \
              "
RDEPENDS:${PN} += "${BPN}-common wsdd"
INSANE_SKIP:${PN} += "dev-so"
INSANE_SKIP:${PN} += "file-rdeps"
```
Not working either.

nickersk, 2022-02-08:
remove this: INSANE_SKIP:${PN} += "file-rdeps"
add this: do_package_qa(){
}

vpont, 2022-02-08:
That did the trick!
Is this particular to this box or something that needs to be fixed? Just to whether I'll need to add that override function everytime I want to build an image.
Thanks for your help!

nickersk, 2022-02-08:
samba build depends on some other things ... and it is real pain to find all the dependecnies

closed 2022-02-08 by nickersk