oe-alliance/oe-alliance-core #758: enigma2-plugin-systemplugins-serviceapp.bb

Masta2002, 2022-03-20:
Das Rezept zum erstellen hat anscheinend einen fehler so das die vorhanden Übersetzungen nicht mit ins Image gebaut werden. Bitte könnte sich das mal wer ansehen.

Die infos dazu im oATV Forum
https://www.opena.tv/openatv-7-0-py3-9-openssl-1-1-1-entwicklung/58911-erledigt-vergesene-uebersetzung-post500988.html#post500988

___________________________________________________

The recipe for creating apparently has an error, so that the existing translations are not built into the image. Please could you take a look.

The information about this in the oATV forum
https://www.opena.tv/openatv-7-0-py3-9-openssl-1-1-1-entwicklung/58911-erledigt-vergesene-uebersetzung-post500988.html#post500988

Masta2002, 2022-04-14:
The error does not only occur in oATV. I have now also tried openVIX and the translation is not installed there either.

jbleyel, 2022-04-14:
The folder 
/usr/lib/enigma2/python/Plugins/SystemPlugins/ServiceApp
contains no locale subfolder after installation.

The ServiceApp ipk do not have mo files.

nickersk, 2022-04-15:
fixed: https://github.com/oe-alliance/oe-alliance-core/commit/4d87f39e861bfe1e291a5c45cb975ed42e8d88fd

Masta2002, 2022-04-15:
Thx for fixing this

closed 2022-04-15 by Masta2002