oe-alliance/oe-alliance-core #774: [OE-A-CORE 5.1] Please update kodi-addon-inputstream-adaptive-nexus to v20.2.2

MastaG, 2022-06-20:
Running OpenATV 7.1 with Kodi 20 on Octagon sf8008.
Installed Netflix from official repository: https://github.com/castagnait/repository.castagnait/raw/kodi/repository.castagnait-2.0.0.zip

New streams starting OK, however resume stream will show green artifacts / corruption on screen.
Please update https://github.com/oe-alliance/oe-alliance-core/blob/5.1/meta-oe/recipes-mediacenter/kodi-addons/kodi-addon-inputstream-adaptive-nexus_git.bb to v20.2.2.
It has possible fixes for this issue, see:
https://github.com/xbmc/inputstream.adaptive/blob/Nexus/inputstream.adaptive/changelog.txt

If not, then please try to find a workaround to disable HiPlayer only on Netflix streams.
Don't know how to achieve this.

nickersk, 2022-06-21:
https://github.com/oe-alliance/oe-alliance-core/commit/57347d37e26dc8aec74337731e4e55735fbaeda1

closed 2022-06-21 by nickersk

nickersk, 2022-06-21:
build broken for some targets, needs more investigation, reverted for now
- see https://github.com/oe-alliance/oe-alliance-core/commit/57347d37e26dc8aec74337731e4e55735fbaeda1#comments

atvcaptain, 2023-05-16:
update version in branch 5.2