oe-alliance/oe-alliance-core #797: enable a CONFIGURATION for wireguard

antrabe, 2022-11-01:
There is a plugin to run wireguard on enigma2 but there is something wrong with ip. Could you enable the kernel to works on octagon sf8008 any image based on oe

m4dhouse, 2022-11-01:
> There is a plugin to run wireguard on enigma2 but there is something wrong with ip. Could you enable the kernel to works on octagon sf8008 any image based on oe

Hi, if you are referring to the Wireguard Vpn plugin, it works perfectly in oe-a on sf8008 and other Hisilicon boxes, there is something wrong with Vu+

antrabe, 2022-11-02:
In which image works better i tested openpli 8.3 and BH 5.1 and doesnt works only on openatv 7.1

antrabe, 2022-11-02:
im just testing new version of the plugin 1.8 and works perfect on openpli 8.3

davesayers2014, 2023-01-11:
This should be fixed with latest images with changes to openresolv

Vu+ models seems to have had changes for wireguard aswel

m4dhouse, 2023-01-11:
> This should be fixed with latest images with changes to openresolv
> 
> Vu+ models seems to have had changes for wireguard aswel

The problem was here: https://github.com/oe-alliance/oe-alliance-core/commit/55005a721b184f966eedb43df3d65a2c165b7343

closed 2023-05-16 by jbleyel