oe-alliance/oe-alliance-core #810: AX HD 61  Wireguard module

Skyrun1973, 2023-01-15:
AX HD 61
Wireguard module is missing or must be enabled .
I'm sure that WireGuard will be the future of VPNs ,i beleve that must are defoult on all Enigma2 devices. 
Regards to all

atvcaptain, 2023-01-15:
on feed  kernel-module-wireguard-4.4.35  i have check openatv 7.2

Skyrun1973, 2023-01-15:
Thanks Captain for checking, but the module exists but is not loadedInviato da Posta per Windows Da: CaptainInviato: domenica 15 gennaio 2023 12:31A: oe-alliance/oe-alliance-coreCc: Skyrun1973; AuthorOggetto: Re: [oe-alliance/oe-alliance-core] AX HD 61 Wireguard module (Issue #810) on feed kernel-module-wireguard-4.4.35 i have check openatv 7.2

Skyrun1973, 2023-01-15:
Thanks Captain for checking, but the module exists but is not loaded

Huevos, 2023-01-15:
https://www.world-of-satellite.com/showthread.php?65067-Wireguard-on-enigma2

Please read. Shows a working setup.

m4dhouse, 2023-01-16:
In the feeds the module is present, it is installed but not loaded, something must be enabled in defconfig

## log ##
modprobe: ERROR: could not insert 'wireguard': Unknown symbol in module, or unknown parameter (see dmesg)
Starting all configured Wireguard interfaces
[#] ip link add wg0 type wireguard
RTNETLINK answers: Operation not supported
Unable to access interface: Protocol not supported
[#] ip link delete dev wg0
Cannot find device "wg0"

davesayers2014, 2023-01-16:
Possibly similar fixes needed as this commit https://github.com/oe-alliance/oe-alliance-core/commit/55005a721b184f966eedb43df3d65a2c165b7343

closed 2023-05-16 by jbleyel