oe-alliance/oe-alliance-core #816: .bash_profile not working

jeepcook, 2023-01-25:
Hello, 

With the 5.2 and openatv 7.2, the content of .bash_profile is not executed. Example, in mine there is just alias ll='ls -al' and the command ll doesn't work, I have to execute manually the alias command.
No problem in 7.1 EOL.

Thanks

nickersk, 2023-01-25:
login as: root
root@vusolose:~# ll
-sh: ll: command not found

root@vusolose:~# **echo "alias ll='ls -la'" > .profile**
root@vusolose:~# more .profile
alias ll='ls -la'

**AFTER RELOGIN:**
root@vusolose:~# ll
drwx------    8 root     root           904 Jan 25 11:28 .
drwxr-xr-x    3 root     root           224 Dec  1 06:22 ..

jeepcook, 2023-01-25:
Hell nickersk, 
it's ok now using .profile instead of .bash_profile.
Thanks for your help !!

closed 2023-01-25 by jeepcook