oe-alliance/oe-alliance-core #818: Change directory with some plugins crashes enigma2

ArnoldSchiller, 2023-03-09:
TypeError: in method 'new_eServiceReference', argument 3 of type 'std::string const &'
Additional information:
Wrong number or type of arguments for overloaded function 'new_eServiceReference'.
  Possible C/C++ prototypes are:
    eServiceReference::eServiceReference()
    eServiceReference::eServiceReference(int,int)
    eServiceReference::eServiceReference(int,int,int)
    eServiceReference::eServiceReference(int,int,int,int)
    eServiceReference::eServiceReference(int,int,int,int,int)
    eServiceReference::eServiceReference(int,int,int,int,int,int)
    eServiceReference::eServiceReference(int,int,int,int,int,int,int)
    eServiceReference::eServiceReference(int,int,std::string const &)
    eServiceReference::eServiceReference(eServiceReference const &)
    eServiceReference::eServiceReference(std::string const &)
01:38:09.5406 
01:38:09.5407 [ePyObject] (PyObject_CallObject(<bound method ActionMap.action of <Components.ActionMap.HelpableActionMap object at 0xaf770030>>,('PluginMovieSelectionActions', 'EMCOK')) failed)

Can not handle characters like ß like 'große' in the file system. All it takes is one file in a directory and a plugin that goes there crashes.

Huevos, 2023-03-09:
Which plugin? Which image distro? Which version? You need to provide logs.

Also this is not the place. This is an enigma problem. It should be reported to the enigma team of the distro you are using, or to the plugin author.

ArnoldSchiller, 2023-03-13:
openAtv 7.2

jbleyel, 2023-03-13:
This is an EMC issue and your file is probably not valid UTF-8.

closed 2023-03-13 by jbleyel

ArnoldSchiller, 2023-03-14:
Well, it can crash in other places as well, not just at EMC. For example if you edit via telnet on the console with vi userbouqet.favorites.tv and have too many spaces at the end of the line. So it happens not only with filenames. Does this mean that the installed vi does not support utf8? The common thing is that you can't see the wrongly encoded space.